SvenVD/rpisurv #1: not working

afburgos, 2015-11-15:
This is what the command line says after configure and launch it:

pi@pi2recepcion /usr/local/bin/rpisurv $ python surveillance.py
Traceback (most recent call last):
  File "surveillance.py", line 12, in <module>
    from config import cfg
  File "/usr/local/bin/rpisurv/config.py", line 4, in <module>
    cfg = yaml.load(ymlfile)
  File "/usr/lib/python2.7/dist-packages/yaml/__init__.py", line 71, in load
    return loader.get_single_data()
  File "/usr/lib/python2.7/dist-packages/yaml/constructor.py", line 37, in get_single_data
    node = self.get_single_node()
  File "/usr/lib/python2.7/dist-packages/yaml/composer.py", line 36, in get_single_node
    document = self.compose_document()
  File "/usr/lib/python2.7/dist-packages/yaml/composer.py", line 55, in compose_document
    node = self.compose_node(None, None)
  File "/usr/lib/python2.7/dist-packages/yaml/composer.py", line 84, in compose_node
    node = self.compose_mapping_node(anchor)
  File "/usr/lib/python2.7/dist-packages/yaml/composer.py", line 133, in compose_mapping_node
    item_value = self.compose_node(node, item_key)
  File "/usr/lib/python2.7/dist-packages/yaml/composer.py", line 84, in compose_node
    node = self.compose_mapping_node(anchor)
  File "/usr/lib/python2.7/dist-packages/yaml/composer.py", line 133, in compose_mapping_node
    item_value = self.compose_node(node, item_key)
  File "/usr/lib/python2.7/dist-packages/yaml/composer.py", line 64, in compose_node
    if self.check_event(AliasEvent):
  File "/usr/lib/python2.7/dist-packages/yaml/parser.py", line 98, in check_event
    self.current_event = self.state()
  File "/usr/lib/python2.7/dist-packages/yaml/parser.py", line 449, in parse_block_mapping_value
    if not self.check_token(KeyToken, ValueToken, BlockEndToken):
  File "/usr/lib/python2.7/dist-packages/yaml/scanner.py", line 116, in check_token
    self.fetch_more_tokens()
  File "/usr/lib/python2.7/dist-packages/yaml/scanner.py", line 257, in fetch_more_tokens
    % ch.encode('utf-8'), self.get_mark())
yaml.scanner.ScannerError: while scanning for the next token
found character '\t' that cannot start any token
  in "conf/surveillance.yml", line 8, column 1

Please, tell me what to do.

Thanks

SvenVD, 2015-11-15:
Hi thanks for testing it out, there seems to be a problem with your config file. You seem to use a "tab" character where it shouldn't be used. (line 8, column 1)

Please run the following command and post results here (but remove passwords in output first):
cat -A /etc/rpisurv

SvenVD, 2015-11-15:
Alternatively validate the config file by pasting it here:

http://codebeautify.org/yaml-validator and click validate

afburgos, 2015-11-20:
Sorry for the delay, it was that, the tab character. Once corrected, it
works like a charm.

One question, do the system reconnect every x mins to avoid time delay in
the images?

One of the log files (system.log I think), is filling with logs every
minute. Sometime it will fulfill the sd card, can it be deactivated once
everything runs smoothly?

Regards

SvenVD, 2015-11-20:
Hi,

I assume you mean this log: /usr/local/bin/rpisurv/logs/main.log?
Currently the default setting for logging is debug mode for ease of troubleshooting. 
This shouldn't fill the SDcard since the logs will be rotated by default, however it seems to not do this at the moment. The cause is a wrong value for maxBytes in the configuration for the logging.

Please change 

        filename: logs/main.log
        maxBytes: 1e6
        backupCount: 10

to

        filename: logs/main.log
        maxBytes: 1000000
        backupCount: 10

in the /usr/local/bin/rpisurv/conf/logging.yml config file to have the log file rotated when it reaches 1Mb in size. I will update the default value in the git repo also.

Secondly:
You can always disable DEBUG mode logging by changing 

loggers:
    l_default:
        level: DEBUG
        handlers:
          - h_console
          - h_rotfile
        propagate: False

to 

loggers:
    l_default:
        level: INFO
        handlers:
          - h_console
          - h_rotfile
        propagate: False

again in the /usr/local/bin/rpisurv/conf/logging.yml config file

SvenVD, 2015-11-20:
One question, do the system reconnect every x mins to avoid time delay in
the images?
> The system checks every 25 seconds if the streams you defined in the config file are still connectable. If a (or more) previous unconnectable streams become available or one or more streams become unavailable it will redraw the screen with the connectable cams

>Next to this the systems checks every 5 seconds if the omxplayer process per stream is still running. If not it will try to restart the omxplayer process for that stream only. This is especially handy for low budget ip cameras which tend to crash the omxplayer process now and then.

closed 2015-11-20 by SvenVD