SvenVD/rpisurv #4: Works!

goodtimeclub, 2016-02-26:
Great stuff here @SvenVD - got it going eventually but ran into some funny issues along the way, namely permissions on running install.sh. `/usr/local/bin/rpisurv` never got created, neither did `/home/pi/surveillance`

Ended up copying the contents of `surveillance/conf` into `/usr/local/bin/rpisurv` & everything fell into place.

Curious if you know of a way to stream an mjpeg stream in this way(autonomous, service-monitoring, stable fps). I'm using [https://github.com/ccrisan/motioneyeos](https://github.com/ccrisan/motioneyeos) for the camera side but am looking to setup a chromeless/fullscreen monitor to stream the images to.

SvenVD, 2016-02-26:
Hi,

Thanks for testing out!
Did you run ./install.sh as the root user/sudo? 

I do not know any software that does the same for mjpeg streams and runs on the pi.

goodtimeclub, 2016-02-26:
ran `sudo sh ~/rpisurv/install.sh` after a fresh install of Raspbian Jessie (2016-02-09) on a Pi2 B. I'll run through all the same steps next week again & report back ;)

re: mjpeg monitor, I'm using a modified version of a omxplayer start/stop/repair script somebody made - still hunting for the holy grail of "IP Camera capture/serve & IP Camera monitor" on 2 separate Pi

SvenVD, 2016-02-26:
Ok thanks! A suggestion is to run the script with sudo sh -x ~/rpisurv/install.sh. And paste the output here.

goodtimeclub, 2016-03-03:
Ok, starting from scratch today: Raspbian Jessie on a Pi2 Model B. 

1. `sudo apt-get update && sudo apt-get upgrade`
2. got Git running by adding an ssh key by following [https://help.github.com/articles/generating-a-new-ssh-key-and-adding-it-to-the-ssh-agent/](https://help.github.com/articles/generating-a-new-ssh-key-and-adding-it-to-the-ssh-agent/)
3. `git clone git@github.com:SvenVD/rpisurv.git`
4. `sudo sh -x ~/rpisurv/install.sh` output: [rpisurv-install-log.txt](https://github.com/SvenVD/rpisurv/files/157112/rpisurv-install-log.txt)

Some poking around shows the files didn't get copied properly:

`pi@raspberrypi:~ $ ls /etc/rpisurv`
**/etc/rpisurv**
`pi@raspberrypi:~ $ ls -la /etc/rpisurv`
**lrwxrwxrwx 1 root root 44 Mar  3 15:50 /etc/rpisurv -> /usr/local/bin/rpisurv/conf/surveillance.yml**
`pi@raspberrypi:~ $ cat /etc/rpisurv`
**cat: /etc/rpisurv: No such file or directory**

Hope this helps!

SvenVD, 2016-03-05:
Hi, thanks for reporting back!

After some digging I found out that /bin/sh does link to the dash shell instead of the bash shell. 

So while `sh ~/rpisurv/install.sh` does not work, `bash ~/rpisurv/install.sh` or `~/rpisurv/install.sh` does work.
However I added an extra check in the installer script to check for bash script in this commit: https://github.com/SvenVD/rpisurv/commit/b9869aba9c4d20e4f225a1a11d20d837abe26ffa

closed 2016-03-05 by SvenVD

goodtimeclub, 2016-03-07:
gotcha ;)

drthanwho, 2017-04-20:
I simply run `sudo /home/pi/rpisurv/install.sh` and it worked for me. Hopefully i didn't do it wrong