SvenVD/rpisurv #5: None Not Connectable ?

Sylvian, 2016-03-23:
Hi,

I've to try install this service on a new RPI and I got Strange error (i take a screen from console)

[image]

And It never start.

I've check the rstp url with VLC on the desktop mode and it works.
When I was in console I've check the network with an apt-get install htop for example and it works too..

So I don't know what to do next if u get a trick or Something to fix this issue :)

__________________________________
Hardware:
Camera IP Axis P1405-LE
RPI 3 Model B with a fresh 2016-03-18-raspbian-jessie
__________________________________

Thanks for your help :)

Sylvian, 2016-03-23:
I found the bug by myself, sorry for inconvenience ^^ This script is working awesome !

I didn't set a port in my rtsp adress, because it was working with port 80 on VLC, but for some unknow reason, I found the rtsp server was set to port 554 by default with Axis.

Then I set it to 554 and it works fine ^^

The last thing I didn't find how to yet is to "center" the video or use the entire screen.

I join a picture you can see there is some space left on bottom and right.

[image]

Again thanks for this usefull script I think you should rename it: "RPI IP Camera Monitor" because I find it after 2 hours on Google.

SvenVD, 2016-03-23:
Hi thanks for testing it out!

It seems your resolution setting as detected by the `fbset -s` command seems to be wrong.

Do you know the native resolution of your screen? I suggest you play a bit with the settings in /boot/config.txt especially the following:

# uncomment to force a console size. By default it will be display's size minus
# overscan.

#framebuffer_width=1920
#framebuffer_height=1080

Sylvian, 2016-03-24:
Hi,

thank you for answering me, I made some research about this but it seems the  `fbset -s` result was perfect, my English is not good enough so I made a picture to explain the real bug ^^

[image]
I've try this on 3 different TV and a PC monitor, It's the same bug everytime I don't know why...
Maybe because I use only one cam ?

SvenVD, 2016-03-24:
In your first screenshot it's saying, autodetected resolution of 1824 and 982. This seems like a non standard resolution and similar to this problem https://www.raspberrypi.org/forums/viewtopic.php?f=91&t=99820. Maybe it's the overscan  options?. I do not have a rpi3 so I can not test. But at first sight this not related to rpi3. Can you test with the overscan options?

Thanks

Sylvian, 2016-04-12:
Hi , I'm coming back a bit late but the problem was solved with uncomment the line "disable_overscan" on the config.txt file.

Thanks for your help :)

SvenVD, 2016-04-12:
Thanks for confirming the resolution

closed 2016-04-12 by SvenVD