SvenVD/rpisurv #6: Video Stretched

AtomicLaser, 2016-03-23:
Great script!

I've managed to get two IP cameras connected at the moment (need to figure out the stream IDs for the other two I have, however the image is looking very stretched out of proportions 

Any ideas how to fix this?

SvenVD, 2016-03-24:
No, this is by design. Rpisurv tries to use all pixels on the screen. For surveillance it's more useful to use all available screen instead of having half of the screen black and unused. You can choose if you want the two cameras in 2 columns or 2 rows. For 2 columns set `nr_of_columns: "2"`, This usually gives a better result then 2 rows. For 2 rows set `nr_of_columns: "1"`

Greetings

AtomicLaser, 2016-03-24:
Thanks, I've managed to get 4 of my cameras online, so I don't see this issue anymore.  Would rather have blank spaces than Rpisurv stretching images out of proportion.  Assume this would also happen if I had a 5 camera system (waiting for another IP cam to come) with 4 cams displaying right and the 5th being stretched?

another thing of note although not too fussy is that some of the image goes off screen.  I've set the correct display parameters in the boot config.  No biggie as it crops off the camera name at the bottom of the display and the time and date from the right of the screen.

[image]

SvenVD, 2016-03-24:
Indeed 5th camera will be stretched.

What is the output of fbset -s?

AtomicLaser, 2016-03-25:
pi@raspberrypi:~ $ fbset -s                                                      
                                                                                 
mode "1920x1080"                                                                 
    geometry 1920 1080 1920 1080 16                                              
    timings 0 0 0 0 0 0 0                                                        
    rgba 5/11,6/5,5/0,0/16                                                       
endmode                                                                          
                                                                                 
pi@raspberrypi

SvenVD, 2016-03-25:
Is 1920x1080 the native resolution of your screen?

rpisurv uses the output of fbset for it's calculations. To  be sure rpisurv parsed the correct resolution you can see this in `/usr/local/bin/rpisurv/logs/main.log` but only after you enabled debug logging in ` /root/rpisurv/surveillance/conf/logging.yml`

Change

```
loggers:
    l_default:
        level: INFO
        handlers:
          - h_console
          - h_rotfile
        propagate: False
```

to

```
loggers:
    l_default:
        level: DEBUG
        handlers:
          - h_console
          - h_rotfile
        propagate: False
```

AtomicLaser, 2016-03-25:
I have a full HD TV, so pretty sure it's native resolution is 1920 x 1080.

I'll have to give the debugger a try when I next have it connected up to the TV, as I have it running on a 5" TFT monitor at the moment

Thanks :)

AtomicLaser, 2016-03-26:
What would be great if there was an option in the config where we could choose if we wanted to use 100% of the screen's pixels.  I don't mind having blank spaces where I don't have enough screens for the quads.

I was looking in the surveillance.py file to see if I can edit it to stop the stretching would this be an edit to lines 140 - 144?

SvenVD, 2016-03-28:
Hi

I added an option to you can override the autocalculated width and/or height of the windows. It's already in master branch.

Please check it out https://github.com/SvenVD/rpisurv/commit/802774244d76c37456c38355f4502bcc1b91f360

closed 2016-04-12 by SvenVD