SvenVD/rpisurv #9: Updates to this tool

goodtimeclub, 2016-04-18:
Wondering how you would recommend safely updating existing instances of this. Any suggestions?

SvenVD, 2016-04-18:
The update process can be done like this:

```
#Save your config
cp -v /usr/local/bin/rpisurv/conf/surveillance.yml /tmp/surveillance.yml.save
#Update to latest
git pull
#Run installer
run ./install.sh
#Put back your config
cp -v /tmp/surveillance.yml.save /usr/local/bin/rpisurv/conf/surveillance.yml
```

Note: this can only be done if there are no breaking changes in config. Otherwise you need to edit the config by hand. Up till now no breaking have been introduced however

goodtimeclub, 2016-04-19:
Thanks, Sven! Thought I'd check to see if that was the case - will keep my Pis up to date ;)
>

closed 2016-04-26 by SvenVD

goodtimeclub, 2016-04-27:
ps worked like a charm - great advances so far ;)

afburgos, 2017-02-15:
I´m trying to update but het this error:

pi@RpiConserjeria ~/rpisurv $ sudo git pull
Updating 2f07e95..1528dd0
error: Your local changes to the following files would be overwritten by merge:
        surveillance/surveillance.py
        surveillance/worker.py
Please, commit your changes or stash them before you can merge.
Aborting

What have to do?

Thanks

afburgos, 2017-02-15:
Also:

pi@RpiConserjeria ~/rpisurv $ sudo git pull https://github.com/SvenVD/rpisurv.git
From https://github.com/SvenVD/rpisurv
 * branch            HEAD       -> FETCH_HEAD
Updating 2f07e95..1528dd0
error: Your local changes to the following files would be overwritten by merge:
        surveillance/surveillance.py
        surveillance/worker.py
Please, commit your changes or stash them before you can merge.
Aborting

afburgos, 2017-02-15:
Ok, solved with 

git reset --hard