SvenVD/rpisurv #10: feature

gabiX87, 2016-04-26:
Hi, I registered to request a feature if it's possible.

I am running your script on my rPI B+. Works like a charm. So my request:
When a rtsp stream goes down for example on a 2x2 matrix (with 4 urls defined in the config file) I do not want to recalculate , but leave the 4 windows and just display a black screen or a message "disconnected" or something. Because for example with 2 streams only the image looks weird , verticaly stretched on 4:3 and 16:10 also. 3 streams is quite ok but , the 2x2 would be the best for window size ratio.
 Could you add an option into the config file to turn on or turn off this recalculation+resizing.

Thanks

SvenVD, 2016-04-26:
Hi, the weird stretching is by design. The more screen space you use the more easily to spot movement in the screen. However if you do not like this you can use the advanced options to prevent this.
See: https://github.com/SvenVD/rpisurv/commit/802774244d76c37456c38355f4502bcc1b91f360
If you enable one or both of these advanced options, autostretching will be disabled. However, you need to calculate the pixels yourself. If you only enable one advanced option. Let's say for height, rpisurv will still calculate the width and vice versa.
One trick to disable the autostretching, but still let rpisurv calculate the streams width and height is to set both advanced option to absurd high amount. Like 10000 for height and 10000 for width, rpisurv will fall back to autocalculation then, but without autostretching of last stream in case you do not have an even number of streams

gabiX87, 2016-04-27:
Thank you it works as I need it. You might wanna add some comments into the config file so people know when they open it.

SvenVD, 2016-04-27:
Hi, with two streams you can set the fixed_height to something half the screen height or something else you like.

An improvement would be that rpisurv detects the native stream resolution somehow and kept that ratio width/height. But that is currently not on the planning.

puck, 2016-05-20:
Hey,

I kinda read the initial request differently than how this thread was taken it. I would expect that if a stream dies, show a blank screen (or a fixed image staying it is unreachable), and then when it is working again, swap the stream back in.

This would certainly be my preference.

gabiX87, 2016-05-20:
Yes.

SvenVD, 2016-05-20:
Hi, I added the following option in commit https://github.com/SvenVD/rpisurv/commit/c8976689b5649491ad3011a4fdd5a90340cccae3

    #rpisurv detects the connectable streams on startup, 
    #if this option is enabled it will keep the first calculated layout for the lifetime of the rpisurv running even if some cameras go down and come up, instead of redrawing the screen
    #When this option is set to True, rpisurv ignores all cameras configured that were not connectable when rpisurv was started
    keep_first_screen_layout: False

Off course this can be used in combination with the advanced settings to achieve what you asked.

Please let me know how it works out.

SvenVD, 2016-06-05:
Any update? Can this ticket be closed?

puck, 2016-06-06:
I can't easily reproduce the reported issue as all my streams come from the same server, so even if I stop one stream, the code still considers the server as contactable, so doesn't try and rearrange the screen. But the logic you describe and what is in the code certainly makes sense to me.

closed 2016-06-11 by SvenVD