SvenVD/rpisurv #11: Warn if gpu_mem is too low?

puck, 2016-05-20:
I found I wasn't able to start more than 2 streams with the default gpu_mem setting. Perhaps having some logic that warns with a suitable warning if you fail to spawn additional streams?  I only thought this might be an issue after looking at the README file.

SvenVD, 2016-05-20:
Normally there should be some  kind of error logged. Did you see something in the logs?

SvenVD, 2016-06-05:
I added a warning in the logs when free gpu mem is too low : https://github.com/SvenVD/rpisurv/commit/3d3461c580aaab93c03b071e448d33f2d42edad3

Please check it out

puck, 2016-06-06:
I set gpu_mem back to default, and I can confirm that the logging works beautifully and I now know exactly what is wrong!

Thank you.

closed 2016-06-06 by puck