SvenVD/rpisurv #15: sad errors

jdetmold, 2016-08-03:
trying to get this working with no luck keep getting the following. the feeds do work in vlc.

[image]
sorry if the photo is not clear enough I can type out the error

the logs show:
```
2016/08/03 10:54:17 - l_default - DEBUG - autodetected resolution of['1824', '984']
2016/08/03 10:54:17 - l_default - DEBUG - nr_of_columns = 2
2016/08/03 10:54:17 - l_default - DEBUG - camera_streams config option exist, using this one
```

and my rpisurv is:
```
essentials:
    #These are the rtsp streams you want the program to display
    #The streams will be displayed in the order you present them here
    camera_streams:
        rtsp_url: "rtsp://**unifiserver**:7447/55b7feff1a3352fa026df0d1_2"
        rtsp_url: "rtsp://**unifiserver**:7447/55b7feff1a3352fa026df0d1_2"

    #How many columns you want the program to use, it will autocalculate the amount of row needed based on the resolution of your screen
    nr_of_columns: "2"

    #rpisurv detects the connectable streams on startup, 
    #if this option is enabled it will keep the first calculated layout for the lifetime of the rpisurv running even if some cameras go down and come up, instead of redrawing the screen
    #When this option is set to True, rpisurv ignores all cameras configured that were not connectable when rpisurv was started
    keep_first_screen_layout: False

#!!!Normal users do not need to edit this section!!!
advanced:
    ##Enable this option if you want to have a fixed width of all your camera streams, 
    ##By default rpisurv autocalculates this value, this can cause streams to get "stretched", 
    ##if this value exceeds the available height, rpisurv will fallback to autocalculation   
    #fixed_width: 500
    
    ##Enable this option if you want to have a fixed height for all your camera streams, 
    ##By default rpisurv autocalculates this value, this can cause streams to get "stretched",
    ##if this value exceeds the available height, rpisurv will fallback to autocalculation  
    #fixed_height: 500

#These are fallbacks if autodection fails;
#Normally you do not need to configure these
fallbacks:
    resolution:
        width: "1920"
        height: "1080"

```

jdetmold, 2016-08-03:
never mind I'm just an idiot!
Great work btw!

closed 2016-08-03 by jdetmold

SvenVD, 2016-08-03:
Hi, probably the url was wrong, * are not valid, can you confirm this can be handy for other users.

Thanks

jdetmold, 2016-08-04:
no the *'s were to hide the server info as i said the feeds worked in vlc it was just missing the -'s before the feed in the config