SvenVD/rpisurv #16: fuzzy images on 3 of 4 cams

skueppers, 2016-08-30:
First of all thanks for your great script which runs (nearly) charming after a few pitfalls.
My current problem is that 3 of my 4 cams only show a "fuzzy" picture - as well as if streaming "by default" oder "over tcp".
[image]
All in all my RasPi uses currently 50-60% CPU

Although this is not a specific problem with rpisurv maybe you have the crucial hint

SvenVD, 2016-08-30:
Did you assign enough memory to the GPU?

skueppers, 2016-08-31:
currently running with gpu_mem=384 "only"
setting gpu_mem to 512 makes the system ghastly slow

SvenVD, 2016-08-31:
What version of omxplayer are you using?

closed 2016-09-09 by SvenVD