SvenVD/rpisurv #17: Issue with 6 cameras (or more)

Yeomans, 2016-09-09:
Hi,

This is a really great application, really like it. I'm having an issue with displaying 6 cameras though and I wonder if it's just a limitation of the hardware?

Here is a video of what happens when configured for 3 columns and autosizing enabled. 
https://sendvid.com/ojrf3td2

For some reason it seems to be confused as to where to place the cameras. 5 works fine with the last image stretch, but as soon as a 6th is added this happens. When I add a 7th, the display cuts out completely when drawing the 3rd row. Very odd.

Using an RPi 3 with gpu_mem=512 connected to a 1920x1080 monitor (tested on 2 monitors just in case). Debug output showed 342MB of free memory with 6 cameras displayed.

Ideally, if the hardware allows for it I would like to get it working with more than 6 cameras eventually.

Thanks in advance.

Yeomans, 2016-09-09:
UPDATE:

So I think this might have just been a problem with the size of the RTSP stream I was connecting. As soon as I changed all my streams from 1024x576 to 640x360 it worked nicely in the 3x3 grid. Not sure if that sort of behavior is expected though so I'll leave this open for comment.

Cheers

Yeomans, 2016-09-09:
UPDATE 2:

It might be an issue with omxplayer player resizing. I tested how far it could go. It's able to display a 4x3 grid with 4 columns, but as soon as it tries to draw the 4th row the cameras flicker. Even setting a height of 270 so it shows a blank row down the bottom causes the flicker. Unfortunately I don't have the ability to output my rtsp stream at a lower resolution so a bit stuck I if want to get more than 12 cameras (or to display their proper aspect ratio).

Yeomans, 2016-09-09:
UPDATE 3.

Sorry for the spam! I'm just keen to get this working. Looks like these issues are with omxplayer. I made my own script to display cameras and cannot get more than 8. If I manually run the omxplayer cmd to open the stream of a 9th camera it gives me the tv-service error. Guess this is just some limitation :(

SvenVD, 2016-09-09:
Yes probably a limitation by omxplayer? Maybe ask the developers of omxplayer?

closed 2016-09-24 by SvenVD