SvenVD/rpisurv #22: not all RTSP are equal?

eaglerare, 2017-04-13:
First, thanks so much for your work on this. Everything about it seemed to work just as expected. I had no problem getting 3 cameras showing on my TV via Pi2 hdmi with rpisurv :+1: +1:

I ran upon an issue, and I'm unsure how diagnose it any further. I decided to add a 4th feed to my system, by using VLC RTSP server to stream the screen of a PC. The idea being that, I would then have my cameras plus an information stream flowing as a 4th source. The use of VLC to stream the screen via RTSP seems to work fine, I can access the stream from other clients, and even from my pi2 bu using the following command: sudo omxplayer --live -o hdmi rtsp://xx.xx.xx.xx:8554/scr.sdp
If I change one of my already tested and functioning stream sources to use this new one, it causes an error in surveillance.py  

Traceback (most recent call last):
  File "surveillance.py", line 10, in <module>
    from config import cfg
  File "/usr/local/bin/rpisurv/config.py", line 4, in <module>
    cfg = yaml.load(ymlfile)
  File "/usr/lib/python2.7/dist-packages/yaml/__init__.py", line 71, in load
    return loader.get_single_data()
  File "/usr/lib/python2.7/dist-packages/yaml/constructor.py", line 37, in get_single_data
    node = self.get_single_node()
  File "/usr/lib/python2.7/dist-packages/yaml/composer.py", line 36, in get_single_node
    document = self.compose_document()
  File "/usr/lib/python2.7/dist-packages/yaml/composer.py", line 55, in compose_document
    node = self.compose_node(None, None)
  File "/usr/lib/python2.7/dist-packages/yaml/composer.py", line 84, in compose_node
    node = self.compose_mapping_node(anchor)
  File "/usr/lib/python2.7/dist-packages/yaml/composer.py", line 133, in compose_mapping_node
    item_value = self.compose_node(node, item_key)
  File "/usr/lib/python2.7/dist-packages/yaml/composer.py", line 84, in compose_node
    node = self.compose_mapping_node(anchor)
  File "/usr/lib/python2.7/dist-packages/yaml/composer.py", line 133, in compose_mapping_node
    item_value = self.compose_node(node, item_key)
  File "/usr/lib/python2.7/dist-packages/yaml/composer.py", line 82, in compose_node
    node = self.compose_sequence_node(anchor)
  File "/usr/lib/python2.7/dist-packages/yaml/composer.py", line 110, in compose_sequence_node
    while not self.check_event(SequenceEndEvent):
  File "/usr/lib/python2.7/dist-packages/yaml/parser.py", line 98, in check_event
    self.current_event = self.state()
  File "/usr/lib/python2.7/dist-packages/yaml/parser.py", line 393, in parse_block_sequence_entry
    "expected <block end>, but found %r" % token.id, token.start_mark)
yaml.parser.ParserError: while parsing a block collection
  in "conf/surveillance.yml", line 6, column 9
expected <block end>, but found '<block mapping start>'
  in "conf/surveillance.yml", line 19, column 10

Any thoughts?

eaglerare, 2017-04-13:
also, I notice that surveillance.yml seems to be just a copy of /etc/rpisurv so I ran a diff of it vs my known good rpisurv I don't see any issues. 

diff ./surveillance.yml /etc/rpisurvwrx 
16c16
<         - rtsp_url: "rtsp://xx.xx.xx.xx:8554/p1.sdp"
---
>         - rtsp_url: "rtsp://pi:unl0ckpi@xx.xx.xx.xx:554/pxay1.sdp"
19c19
<          rtsp_over_tcp: true
---
>          #rtsp_over_tcp: true

SvenVD, 2017-04-14:
Did you use tabs in /surveillance.yml

SvenVD, 2017-04-19:
The error you posted seems to be an a parse error in trying to read the  surveillance.yml file. Can you paste the complete  surveillance.yml here, so we can have a look what might be wrong?

Thanks

closed 2017-05-01 by SvenVD

blackangelPL, 2018-03-19:
I have the same problem

Mar 19 13:44:06 raspberrypi systemd[1]: rpisurv.service start request repeated too quickly, refusing to start.
Mar 19 13:44:06 raspberrypi systemd[1]: Failed to start Rpisurv Raspberry Pi Surveillance.
Mar 19 13:44:06 raspberrypi systemd[1]: Unit rpisurv.service entered failed state.
pi@raspberrypi:~/rpisurv $ mc
  GNU nano 2.2.6                                         File: /var/log/syslog                                                                                Modified

Mar 19 14:31:03 raspberrypi rpisurv[4147]: while not self.check_event(MappingEndEvent):
Mar 19 14:31:03 raspberrypi rpisurv[4147]: File "/usr/lib/python2.7/dist-packages/yaml/parser.py", line 98, in check_event
Mar 19 14:31:03 raspberrypi rpisurv[4147]: self.current_event = self.state()
Mar 19 14:31:03 raspberrypi rpisurv[4147]: File "/usr/lib/python2.7/dist-packages/yaml/parser.py", line 439, in parse_block_mapping_key
Mar 19 14:31:03 raspberrypi rpisurv[4147]: "expected <block end>, but found %r" % token.id, token.start_mark)
Mar 19 14:31:03 raspberrypi rpisurv[4147]: yaml.parser.ParserError: while parsing a block mapping
Mar 19 14:31:03 raspberrypi rpisurv[4147]: in "conf/surveillance.yml", line 1, column 1
Mar 19 14:31:03 raspberrypi rpisurv[4147]: expected <block end>, but found '-'
Mar 19 14:31:03 raspberrypi rpisurv[4147]: in "conf/surveillance.yml", line 14, column 1
Mar 19 14:31:03 raspberrypi systemd[1]: rpisurv.service: main process exited, code=exited, status=1/FAILURE
Mar 19 14:31:03 raspberrypi systemd[1]: Unit rpisurv.service entered failed state.
Mar 19 14:31:04 raspberrypi systemd[1]: rpisurv.service holdoff time over, scheduling restart.
Mar 19 14:31:04 raspberrypi systemd[1]: Stopping Rpisurv Raspberry Pi Surveillance...
Mar 19 14:31:04 raspberrypi systemd[1]: Starting Rpisurv Raspberry Pi Surveillance...
Mar 19 14:31:04 raspberrypi systemd[1]: rpisurv.service start request repeated too quickly, refusing to start.
Mar 19 14:31:04 raspberrypi systemd[1]: Failed to start Rpisurv Raspberry Pi Surveillance.
Mar 19 14:31:04 raspberrypi systemd[1]: Unit rpisurv.service entered failed state.
Mar 19 14:31:08 raspberrypi vncserver-x11[658]: AgentInitCheck: no response from agent

if i use omxplayer 
omxplayer --live --win "0 100 1824 642" rtsp://10.20.2.210/user=admin_password=_channel=0_stream=1.sdp
it works 

When i put the same adres to rpirurv it wont start