SvenVD/rpisurv #23: Issue with RTSP validation "Not Connectable (failed rtsp validation, no response from rtsp server)"

WallnerK, 2017-04-28:
Hi!

First of all thanks for that great piece of software. I'm currently testing serveral setups for potential commercial use.

Yesterday I tried to connect Intellinet IP-Dome cameras - those come with RTSP support and quite good RTSP-configuration possibilities.
All cameras worked perfect on the first try through VLC. 
But it seems like rpisurv couldn't validate the existance of a RTSP-Server.

I worked arround the issue by changing the code in surveillance.py - to evaluate the failed test for RTSP-validation to true.

` #If we come to this point it means we have some data received, before we return True we need to check if are connected to an RTSP server
        if not rtsp_response:
            logger.error(self.hostname + " : " + str(self.port) +" Not Connectable (failed rtsp validation, no response from rtsp server)")
            return True`

All streams connected immediately.

Of course I'm not very happy with that - I don't fully understand the code though (not a programmer).
Is there anything else I could do to fix that validation? Is it a configuration problem on server (camera) side?

SvenVD, 2017-04-28:
As I do not have an  Intellinet IP-Dome  camera I can not troubleshoot further. A network packet dump from the RTSP connection setup between rpisurv and the camera is needed to troubleshoot this further. Is this camera RTSP 1.0 compatible?

Thanks

SvenVD, 2017-04-28:
I am interested in what kind of commercial use you are testing, contact me @ svenvd.github@gmail.com

closed 2017-06-11 by SvenVD