SvenVD/rpisurv #25: multiple streams and blanking hdmi

arrfour, 2017-06-14:
I'm struggling with an issue displaying 3 or more 1080p rtsp streams to an HDMI 1080p display.
A single RTSP works flawlessly and I've tested with "Automatic" display detection on an HDMI and DVI monitors discretely.
I've set gpu_mem=512 and I'm running an RPi 3 with updated raspbian pixel.

If I configure RPiSurv for 4 RTSP streams, I have to force HDMI to 1280x720p with:
hdmi_ignore_edid=0xa5000080
hdmi_group=2
# explicit 720P
hdmi_mode=85

Otherwise, if I comment these settings out and just run with automatic display detection, which is connected to a 1080P Asus display, the screen blanks after the first of the 4 streams is drawn and the monitor reports "power save mode"

I've tried the same on an older Dell via DVI and I get the same result.

Any ideas?

SvenVD, 2017-07-18:
Probably 4 full HD streams is too much for the RPI. Can you try to downscale the streams?

SvenVD, 2017-07-18:
What I also experienced is you can not show more pixels on the screen than the connected monitor supports

arrfour, 2017-07-19:
I'll try reducing the stream resolution at the camera, but I didn't know if rpisurv might be able to modify the calculations for the screen. I'm using Pi3 as my platform so it didn't seem like a hardware limitation thing. 

Is there an area in the code where I might be able to limit the actual size of each stream? I tried to define the max width and height in the rpisurv config file, but no luck.

drthanwho, 2017-07-19:
I had the same problem. Had 3 cameras at 1080p but after the first loaded it would go blank. Had to lower the resolution for all cameras to below 720p for the streams to work.  Haven't found a work around unfortunately.

si458, 2017-07-31:
i have no issues with 4 1080p Dahua IP cameras, on a Pi Zero W with GPU set at 256MB AND USING WIFI, are you sure ur streams are set correctly on the cameras?

arrfour, 2017-07-31:
@si458 are you using mjpeg or h264 streams on those cameras? I'm using OpenEye cameras set to 1080p/30 on h264 when I get the issue. I'm on RPi3 with GPU set at 512MB

si458, 2017-07-31:
I'm using rtsp from the cameras so I'm guessing h264 stream??? Wouldn't no unless u can tell me how to find it out

drthanwho, 2017-07-31:
I have 3 Geovision cameras. Tried one in 1080p and worked fine but when I try all 3 together moments after booting and trying to load the streams it goes to a black screen. They are on H264 mode.

si458, 2017-07-31:
daft one to try, connect to one camera only, run script, check the memory usage of the GPU (mine was about 32mb, it will show you how much is left at beginning of script then after its connected), it it maybe too high as the memory usage would be x 4 ?

closed 2017-09-14 by SvenVD