SvenVD/rpisurv #28: get_free_gpumem crashing

si458, 2017-08-01:
```
Traceback (most recent call last):
  File "surveillance.py", line 329, in <module>
    check_free_gpumem()
  File "surveillance.py", line 223, in check_free_gpumem
    free_gpumem = get_free_gpumem()
  File "surveillance.py", line 227, in get_free_gpumem
    free_gpumem=str(regex_result.group(1))
AttributeError: 'NoneType' object has no attribute 'group'
```
logger.debug(gpumemresult) give me the following output
```
Relocatable heap version 4 found at 0x14000000
total space allocated is 172M, with 170M relocatable, 2.3M legacy and 0 offline
1 legacy blocks of size 2359296

free list at 0x1e401ba0

invalid space 0x0 in free list at 0x1e3fc920
993 free memory in 2 free block(s)
largest free block is 961 bytes
0x14000000: legacy block 2.3M
0x14240000: free 143M
[ 108] 0x1d0d6e40: used 1.0M (refcount 1 lock count 0, size  1048576, align   32, data 0x1d0d6e60, d3ruAl) 'khdispatch_workspace'
0x1d1d6e80: free 14M
[ 109] 0x1dfeb5a0: used   64 (refcount 1 lock count 0, size        0, align   32, data 0x1dfeb5c0, d3ruAl) 'khdispatch_readahead'
[ 107] 0x1dfeb5e0: used   96 (refcount 1 lock count 0, size       22, align    1, data 0x1dfeb600, d1rual) 'khrn_hw_null_render'
[ 106] 0x1dfeb640: used  160 (refcount 1 lock count 0, size       96, align    4, data 0x1dfeb660, d0rual) 'KHRN_MAP_64_T.storage'
[ 105] 0x1dfeb6e0: used  576 (refcount 1 lock count 0, size      512, align    4, data 0x1dfeb700, d0rual) 'KHRN_MAP_T.storage'
[ 104] 0x1dfeb920: used 1.1K (refcount 1 lock count 0, size     1024, align    4, data 0x1dfeb940, d0rual) 'KHRN_PID_MAP_T.storage'
[ 103] 0x1dfebd60: used  576 (refcount 1 lock count 0, size      512, align    4, data 0x1dfebd80, d0rual) 'KHRN_MAP_T.storage'
[ 102] 0x1dfebfa0: used  576 (refcount 1 lock count 0, size      512, align    4, data 0x1dfebfc0, d0rual) 'KHRN_MAP_T.storage'
[ 101] 0x1dfec1e0: used  576 (refcount 1 lock count 0, size      512, align    4, data 0x1dfec200, d0rual) 'KHRN_MAP_T.storage'
0x1dfec420: free 69K
[ 524] 0x1dffd720: used  960 (refcount 1 lock count 0, size      884, align   32, data 0x1dffd740, d0rual) 'audio_server'

0x1dffd720: corrupt trailer (space 993 != 961)
[  74] 0x1dffdae0: used  960 (refcount 1 lock count 0, size      880, align   32, data 0x1dffdb00, d0rual) 'audio_server'
[ 537] 0x1dffdea0: used  960 (refcount 1 lock count 0, size      884, align   32, data 0x1dffdec0, d0rual) 'audio_server'
0x1dffe260: corrupt entry (space 0x0)
resynced at 0x1dffe240 (skipped -32 bytes)
[1515890087] 0x1dffe240: used  960 (refcount 1515890088 lock count 7743, size 503307904, align    1, data 0x1dffe260, d3rUAL) '0x5a5aa5a5'
[  85] 0x1dffe600: used  960 (refcount 1 lock count 0, size      884, align   32, data 0x1dffe620, d0rual) 'audio_server'
[ 142] 0x1dffe9c0: used  960 (refcount 1 lock count 0, size      884, align   32, data 0x1dffe9e0, d0rual) 'audio_server'
[ 492] 0x1dffed80: used 4.0M (refcount 1 lock count 0, size  4177920, align   32, data 0x1dffeda0, d1rual) 'dispmanx_resource'
0x1e3fadc0: free 448
0x1e3faf80: corrupt entry (space 0x0)
resynced at 0x1e3fb140 (skipped 448 bytes)
[ 505] 0x1e3fb140: used  960 (refcount 1 lock count 0, size      880, align   32, data 0x1e3fb160, d0rual) 'audio_server'
[ 579] 0x1e3fb500: used  960 (refcount 1 lock count 0, size      884, align   32, data 0x1e3fb520, d0rual) 'audio_server'
[  20] 0x1e3fb8c0: used  960 (refcount 1 lock count 0, size      880, align   32, data 0x1e3fb8e0, d0rual) 'audio_server'
0x1e3fbc80: free 384
0x1e3fbe00: corrupt entry (space 0x0)
resynced at 0x1e3fbe00 (skipped 0 bytes)
[ 507] 0x1e3fbe00: used  960 (refcount 1 lock count 0, size      884, align   32, data 0x1e3fbe20, d0rual) 'audio_server'
[  89] 0x1e3fc1c0: used  960 (refcount 1 lock count 0, size      880, align   32, data 0x1e3fc1e0, d0rual) 'audio_server'
0x1e3fc1c0: corrupt trailer (space 0 != 961)
0x1e3fc1c0: corrupt trailer (guards[0] = 0x00000000)
0x1e3fc1c0: corrupt trailer (guards[1] = 0x00000000)
0x1e3fc1c0: corrupt trailer (guards[2] = 0x00000000)
0x1e3fc1c0: corrupt trailer (guards[3] = 0x00000000)
0x1e3fc1c0: corrupt trailer (guards[4] = 0x00000000)
0x1e3fc580: corrupt entry (space 0x0)
resynced at 0x1e3fc400 (skipped -384 bytes)
[ 297] 0x1e3fc400: used  960 (refcount 1 lock count 0, size      880, align   32, data 0x1e3fc420, d0rual) 'audio_server'
[ 522] 0x1e3fc7c0: used  960 (refcount 1 lock count 0, size      880, align   32, data 0x1e3fc7e0, d0rual) 'audio_server'
[ 450] 0x1e3fcb80: used  576 (refcount 1 lock count 0, size      512, align    4, data 0x1e3fcba0, d0rual) 'ILCS VC buffer pool'
[ 348] 0x1e3fcdc0: used  960 (refcount 1 lock count 0, size      884, align   32, data 0x1e3fcde0, d0rual) 'audio_server'
[  98] 0x1e3fd180: used  960 (refcount 1 lock count 0, size      884, align   32, data 0x1e3fd1a0, d0rual) 'audio_server'
[ 513] 0x1e3fd540: used  960 (refcount 1 lock count 0, size      880, align   32, data 0x1e3fd560, d0rual) 'audio_server'
[... 56 lines cut ...]
[1515890087] 0x1e4017e0: used  960 (refcount 1515890088 lock count 7914, size 507502400, align 4096, data 0x1e402000, d2rUAl) '0x5a5aa5a5'
0x1e4017e0: corrupt trailer (space 0 != 961)
0x1e4017e0: corrupt trailer (guards[0] = 0x00000000)
0x1e4017e0: corrupt trailer (guards[1] = 0x00000000)
0x1e4017e0: corrupt trailer (guards[2] = 0x00000000)
0x1e4017e0: corrupt trailer (guards[3] = 0x00000000)
[1515890087] 0x1e401ba0: used  960 (refcount 1515890088 lock count 7914, size 507502400, align 4096, data 0x1e402000, d2rUAl) '0x5a5aa5a5'
0x1e401ba0: corrupt trailer (space 0 != 961)
0x1e401ba0: corrupt trailer (guards[0] = 0x00000000)
0x1e401ba0: corrupt trailer (guards[1] = 0x00000000)
0x1e401ba0: corrupt trailer (guards[2] = 0x00000000)
0x1e401ba0: corrupt trailer (guards[3] = 0x00000000)
0x1e401ba0: corrupt trailer (guards[4] = 0x00000000)
[1515890087] 0x1e401f60: used  960 (refcount 1515890088 lock count 7914, size 507502400, align 4096, data 0x1e402000, d2rUAl) '0x5a5aa5a5'
0x1e401f60: corrupt trailer (space 0 != 961)
0x1e401f60: corrupt trailer (guards[0] = 0x00000000)
0x1e401f60: corrupt trailer (guards[1] = 0x00000000)
0x1e401f60: corrupt trailer (guards[2] = 0x00000000)
0x1e401f60: corrupt trailer (guards[3] = 0x00000000)
0x1e401f60: corrupt trailer (guards[4] = 0x00000000)
0x1e402320: corrupt entry (space 0x0)
could not resync
heap corruption detected
small allocs not requested
```

SvenVD, 2017-08-01:
Hi, what version of raspbian are you running?

What is the output of /bin/fbset -s on the terminal?

si458, 2017-08-02:
latest raspbian out from raspberry pi website
2017-07-05 with all upgrades/dist-upgrades installed

`pi@raspberrypi:~ $  /bin/fbset -s
mode "1920x1080"
    geometry 1920 1080 1920 1080 32
    timings 0 0 0 0 0 0 0
    rgba 8/16,8/8,8/0,8/24
endmode
`

SvenVD, 2017-08-02:
Do you run rpisurv as root?

si458, 2017-08-02:
`sudo python surveillance.py`     works fine every so often but then others i get that error

SvenVD, 2017-08-05:
Maybe corrupt sdcard or hardware? I can't reproduce

si458, 2017-08-07:
ive managed to get around this issue
`sudo vcgencmd cache_flush` 
this will clear the GPU memory cache
if i run this before i run the python script, it then works without any issue?
maybe you should clear thr cache at the start of running the script?

SvenVD, 2017-08-07:
On a second look. I think the issue is you are running out of memory, you above output shows only 1M left

SvenVD, 2017-08-07:
Did you add s, add gpu_mem=512 to your /boot/config.txt? Because you only seem to have like 170M of memory

si458, 2017-08-08:
im running the script on a Pi Zero which only has 512MB ram, so only 256MB is possible for the GPU, but as explained i managed to fix the issue by running the cache_flush every time BEFORE i run the main python survillance.py script

si458, 2017-08-08:
#33 this will allow user to skip the memory usage check which helps with this issue :)

SvenVD, 2017-08-25:
I found and hopefully fixed the root cause of this check failing in commit https://github.com/SvenVD/rpisurv/commit/462f561001376b4e778c439105c8887a94718640.

Parsing crashed because "vcdbg reloc" ditches the M  unit when free mem is lower then 1Megabyte.

Probably same issue: https://github.com/SvenVD/rpisurv/issues/37

si458, 2017-08-31:
sorry for delay, been ill and maybe with work

> 2017/08/31 13:28:40 - l_default - DEBUG - Drawing placeholder with coordinates: 960, 720 and width: 480 height: 360
2017/08/31 13:28:41 - l_default - DEBUG - cam stream name =cam_stream12
2017/08/31 13:28:41 - l_default - DEBUG - start stream cam_stream12
2017/08/31 13:28:41 - l_default - DEBUG - Drawing placeholder with coordinates: 1440, 720 and width: 480 height: 360
2017/08/31 13:28:41 - l_default - INFO - Unique id of this installation is 54446563015ae798a28077730f37ae7218ff66fadfea92da067dcafdb35e39e8
2017/08/31 13:28:41 - l_default - DEBUG - Start_time is 1504182521.17
2017/08/31 13:28:41 - l_default - DEBUG - Current time detected is 1504182521.18 runtime calculated is 0
2017/08/31 13:28:41 - l_default - DEBUG - Free gpu memory value is 7K
2017/08/31 13:28:41 - l_default - DEBUG - Free memory in bytes: 7.0
2017/08/31 13:28:41 - l_default - ERROR - Free gpu mem is 7.0 bytes which is less than 80000000 bytes. Streams might fail to start. Consider assigning more memory to gpu in /boot/config.txt with the gpu_mem option
2017/08/31 13:29:06 - l_default - DEBUG - stats_counter is 1. Only sending every 40
2017/08/31 13:29:06 - l_default - DEBUG - Free gpu memory value is 178M
2017/08/31 13:29:06 - l_default - DEBUG - Free memory in bytes: 186646528.0
2017/08/31 13:29:31 - l_default - DEBUG - stats_counter is 2. Only sending every 40
2017/08/31 13:29:31 - l_default - DEBUG - Free gpu memory value is 178M
2017/08/31 13:29:31 - l_default - DEBUG - Free memory in bytes: 186646528.0
2017/08/31 13:29:56 - l_default - DEBUG - stats_counter is 3. Only sending every 40
2017/08/31 13:29:57 - l_default - DEBUG - Free gpu memory value is 178M
2017/08/31 13:29:57 - l_default - DEBUG - Free memory in bytes: 186646528.0
2017/08/31 13:30:22 - l_default - DEBUG - stats_counter is 4. Only sending every 40
2017/08/31 13:30:22 - l_default - DEBUG - Free gpu memory value is 178M
2017/08/31 13:30:22 - l_default - DEBUG - Free memory in bytes: 186646528.0

first time round the checks shows 7K which seemed abit odd?
but then every other check shows 178M free
which seems correct as i have 'gpu_mem=379' set, 12 camera streams connected, which shows about 16MB per stream

SvenVD, 2017-08-31:
7K is correct, it means 7 kilobytes left. The next line calculating it to bytes was not correct. I fixed this bug in latest release.

si458, 2017-09-04:
are you sure its correct? just done latest git pull and im sure its still wrong?

> 2017/09/04 11:08:20 - l_default - DEBUG - Start_time is 1504519700.83
2017/09/04 11:08:20 - l_default - DEBUG - Current time detected is 1504519700.83 runtime calculated is 0
2017/09/04 11:08:20 - l_default - DEBUG - Free gpu memory value is 3K
2017/09/04 11:08:20 - l_default - DEBUG - Free memory in bytes: 3072.0
2017/09/04 11:08:20 - l_default - ERROR - Free gpu mem is 3072.0 bytes which is less than 80000000 bytes. Streams might fail to start. Consider assigning more memory to gpu in /boot/config.txt with the gpu_mem option
2017/09/04 11:08:46 - l_default - DEBUG - stats_counter is 1. Only sending every 40
2017/09/04 11:08:46 - l_default - DEBUG - Free gpu memory value is 178M
2017/09/04 11:08:46 - l_default - DEBUG - Free memory in bytes: 186646528.0
2017/09/04 11:09:11 - l_default - DEBUG - stats_counter is 2. Only sending every 40
2017/09/04 11:09:11 - l_default - DEBUG - Free gpu memory value is 178M
2017/09/04 11:09:11 - l_default - DEBUG - Free memory in bytes: 186646528.0

SvenVD, 2017-09-07:
What part do you think is incorrect? When starting raspbian reports 3K free and a bit later it is 178M free. rpisurv now correctly parses what raspbian reports. If raspbian reports are wrong I can not change this.

si458, 2017-09-07:
It's ok, it works now and much better than what it use to be like, not sure why when you first run the rpisurv it shows 7k free? Then a few seconds later 176mb but it's fine :)

closed 2017-09-14 by SvenVD