SvenVD/rpisurv #35: systemctl is showing 'python survellance.py' running multiple times

si458, 2017-08-09:
systemctl is showing 'python surviellance.py' running multiple times and crashing the PI at times

https://pastebin.com/KizBLEWF

but dont see this issue when run manually 'sudo python surveillance.py'

SvenVD, 2017-08-09:
You should not run sudo python surveillance.py manually. You are just spawning multiple instances of rpisurv without killing previous instances, which will eventually eat up all resources. If you have systemd, only use systemd to stop start or restart rpisurv.

si458, 2017-08-09:
I see the multiple Python scripts being run at start up by using the systemctl 

But also it i stop the systemctl and run sudo Python manually I still see multiple Python scripts being run?

I'm not mixing both methods of start ups

SvenVD, 2017-08-09:
There are multiple python scripts running at the same time after starting rpisurv, one for each stream (watchdog) and a master process. You should not have more then those.

si458, 2017-08-13:
ah right ok, that makes more sense, the script/monitoring runs alot smoother on my Pi 3 then my Pi Zero
if i do more than 4 cameras on the Pi Zero it struggles but the Pi 3 works without any issue!

closed 2017-08-13 by si458