SvenVD/rpisurv #37: Streams reboot every few minutes

gavzzorz, 2017-08-21:
Hi,

I have an issue where every few minutes the screen will go back to the Raspberry Pi Terminal screen then load back up RPISurv with the connecting image then load back into the streams.
it does this every 5 mins or so.

Video of issue
https://www.dropbox.com/s/k1r9d7un2kz9tkg/2017-08-21%2011.46.44.mp4?dl=0

arrfour, 2017-08-21:
The first thing hung I would check is power. What are you using to power your RPi?

gavzzorz, 2017-08-22:
the RPI is a genuine RPI3 starter kit as linked below that comes with the official PSU...
https://www.amazon.co.uk/Raspberry-Pi-Official-Desktop-Starter/dp/B01CI5879A/ref=sr_1_2?ie=UTF8&qid=1503388074&sr=8-2&keywords=rpi3

Please see the below link to all the logs. i have replaced the RTSP Address in the logs as this is in a school and could be accessed remotely.

https://www.dropbox.com/sh/j8136iohunsu9dy/AAAp513A7X45RuAS8xWy_uW3a?dl=0

Hope that helps.

Thanks

SvenVD, 2017-08-24:
It seems the main process is crashing, after which rpisurv tries to revive itself via systemd (It will do anything it can to stay alive :-). To confirm this theory please post output of journalctl -u rpisurv. The question is now why it crashes, it could be hardware related.

gavzzorz, 2017-08-25:
im thinking its not hardware related as i had the same issue with a RPI Model B+ the other day and then the same again when the RPI 3 Arrived.

I have used this before with no issues what so ever on an earlier version of Raspbian (Pixel) to i'm thinking its something to do with Raspbian Stretch.

please see the below logs for Journalctl...
https://www.dropbox.com/s/kwdn80nb3pdlkk1/JournalCtl_Log.txt?dl=0

Cheers

SvenVD, 2017-08-25:
Hi, check https://github.com/SvenVD/rpisurv/issues/28 I found and hopefully fixed the bug. Can you install the latest version and test again?

Thanks

gavzzorz, 2017-08-29:
Hi SvenVD,

I have updated the Pi this morning and so far so good :).

Thanks for sorting this!

closed 2017-09-14 by SvenVD