SvenVD/rpisurv #39: wont start

syadnom, 2017-09-10:
I just installed (first time, pi3) today and I can't get this to start.

No log directory exists in /usr/local/bin/rpisurv 

I've tried to run the surveillance.py directly but it fails.

error:
```

root@raspberrypi:/usr/local/bin/rpisurv# python surveillance.py
Traceback (most recent call last):
  File "surveillance.py", line 10, in <module>
    from config import cfg
  File "/usr/local/bin/rpisurv/config.py", line 4, in <module>
    cfg = yaml.load(ymlfile)
  File "/usr/lib/python2.7/dist-packages/yaml/__init__.py", line 71, in load
    return loader.get_single_data()
  File "/usr/lib/python2.7/dist-packages/yaml/constructor.py", line 37, in get_single_data
    node = self.get_single_node()
  File "/usr/lib/python2.7/dist-packages/yaml/composer.py", line 36, in get_single_node
    document = self.compose_document()
  File "/usr/lib/python2.7/dist-packages/yaml/composer.py", line 55, in compose_document
    node = self.compose_node(None, None)
  File "/usr/lib/python2.7/dist-packages/yaml/composer.py", line 84, in compose_node
    node = self.compose_mapping_node(anchor)
  File "/usr/lib/python2.7/dist-packages/yaml/composer.py", line 133, in compose_mapping_node
    item_value = self.compose_node(node, item_key)
  File "/usr/lib/python2.7/dist-packages/yaml/composer.py", line 84, in compose_node
    node = self.compose_mapping_node(anchor)
  File "/usr/lib/python2.7/dist-packages/yaml/composer.py", line 133, in compose_mapping_node
    item_value = self.compose_node(node, item_key)
  File "/usr/lib/python2.7/dist-packages/yaml/composer.py", line 82, in compose_node
    node = self.compose_sequence_node(anchor)
  File "/usr/lib/python2.7/dist-packages/yaml/composer.py", line 111, in compose_sequence_node
    node.value.append(self.compose_node(node, index))
  File "/usr/lib/python2.7/dist-packages/yaml/composer.py", line 84, in compose_node
    node = self.compose_mapping_node(anchor)
  File "/usr/lib/python2.7/dist-packages/yaml/composer.py", line 127, in compose_mapping_node
    while not self.check_event(MappingEndEvent):
  File "/usr/lib/python2.7/dist-packages/yaml/parser.py", line 98, in check_event
    self.current_event = self.state()
  File "/usr/lib/python2.7/dist-packages/yaml/parser.py", line 428, in parse_block_mapping_key
    if self.check_token(KeyToken):
  File "/usr/lib/python2.7/dist-packages/yaml/scanner.py", line 116, in check_token
    self.fetch_more_tokens()
  File "/usr/lib/python2.7/dist-packages/yaml/scanner.py", line 257, in fetch_more_tokens
    % ch.encode('utf-8'), self.get_mark())
yaml.scanner.ScannerError: while scanning for the next token
found character '\t' that cannot start any token
  in "conf/surveillance.yml", line 15, column 1
```

> root@raspberrypi:/usr/local/bin/rpisurv# systemctl status rpisurv
> ● rpisurv.service - Rpisurv Raspberry Pi Surveillance
>    Loaded: loaded (/etc/systemd/system/rpisurv.service; enabled; vendor preset: enabled)
>    Active: failed (Result: exit-code) since Sun 2017-09-10 11:59:52 MDT; 5min ago
>   Process: 629 ExecStart=/usr/bin/rpisurv (code=exited, status=1/FAILURE)
>  Main PID: 629 (code=exited, status=1/FAILURE)
> 
> Sep 10 11:59:51 raspberrypi systemd[1]: rpisurv.service: Unit entered failed state.
> Sep 10 11:59:51 raspberrypi systemd[1]: rpisurv.service: Failed with result 'exit-code'.
> Sep 10 11:59:52 raspberrypi systemd[1]: rpisurv.service: Service hold-off time over, scheduling restart.
> Sep 10 11:59:52 raspberrypi systemd[1]: Stopped Rpisurv Raspberry Pi Surveillance.
> Sep 10 11:59:52 raspberrypi systemd[1]: rpisurv.service: Start request repeated too quickly.
> Sep 10 11:59:52 raspberrypi systemd[1]: Failed to start Rpisurv Raspberry Pi Surveillance.
> Sep 10 11:59:52 raspberrypi systemd[1]: rpisurv.service: Unit entered failed state.
> Sep 10 11:59:52 raspberrypi systemd[1]: rpisurv.service: Failed with result 'exit-code'.
> 

I ran the install script to get this far.

only changes to /etc/rpisurv are rtsp:

> essentials:
>     #These are the rtsp streams you want the program to display
>     #The streams will be displayed in the order you present them here
>     camera_streams:
>         #Dlink dcs-5222l example
> #        - rtsp_url: "rtsp://<user>:<password>@<ip or dnsname>:<port>/play1.sdp"
>           #Enable this option if you want the rtsp stream to stream over tcp instead of udp, this may solve a "smearing" effect on some setups. Defaults to false
>           #Note that you need a version of omxplayer older then 14 March 2016 for this option to work
> #          rtsp_over_tcp: true
>         #Foscam-fi9821w example
> #        - rtsp_url: "rtsp://<user>:<password>@<ip or dnsname>:<port>/videoMain"
>         #Dahua IPC-HDW4200S example or IPC-HDW4300S
> #        - rtsp_url: "rtsp://<user>:<password>@<ip or dnsname>:<port>/cam/realmonitor?channel=1&subtype=1"
>          - rtsp_url: "rtsp://nvr:7447/59b1c01b90084a188c8a3951_2"
>          - rtsp_url: "rtsp://nvr:7447/59b1c01b90084a188c8a3952_2"
>          - rtsp_url: "rtsp://nvr:7447/575b2a5b45ce363182a9ec08_2"
>          - rtsp_url: "rtsp://nvr:7447/569d46c645ce0ec31ff6f846_2"

Those rtsp urls work in omxplayer directly.

Thanks

SvenVD, 2017-09-10:
You may not use tabs in a yaml file. Please remove all tabs.

syadnom, 2017-09-10:
removed all tabes, replaced with spaces.  Same result.

complete file:

```
essentials:
    #These are the rtsp streams you want the program to display
    #The streams will be displayed in the order you present them here
    camera_streams:
        #Dlink dcs-5222l example
#        - rtsp_url: "rtsp://<user>:<password>@<ip or dnsname>:<port>/play1.sdp"
          #Enable this option if you want the rtsp stream to stream over tcp instead of udp, this may solve a "smearing" effect on some setups. Defaults to false
          #Note that you need a version of omxplayer older then 14 March 2016 for this option to work
#          rtsp_over_tcp: true
        #Foscam-fi9821w example
#        - rtsp_url: "rtsp://<user>:<password>@<ip or dnsname>:<port>/videoMain"
        #Dahua IPC-HDW4200S example or IPC-HDW4300S
#        - rtsp_url: "rtsp://<user>:<password>@<ip or dnsname>:<port>/cam/realmonitor?channel=1&subtype=1"
         - rtsp_url: "rtsp://nvr:7447/59b1c01b90084a188c8a3951_2"
  - rtsp_url: "rtsp://nvr:7447/59b1c01b90084a188c8a3952_2"
  - rtsp_url: "rtsp://nvr:7447/575b2a5b45ce363182a9ec08_2"
  - rtsp_url: "rtsp://nvr:7447/569d46c645ce0ec31ff6f846_2"

    #How many columns you want the program to use, it will autocalculate the amount of row needed based on the resolution of your screen
    nr_of_columns: "2"

    #Autostretching makes sures all of your pixels will be uses to display streams, so you have a maximum surveillance area.
    #But this will probably destroy the aspect ratio of your last stream.
    #autostretch: False

    #rpisurv detects the connectable streams on startup,
    #if this option is enabled it will keep the first calculated layout for the lifetime of the rpisurv running even if some cameras go down and come up, instead of redrawing the screen
    #When this option is set to True, rpisurv ignores all cameras configured that were not connectable when rpisurv was started
    keep_first_screen_layout: False

#!!!Normal users do not need to edit this section!!!
advanced:
    ##Enable this option if you want to have a fixed width of all your camera streams,
    ##By default rpisurv autocalculates this value, this can cause streams to get "stretched",
    ##if this value exceeds the available width, rpisurv will fallback to autocalculation
    #fixed_width: 500

    ##Enable this option if you want to have a fixed height for all your camera streams,
    ##By default rpisurv autocalculates this value, this can cause streams to get "stretched",
    ##if this value exceeds the available height, rpisurv will fallback to autocalculation
    #fixed_height: 500

    #Rpisurv sends usage stats to it's statistics server to give the rpisurv community an idea how widespread this software is being used
    #No performance impact is measured when using this option, no sensitive data is being sent. All data is anonymised.
    #By default this is true
    #update_stats: True

    #By default rpisurv checks every 25 seconds if it needs to redraw the screen
    #interval_check_status: 25

    #By default rpisurv will check memory usage, set to False to skip memory usage check
    #memory_usage_check: True

#These are fallbacks if autodection fails;
#Normally you do not need to configure these
fallbacks:
    resolution:
        width: "1920"
        height: "1080"
```

SvenVD, 2017-09-10:
First error was very clear:

yaml.scanner.ScannerError: while scanning for the next token
found character '\t' that cannot start any token
  in "conf/surveillance.yml", line 15, column 1

The issue was a tab in the yaml file. Do you have a new error log?

syadnom, 2017-09-10:
no, there is no log folder in /usr/local/bin/rpisurv

SvenVD, 2017-09-10:
Ok I see the issue. You should indent  the following lines like it was done in the example file

- rtsp_url: "rtsp://nvr:7447/59b1c01b90084a188c8a3952_2"
  - rtsp_url: "rtsp://nvr:7447/575b2a5b45ce363182a9ec08_2"
  - rtsp_url: "rtsp://nvr:7447/569d46c645ce0ec31ff6f846_2"

syadnom, 2017-09-10:
I've matched the indentation to the example with spaces.  Still error.

I'm still getting this when I try to run rpisurv directly:
```
    % ch.encode('utf-8'), self.get_mark())
yaml.scanner.ScannerError: while scanning for the next token
found character '\t' that cannot start any token
  in "conf/surveillance.yml", line 15, column 1

```

I haven't altered that file at all...

syadnom, 2017-09-10:
I think I found the issue.  surveillance.yml looks like it needs linked to /etc/rpisurv but it isn't.

syadnom, 2017-09-10:
ok, that was it.  it starting now.  not yet pulling the video stream for some reason but I'll continue checking on that.

SvenVD, 2017-09-10:
Normally the installer symlinks /etc/rpisurv to surveillance.yml

closed 2017-09-14 by SvenVD