SvenVD/rpisurv #41: help installing rpisurv

netstatmen, 2017-09-14:
i followed the steps but cant get it to work, can you help me?

pi@rpisurv:/usr/local/bin/rpisurv/logs $ sudo systemctl status rpisurv
● rpisurv.service - Rpisurv Raspberry Pi Surveillance
   Loaded: loaded (/etc/systemd/system/rpisurv.service; enabled; vendor preset:
   Active: failed (Result: exit-code) since Wed 2017-09-13 23:32:42 MST; 1min 21
  Process: 561 ExecStart=/usr/bin/rpisurv (code=exited, status=1/FAILURE)
 Main PID: 561 (code=exited, status=1/FAILURE)

Sep 13 23:32:42 rpisurv systemd[1]: rpisurv.service: Failed with result 'exit-co
Sep 13 23:32:42 rpisurv systemd[1]: rpisurv.service: Service hold-off time over,
Sep 13 23:32:42 rpisurv systemd[1]: Stopped Rpisurv Raspberry Pi Surveillance.
Sep 13 23:32:42 rpisurv systemd[1]: rpisurv.service: Start request repeated too
Sep 13 23:32:42 rpisurv systemd[1]: Failed to start Rpisurv Raspberry Pi Surveil
Sep 13 23:32:42 rpisurv systemd[1]: rpisurv.service: Unit entered failed state.
Sep 13 23:32:42 rpisurv systemd[1]: rpisurv.service: Failed with result 'exit-co
lines 1-13/13 (END)
● rpisurv.service - Rpisurv Raspberry Pi Surveillance
   Loaded: loaded (/etc/systemd/system/rpisurv.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Wed 2017-09-13 23:32:42 MST; 1min 21s ago
  Process: 561 ExecStart=/usr/bin/rpisurv (code=exited, status=1/FAILURE)
 Main PID: 561 (code=exited, status=1/FAILURE)

Sep 13 23:32:42 rpisurv systemd[1]: rpisurv.service: Failed with result 'exit-code'.
Sep 13 23:32:42 rpisurv systemd[1]: rpisurv.service: Service hold-off time over, scheduling restart.
Sep 13 23:32:42 rpisurv systemd[1]: Stopped Rpisurv Raspberry Pi Surveillance.
Sep 13 23:32:42 rpisurv systemd[1]: rpisurv.service: Start request repeated too quickly.
Sep 13 23:32:42 rpisurv systemd[1]: Failed to start Rpisurv Raspberry Pi Surveillance.
Sep 13 23:32:42 rpisurv systemd[1]: rpisurv.service: Unit entered failed state.
Sep 13 23:32:42 rpisurv systemd[1]: rpisurv.service: Failed with result 'exit-code'.
~
~
~
~
~
~
~
~
~
~
lines 1-13/13 (END)
● rpisurv.service - Rpisurv Raspberry Pi Surveillance
   Loaded: loaded (/etc/systemd/system/rpisurv.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Wed 2017-09-13 23:32:42 MST; 1min 21s ago
  Process: 561 ExecStart=/usr/bin/rpisurv (code=exited, status=1/FAILURE)
 Main PID: 561 (code=exited, status=1/FAILURE)

Sep 13 23:32:42 rpisurv systemd[1]: rpisurv.service: Failed with result 'exit-code'.
Sep 13 23:32:42 rpisurv systemd[1]: rpisurv.service: Service hold-off time over, scheduling restart.
Sep 13 23:32:42 rpisurv systemd[1]: Stopped Rpisurv Raspberry Pi Surveillance.
Sep 13 23:32:42 rpisurv systemd[1]: rpisurv.service: Start request repeated too quickly.
Sep 13 23:32:42 rpisurv systemd[1]: Failed to start Rpisurv Raspberry Pi Surveillance.
Sep 13 23:32:42 rpisurv systemd[1]: rpisurv.service: Unit entered failed state.
Sep 13 23:32:42 rpisurv systemd[1]: rpisurv.service: Failed with result 'exit-code'.
~
~
~
~
~
~
~
~
~

netstatmen, 2017-09-14:
this is the complete main.log

2017/09/13 23:29:28 - l_default - DEBUG - autodetected resolution of['1824', '984']
2017/09/13 23:29:28 - l_default - DEBUG - nr_of_columns = 2
2017/09/13 23:29:28 - l_default - DEBUG - interval_check_status = 25
2017/09/13 23:29:28 - l_default - DEBUG - camera_streams config option exist, using this one
2017/09/13 23:29:30 - l_default - DEBUG - autodetected resolution of['1824', '984']
2017/09/13 23:29:30 - l_default - DEBUG - nr_of_columns = 2
2017/09/13 23:29:30 - l_default - DEBUG - interval_check_status = 25
2017/09/13 23:29:30 - l_default - DEBUG - camera_streams config option exist, using this one
2017/09/13 23:29:33 - l_default - DEBUG - autodetected resolution of['1824', '984']
2017/09/13 23:29:33 - l_default - DEBUG - nr_of_columns = 2
2017/09/13 23:29:33 - l_default - DEBUG - interval_check_status = 25
2017/09/13 23:29:33 - l_default - DEBUG - camera_streams config option exist, using this one
2017/09/13 23:29:35 - l_default - DEBUG - autodetected resolution of['1824', '984']
2017/09/13 23:29:35 - l_default - DEBUG - nr_of_columns = 2
2017/09/13 23:29:35 - l_default - DEBUG - interval_check_status = 25
2017/09/13 23:29:35 - l_default - DEBUG - camera_streams config option exist, using this one
2017/09/13 23:29:37 - l_default - DEBUG - autodetected resolution of['1824', '984']
2017/09/13 23:29:37 - l_default - DEBUG - nr_of_columns = 2
2017/09/13 23:29:37 - l_default - DEBUG - interval_check_status = 25
2017/09/13 23:29:37 - l_default - DEBUG - camera_streams config option exist, using this one
2017/09/13 23:29:39 - l_default - DEBUG - autodetected resolution of['1824', '984']
2017/09/13 23:29:39 - l_default - DEBUG - nr_of_columns = 2
2017/09/13 23:29:39 - l_default - DEBUG - interval_check_status = 25
2017/09/13 23:29:39 - l_default - DEBUG - camera_streams config option exist, using this one
2017/09/13 23:29:41 - l_default - DEBUG - autodetected resolution of['1824', '984']
2017/09/13 23:29:41 - l_default - DEBUG - nr_of_columns = 2
2017/09/13 23:29:41 - l_default - DEBUG - interval_check_status = 25
2017/09/13 23:29:41 - l_default - DEBUG - camera_streams config option exist, using this one
2017/09/13 23:29:52 - l_default - DEBUG - autodetected resolution of['1824', '984']
2017/09/13 23:29:52 - l_default - DEBUG - nr_of_columns = 2
2017/09/13 23:29:52 - l_default - DEBUG - interval_check_status = 25
2017/09/13 23:29:52 - l_default - DEBUG - camera_streams config option exist, using this one
2017/09/13 23:29:54 - l_default - DEBUG - autodetected resolution of['1824', '984']
2017/09/13 23:29:54 - l_default - DEBUG - nr_of_columns = 2
2017/09/13 23:29:54 - l_default - DEBUG - interval_check_status = 25
2017/09/13 23:29:54 - l_default - DEBUG - camera_streams config option exist, using this one
2017/09/13 23:29:56 - l_default - DEBUG - autodetected resolution of['1824', '984']
2017/09/13 23:29:56 - l_default - DEBUG - nr_of_columns = 2
2017/09/13 23:29:56 - l_default - DEBUG - interval_check_status = 25
2017/09/13 23:29:56 - l_default - DEBUG - camera_streams config option exist, using this one
2017/09/13 23:29:58 - l_default - DEBUG - autodetected resolution of['1824', '984']
2017/09/13 23:29:58 - l_default - DEBUG - nr_of_columns = 2
2017/09/13 23:29:58 - l_default - DEBUG - interval_check_status = 25
2017/09/13 23:29:58 - l_default - DEBUG - camera_streams config option exist, using this one
2017/09/13 23:30:00 - l_default - DEBUG - autodetected resolution of['1824', '984']
2017/09/13 23:30:00 - l_default - DEBUG - nr_of_columns = 2
2017/09/13 23:30:00 - l_default - DEBUG - interval_check_status = 25
2017/09/13 23:30:00 - l_default - DEBUG - camera_streams config option exist, using this one
2017/09/13 23:30:02 - l_default - DEBUG - autodetected resolution of['1824', '984']
2017/09/13 23:30:02 - l_default - DEBUG - nr_of_columns = 2
2017/09/13 23:30:02 - l_default - DEBUG - interval_check_status = 25
2017/09/13 23:30:02 - l_default - DEBUG - camera_streams config option exist, using this one
2017/09/13 23:30:04 - l_default - DEBUG - autodetected resolution of['1824', '984']
2017/09/13 23:30:04 - l_default - DEBUG - nr_of_columns = 2
2017/09/13 23:30:04 - l_default - DEBUG - interval_check_status = 25
2017/09/13 23:30:04 - l_default - DEBUG - camera_streams config option exist, using this one
2017/09/13 23:30:06 - l_default - DEBUG - autodetected resolution of['1824', '984']
2017/09/13 23:30:06 - l_default - DEBUG - nr_of_columns = 2
2017/09/13 23:30:06 - l_default - DEBUG - interval_check_status = 25
2017/09/13 23:30:06 - l_default - DEBUG - camera_streams config option exist, using this one
2017/09/13 23:30:08 - l_default - DEBUG - autodetected resolution of['1824', '984']
2017/09/13 23:30:08 - l_default - DEBUG - nr_of_columns = 2
2017/09/13 23:30:08 - l_default - DEBUG - interval_check_status = 25
2017/09/13 23:30:08 - l_default - DEBUG - camera_streams config option exist, using this one
2017/09/13 23:30:10 - l_default - DEBUG - autodetected resolution of['1824', '984']
2017/09/13 23:30:10 - l_default - DEBUG - nr_of_columns = 2
2017/09/13 23:30:10 - l_default - DEBUG - interval_check_status = 25
2017/09/13 23:30:10 - l_default - DEBUG - camera_streams config option exist, using this one
[... 59 lines cut ...]
2017/09/13 23:32:40 - l_default - DEBUG - camera_streams config option exist, using this one
2017/09/13 23:32:42 - l_default - DEBUG - autodetected resolution of['1824', '984']
2017/09/13 23:32:42 - l_default - DEBUG - nr_of_columns = 2
[... cut ...]
2017/09/13 23:48:04 - l_default - DEBUG - nr_of_columns = 2
2017/09/13 23:48:04 - l_default - DEBUG - interval_check_status = 25
2017/09/13 23:48:04 - l_default - DEBUG - camera_streams config option exist, using this one
2017/09/13 23:48:06 - l_default - DEBUG - autodetected resolution of['1824', '984']
2017/09/13 23:48:06 - l_default - DEBUG - nr_of_columns = 2
2017/09/13 23:48:06 - l_default - DEBUG - interval_check_status = 25
2017/09/13 23:48:06 - l_default - DEBUG - camera_streams config option exist, using this one

SvenVD, 2017-09-14:
rpisurv seems to be crashing:
Please output:

lsb_release -a
journalctl -u rpisurv
raspberry pi model

netstatmen, 2017-09-14:
raspberry pi 2 B
**pi@rpisurv:~ $ lsb_release -a**
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 9.1 (stretch)
Release:        9.1
Codename:       stretch
pi@rpisurv:~ $

_____________________________________________________________________________________________________________

**pi@rpisurv:~ $ journalctl -u rpisurv**
-- Logs begin at Thu 2016-11-03 10:16:43 MST, end at Thu 2017-09-14 13:26:03 MST
Sep 14 13:24:04 rpisurv systemd[1]: Started Rpisurv Raspberry Pi Surveillance.
Sep 14 13:24:06 rpisurv rpisurv[438]: Traceback (most recent call last):
Sep 14 13:24:06 rpisurv rpisurv[438]:   File "surveillance.py", line 10, in <mod
Sep 14 13:24:06 rpisurv rpisurv[438]:     from config import cfg
Sep 14 13:24:06 rpisurv rpisurv[438]:   File "/usr/local/bin/rpisurv/config.py",
Sep 14 13:24:06 rpisurv rpisurv[438]:     cfg = yaml.load(ymlfile)
Sep 14 13:24:06 rpisurv rpisurv[438]:   File "/usr/lib/python2.7/dist-packages/y
Sep 14 13:24:06 rpisurv rpisurv[438]:     return loader.get_single_data()
Sep 14 13:24:06 rpisurv rpisurv[438]:   File "/usr/lib/python2.7/dist-packages/y
Sep 14 13:24:06 rpisurv rpisurv[438]:     node = self.get_single_node()
Sep 14 13:24:06 rpisurv rpisurv[438]:   File "/usr/lib/python2.7/dist-packages/y
Sep 14 13:24:06 rpisurv rpisurv[438]:     document = self.compose_document()
Sep 14 13:24:06 rpisurv rpisurv[438]:   File "/usr/lib/python2.7/dist-packages/y
Sep 14 13:24:06 rpisurv rpisurv[438]:     node = self.compose_node(None, None)
Sep 14 13:24:06 rpisurv rpisurv[438]:   File "/usr/lib/python2.7/dist-packages/y
Sep 14 13:24:06 rpisurv rpisurv[438]:     node = self.compose_mapping_node(ancho
Sep 14 13:24:06 rpisurv rpisurv[438]:   File "/usr/lib/python2.7/dist-packages/y
Sep 14 13:24:06 rpisurv rpisurv[438]:     item_value = self.compose_node(node, i
Sep 14 13:24:06 rpisurv rpisurv[438]:   File "/usr/lib/python2.7/dist-packages/y
Sep 14 13:24:06 rpisurv rpisurv[438]:     node = self.compose_mapping_node(ancho
Sep 14 13:24:06 rpisurv rpisurv[438]:   File "/usr/lib/python2.7/dist-packages/y
Sep 14 13:24:06 rpisurv rpisurv[438]:     while not self.check_event(MappingEndE
lines 1-23...skipping...
-- Logs begin at Thu 2016-11-03 10:16:43 MST, end at Thu 2017-09-14 13:26:03 MST
Sep 14 13:24:04 rpisurv systemd[1]: Started Rpisurv Raspberry Pi Surveillance.
Sep 14 13:24:06 rpisurv rpisurv[438]: Traceback (most recent call last):
Sep 14 13:24:06 rpisurv rpisurv[438]:   File "surveillance.py", line 10, in <mod
Sep 14 13:24:06 rpisurv rpisurv[438]:     from config import cfg
Sep 14 13:24:06 rpisurv rpisurv[438]:   File "/usr/local/bin/rpisurv/config.py",
Sep 14 13:24:06 rpisurv rpisurv[438]:     cfg = yaml.load(ymlfile)
Sep 14 13:24:06 rpisurv rpisurv[438]:   File "/usr/lib/python2.7/dist-packages/y
Sep 14 13:24:06 rpisurv rpisurv[438]:     return loader.get_single_data()
Sep 14 13:24:06 rpisurv rpisurv[438]:   File "/usr/lib/python2.7/dist-packages/y
Sep 14 13:24:06 rpisurv rpisurv[438]:     node = self.get_single_node()
Sep 14 13:24:06 rpisurv rpisurv[438]:   File "/usr/lib/python2.7/dist-packages/y
Sep 14 13:24:06 rpisurv rpisurv[438]:     document = self.compose_document()
Sep 14 13:24:06 rpisurv rpisurv[438]:   File "/usr/lib/python2.7/dist-packages/y
Sep 14 13:24:06 rpisurv rpisurv[438]:     node = self.compose_node(None, None)
Sep 14 13:24:06 rpisurv rpisurv[438]:   File "/usr/lib/python2.7/dist-packages/y
Sep 14 13:24:06 rpisurv rpisurv[438]:     node = self.compose_mapping_node(ancho
Sep 14 13:24:06 rpisurv rpisurv[438]:   File "/usr/lib/python2.7/dist-packages/y
Sep 14 13:24:06 rpisurv rpisurv[438]:     item_value = self.compose_node(node, i
Sep 14 13:24:06 rpisurv rpisurv[438]:   File "/usr/lib/python2.7/dist-packages/y
Sep 14 13:24:06 rpisurv rpisurv[438]:     node = self.compose_mapping_node(ancho
Sep 14 13:24:06 rpisurv rpisurv[438]:   File "/usr/lib/python2.7/dist-packages/y
Sep 14 13:24:06 rpisurv rpisurv[438]:     while not self.check_event(MappingEndE
Sep 14 13:24:06 rpisurv rpisurv[438]:   File "/usr/lib/python2.7/dist-packages/y
Sep 14 13:24:06 rpisurv rpisurv[438]:     self.current_event = self.state()
Sep 14 13:24:06 rpisurv rpisurv[438]:   File "/usr/lib/python2.7/dist-packages/y
Sep 14 13:24:06 rpisurv rpisurv[438]:     "expected <block end>, but found %r" %
Sep 14 13:24:06 rpisurv rpisurv[438]: yaml.parser.ParserError: while parsing a b
Sep 14 13:24:06 rpisurv rpisurv[438]:   in "conf/surveillance.yml", line 4, colu
Sep 14 13:24:06 rpisurv rpisurv[438]: expected <block end>, but found '<block se
Sep 14 13:24:06 rpisurv rpisurv[438]:   in "conf/surveillance.yml", line 16, col
Sep 14 13:24:06 rpisurv systemd[1]: rpisurv.service: Main process exited, code=e
Sep 14 13:24:06 rpisurv systemd[1]: rpisurv.service: Unit entered failed state.
Sep 14 13:24:06 rpisurv systemd[1]: rpisurv.service: Failed with result 'exit-co
[... 82 lines cut ...]
Sep 14 13:24:06 rpisurv rpisurv[438]:     while not self.check_event(MappingEndEvent):
Sep 14 13:24:06 rpisurv rpisurv[438]:   File "/usr/lib/python2.7/dist-packages/yaml/parser.py", line 98, in check_event
Sep 14 13:24:06 rpisurv rpisurv[438]:     self.current_event = self.state()
Sep 14 13:24:06 rpisurv rpisurv[438]:   File "/usr/lib/python2.7/dist-packages/yaml/parser.py", line 439, in parse_block_mapping_key
Sep 14 13:24:06 rpisurv rpisurv[438]:     "expected <block end>, but found %r" % token.id, token.start_mark)
[... cut ...]
Sep 14 13:24:07 rpisurv rpisurv[468]:   File "/usr/local/bin/rpisurv/config.py", line 4, in <module>
Sep 14 13:24:07 rpisurv rpisurv[468]:     cfg = yaml.load(ymlfile)
Sep 14 13:24:07 rpisurv rpisurv[468]:   File "/usr/lib/python2.7/dist-packages/yaml/__init__.py", line 71, in load
Sep 14 13:24:07 rpisurv rpisurv[468]:     return loader.get_single_data()
Sep 14 13:24:07 rpisurv rpisurv[468]:   File "/usr/lib/python2.7/dist-packages/yaml/constructor.py", line 37, in get_single_data
Sep 14 13:24:07 rpisurv rpisurv[468]:     node = self.get_single_node()
lines 1-46

SvenVD, 2017-09-14:
There is something wrong in your config. What is the output of cat /etc/rpisurv?

netstatmen, 2017-09-14:
**pi@rpisurv:~ $ cat /etc/rpisurv**
essentials:
    #These are the rtsp streams you want the program to display
    #The streams will be displayed in the order you present them here
    camera_streams:
        #Dlink dcs-5222l example
       # - rtsp_url: "rtsp://<user>:<password>@<ip or dnsname>:<port>/play1.sdp"
          #Enable this option if you want the rtsp stream to stream over tcp ins                             tead of udp, this may solve a "smearing" effect on some setups. Defaults to fals                             e
          #Note that you need a version of omxplayer older then 14 March 2016 fo                             r this option to work
          rtsp_over_tcp: true
        #Foscam-fi9821w example
       # - rtsp_url: "rtsp://<user>:<password>@<ip or dnsname>:<port>/videoMain"
        #Dahua IPC-HDW4200S example or IPC-HDW4300S
       # - rtsp_url: "rtsp://<user>:<password>@<ip or dnsname>:<port>/cam/realmo                             nitor?channel=1&subtype=1"

        #camera 1
        - rtsp_url: "rtsp://nvr.dyndns.org:7447/886c4771-c9af-3ff2-a6de-30ea0                             1603a9a_1"

        #camera 2
        - rtsp_url: "rtsp://nvr.dyndns.org:7447/bbh44a27-e83f-3ed5-9a0e-6f5e0                             975df1a_1"

        #camera 3
        - rtsp_url: "rtsp://nvr.dyndns.org:7447/94fsea3a-29f0-3823-8c76-84557                             186c8c4_1"

        #camera 4
        - rtsp_url: "rtsp://nvr.dyndns.org:7447/563444d9-0f3d-3643-8467-1cc99                             4aks496_1"

        #camera 5
        - rtsp_url: "rtsp://nvr.dyndns.org:7447/414f4kfe-2a28-3bc3-b3f2-a3se2                             cf5ce06_1"

        #camera 6
        - rtsp_url: "rtsp://nvr.dyndns.org:7447/c71sk4bd-4503-3851-94b6-8c0bb                             fe6147a_1"

        #camera 7
        - rtsp_url: "rtsp://nvr.dyndns.org:7447/8ba49f32-4f1a-36af-bc51-9d22c                             74c34c6_1"

        #camera 8
        - rtsp_url: "rtsp://nvr.dyndns.org:7447/3a254kc-0237-334d-bacb-69796                             d09e4cd_1"

    #How many columns you want the program to use, it will autocalculate the amo                             unt of row needed based on the resolution of your screen
    nr_of_columns: "3"

    #Autostretching makes sures all of your pixels will be uses to display strea                             ms, so you have a maximum surveillance area.
    #But this will probably destroy the aspect ratio of your last stream.
    #autostretch: False

    #rpisurv detects the connectable streams on startup,
    #if this option is enabled it will keep the first calculated layout for the                              lifetime of the rpisurv running even if some cameras go down and come up, instea                             d of redrawing the screen
    #When this option is set to True, rpisurv ignores all cameras configured tha                             t were not connectable when rpisurv was started
    keep_first_screen_layout: False

#!!!Normal users do not need to edit this section!!!
advanced:
    ##Enable this option if you want to have a fixed width of all your camera st                             reams,
    ##By default rpisurv autocalculates this value, this can cause streams to ge                             t "stretched",
    ##if this value exceeds the available width, rpisurv will fallback to autoca                             lculation
    #fixed_width: 500

    ##Enable this option if you want to have a fixed height for all your camera                              streams,
    ##By default rpisurv autocalculates this value, this can cause streams to ge                             t "stretched",
    ##if this value exceeds the available height, rpisurv will fallback to autoc                             alculation
    #fixed_height: 500

    #Rpisurv sends usage stats to it's statistics server to give the rpisurv com                             munity an idea how widespread this software is being used
    #No performance impact is measured when using this option, no sensitive data                              is being sent. All data is anonymised.
    #By default this is true
    #update_stats: True

    #By default rpisurv checks every 25 seconds if it needs to redraw the screen
    #interval_check_status: 25

    #By default rpisurv will check memory usage, set to False to skip memory usa                             ge check
    #memory_usage_check: True

#These are fallbacks if autodection fails;
#Normally you do not need to configure these
fallbacks:
    resolution:
        width: "1920"
        height: "1080"

SvenVD, 2017-09-14:
Make sure your indentation is correct. Refer to example file

netstatmen, 2017-09-14:
thank you for your advice, it is working now, but the screen has a lot of blinking rows, how could i fix this?

netstatmen, 2017-09-14:
I just noticed that if I stream 8 cameras it works flawlessly, but when I add the 9th camera, all the 9 camera feeds start to blink and with a lot of pixel rows across the screen

netstatmen, 2017-09-14:
I resolved the blinkering by lowering the resolution to the lowest. it is now working with 9 cameras. Thank you

closed 2017-09-14 by netstatmen