SvenVD/rpisurv #44: Delays in display

skowron0, 2017-11-07:
Hi,

I usage five china camera 2Mpx. In day mode all works wery well.
After change to night mode within an hour the image is delayed by a few minutes.
Do you have a solution?
I use substream 640x480px 15fps.

Wersion of my cameras:
[image]

best regards
Krzysztof

SvenVD, 2017-11-07:
This is probably an issue with your cameras? Switching to nightvision should make no difference for rpisurv.

skowron0, 2017-11-08:
The problem is strange because when I use this stream link on vlc or ONVIF Device Manager this problem does not exist. This is only visible when I use omxplayer.

SvenVD, 2017-11-08:
I am just guessing since I can't help you with specific camera issues. But maybe the delay just builds up over time? Try running the stream with vlc for a longer time and see if delay cumulates. 

What happens if you restart rpisurv? Is the delay gone? You could schedule a workaround to restart rpisurv every hour or so.

skowron0, 2017-11-09:
Yes, the delay just builds up over time - only when use omxplayer. When I use vlc there are no delays even after a few hours.
When I restart rpisurv delay is gone.
Within 30 minutes the delay can be up to 8 minutes.
When I disconnect the camera from the network, the video is displayed as if delayed, as if the omxplayer had accumulated the stream.

SvenVD, 2017-11-10:
You should ask the omxplayer guys.

Maybe you can try setting  --video_fifo 1 --video_queue 1 in worker.py als one of the arguments of omxplayer

skowron0, 2017-11-28:
All the time I have this problem, unfortunately your suggest did not work.
When I have statistics turned on, the parameter marked in red is less than 1 second. In night mode this parameter is constantly increasing. Could this be causing the instability of fps?
I see that you have a lot of experience with omxplayer and maybe you can suggest something else.
I also described the problem on the omxplayer page but no one writes ...
[image]

afburgos, 2017-11-28:
This happens to me as well, the cameras are 15 fps and happens by day and
by night.  The only solution I found is reestart the pi everyday.

<https://user-images.githubusercontent.com/7578684/33303511-d143a3c2-d403-11e7-9da8-650eb7b9d0b5.png>

<https://github.com/SvenVD/rpisurv/issues/44#issuecomment-347416641>, or mute
the thread

SvenVD, 2017-11-29:
Are these wireless cameras?

afburgos, 2017-11-29:
Not mine. All them are wired.

skowron0, 2017-11-30:
I also have everything connected by cable.

skowron0, 2017-12-04:
I found problem with gpu_mem in my log.
In config.txt i setted gpu_mem_1024=548.
Sometimes the value of gpu memory drops to zero, then returns several hundred MB.

my log:

2017/12/04 08:27:21 - l_default - DEBUG - stats_counter is 21. Only sending every 40
2017/12/04 08:27:21 - l_default - DEBUG - Free gpu memory value is 405M
2017/12/04 08:27:21 - l_default - DEBUG - Free memory in bytes: 424673280.0
2017/12/04 08:27:21 - l_default - DEBUG - 192.168.7.55 : 1051 Connectable
2017/12/04 08:27:21 - l_default - DEBUG - 192.168.7.59 : 1091 Connectable
2017/12/04 08:27:21 - l_default - DEBUG - 192.168.7.58 : 1081 Connectable
2017/12/04 08:27:21 - l_default - DEBUG - 192.168.7.57 : 1071 Connectable
2017/12/04 08:27:21 - l_default - DEBUG - 192.168.7.52 : 1021 Connectable
2017/12/04 08:27:46 - l_default - DEBUG - stats_counter is 22. Only sending every 40
2017/12/04 08:27:46 - l_default - DEBUG - Free gpu memory value is 405M
2017/12/04 08:27:46 - l_default - DEBUG - Free memory in bytes: 424673280.0
2017/12/04 08:27:46 - l_default - DEBUG - 192.168.7.55 : 1051 Connectable
2017/12/04 08:27:46 - l_default - DEBUG - 192.168.7.59 : 1091 Connectable
2017/12/04 08:27:46 - l_default - DEBUG - 192.168.7.58 : 1081 Connectable
2017/12/04 08:27:47 - l_default - DEBUG - 192.168.7.57 : 1071 Connectable
2017/12/04 08:27:47 - l_default - DEBUG - 192.168.7.52 : 1021 Connectable
2017/12/04 08:28:12 - l_default - DEBUG - stats_counter is 23. Only sending every 40
2017/12/04 08:28:12 - l_default - DEBUG - Free gpu memory value is 405M
2017/12/04 08:28:12 - l_default - DEBUG - Free memory in bytes: 424673280.0
2017/12/04 08:28:12 - l_default - DEBUG - 192.168.7.55 : 1051 Connectable
2017/12/04 08:28:12 - l_default - DEBUG - 192.168.7.59 : 1091 Connectable
2017/12/04 08:28:12 - l_default - DEBUG - 192.168.7.58 : 1081 Connectable
2017/12/04 08:28:12 - l_default - DEBUG - 192.168.7.57 : 1071 Connectable
2017/12/04 08:28:12 - l_default - DEBUG - 192.168.7.52 : 1021 Connectable
2017/12/04 08:28:37 - l_default - DEBUG - stats_counter is 24. Only sending every 40
2017/12/04 08:28:37 - l_default - DEBUG - Free gpu memory value is 405M
2017/12/04 08:28:37 - l_default - DEBUG - Free memory in bytes: 424673280.0
2017/12/04 08:28:37 - l_default - DEBUG - 192.168.7.55 : 1051 Connectable
2017/12/04 08:28:37 - l_default - DEBUG - 192.168.7.59 : 1091 Connectable
2017/12/04 08:28:37 - l_default - DEBUG - 192.168.7.58 : 1081 Connectable
2017/12/04 08:28:37 - l_default - DEBUG - 192.168.7.57 : 1071 Connectable
2017/12/04 08:28:37 - l_default - DEBUG - 192.168.7.52 : 1021 Connectable
2017/12/04 08:29:02 - l_default - DEBUG - stats_counter is 25. Only sending every 40
2017/12/04 08:29:02 - l_default - DEBUG - Free gpu memory value is 405M
2017/12/04 08:29:02 - l_default - DEBUG - Free memory in bytes: 424673280.0
2017/12/04 08:29:02 - l_default - DEBUG - 192.168.7.55 : 1051 Connectable
2017/12/04 08:29:02 - l_default - DEBUG - 192.168.7.59 : 1091 Connectable
2017/12/04 08:29:02 - l_default - DEBUG - 192.168.7.58 : 1081 Connectable
2017/12/04 08:29:02 - l_default - DEBUG - 192.168.7.57 : 1071 Connectable
2017/12/04 08:29:02 - l_default - DEBUG - 192.168.7.52 : 1021 Connectable
2017/12/04 08:29:27 - l_default - DEBUG - stats_counter is 26. Only sending every 40
2017/12/04 08:29:27 - l_default - DEBUG - Free gpu memory value is 405M
2017/12/04 08:29:27 - l_default - DEBUG - Free memory in bytes: 424673280.0
2017/12/04 08:29:27 - l_default - DEBUG - 192.168.7.55 : 1051 Connectable
2017/12/04 08:29:27 - l_default - DEBUG - 192.168.7.59 : 1091 Connectable
2017/12/04 08:29:27 - l_default - DEBUG - 192.168.7.58 : 1081 Connectable
2017/12/04 08:29:27 - l_default - DEBUG - 192.168.7.57 : 1071 Connectable
2017/12/04 08:29:27 - l_default - DEBUG - 192.168.7.52 : 1021 Connectable
2017/12/04 08:29:52 - l_default - DEBUG - stats_counter is 27. Only sending every 40
2017/12/04 08:29:52 - l_default - DEBUG - Free gpu memory value is 405M
2017/12/04 08:29:52 - l_default - DEBUG - Free memory in bytes: 424673280.0
2017/12/04 08:29:52 - l_default - DEBUG - 192.168.7.55 : 1051 Connectable
2017/12/04 08:29:53 - l_default - DEBUG - 192.168.7.59 : 1091 Connectable
2017/12/04 08:29:53 - l_default - DEBUG - 192.168.7.58 : 1081 Connectable
2017/12/04 08:29:53 - l_default - DEBUG - 192.168.7.57 : 1071 Connectable
2017/12/04 08:29:53 - l_default - DEBUG - 192.168.7.52 : 1021 Connectable
2017/12/04 08:30:18 - l_default - DEBUG - stats_counter is 28. Only sending every 40
2017/12/04 08:30:18 - l_default - DEBUG - Free gpu memory value is 405M
2017/12/04 08:30:18 - l_default - DEBUG - Free memory in bytes: 424673280.0
2017/12/04 08:30:18 - l_default - DEBUG - 192.168.7.55 : 1051 Connectable
2017/12/04 08:30:18 - l_default - DEBUG - 192.168.7.59 : 1091 Connectable
2017/12/04 08:30:18 - l_default - DEBUG - 192.168.7.58 : 1081 Connectable
2017/12/04 08:30:18 - l_default - DEBUG - 192.168.7.57 : 1071 Connectable
2017/12/04 08:30:18 - l_default - DEBUG - 192.168.7.52 : 1021 Connectable
[... 75 lines cut ...]
2017/12/04 08:34:30 - l_default - DEBUG - stats_counter is 38. Only sending every 40
2017/12/04 08:34:30 - l_default - DEBUG - Free gpu memory value is 405M
2017/12/04 08:34:30 - l_default - DEBUG - Free memory in bytes: 424673280.0
2017/12/04 08:34:30 - l_default - DEBUG - 192.168.7.55 : 1051 Connectable
2017/12/04 08:34:30 - l_default - DEBUG - 192.168.7.59 : 1091 Connectable
2017/12/04 08:34:30 - l_default - DEBUG - 192.168.7.58 : 1081 Connectable
[... cut ...]
2017/12/04 08:35:20 - l_default - DEBUG - Free gpu memory value is 405M
2017/12/04 08:35:20 - l_default - DEBUG - Free memory in bytes: 424673280.0
2017/12/04 08:35:20 - l_default - DEBUG - 192.168.7.55 : 1051 Connectable
2017/12/04 08:35:20 - l_default - DEBUG - 192.168.7.59 : 1091 Connectable
2017/12/04 08:35:20 - l_default - DEBUG - 192.168.7.58 : 1081 Connectable
2017/12/04 08:35:20 - l_default - DEBUG - 192.168.7.57 : 1071 Connectable
2017/12/04 08:35:20 - l_default - DEBUG - 192.168.7.52 : 1021 Connectable

SvenVD, 2017-12-04:
Try running with one camera feed, is this also giving a delay?

skowron0, 2017-12-07:
Log when I connect one cam:
2017/12/07 07:36:20 - l_default - DEBUG - 192.168.7.58 : 1081 Connectable
2017/12/07 07:36:46 - l_default - DEBUG - stats_counter is 19. Only sending every 40
2017/12/07 07:36:46 - l_default - DEBUG - Free gpu memory value is 0
2017/12/07 07:36:46 - l_default - DEBUG - Free memory in bytes: 0.0
2017/12/07 07:36:46 - l_default - ERROR - Free gpu mem is 0.0 bytes which is less than 80000000 bytes. Streams might fail to start. Consider assigning more memory to gpu in /boot/config.txt with the gpu_mem option
2017/12/07 07:36:46 - l_default - DEBUG - 192.168.7.58 : 1081 Connectable
2017/12/07 07:37:11 - l_default - DEBUG - stats_counter is 20. Only sending every 40
2017/12/07 07:37:11 - l_default - DEBUG - Free gpu memory value is 493M
2017/12/07 07:37:11 - l_default - DEBUG - Free memory in bytes: 516947968.0
2017/12/07 07:37:11 - l_default - DEBUG - 192.168.7.58 : 1081 Connectable
2017/12/07 07:37:36 - l_default - DEBUG - stats_counter is 21. Only sending every 40
2017/12/07 07:37:36 - l_default - DEBUG - Free gpu memory value is 288
2017/12/07 07:37:36 - l_default - DEBUG - Free memory in bytes: 288.0
2017/12/07 07:37:36 - l_default - ERROR - Free gpu mem is 288.0 bytes which is less than 80000000 bytes. Streams might fail to start. Consider assigning more memory to gpu in /boot/config.txt with the gpu_mem option
2017/12/07 07:37:36 - l_default - DEBUG - 192.168.7.58 : 1081 Connectable
2017/12/07 07:38:01 - l_default - DEBUG - stats_counter is 22. Only sending every 40
2017/12/07 07:38:01 - l_default - DEBUG - Free gpu memory value is 493M
2017/12/07 07:38:01 - l_default - DEBUG - Free memory in bytes: 516947968.0
2017/12/07 07:38:01 - l_default - DEBUG - 192.168.7.58 : 1081 Connectable
2017/12/07 07:38:26 - l_default - DEBUG - stats_counter is 23. Only sending every 40
2017/12/07 07:38:26 - l_default - DEBUG - Free gpu memory value is 9K
2017/12/07 07:38:26 - l_default - DEBUG - Free memory in bytes: 9216.0
2017/12/07 07:38:26 - l_default - ERROR - Free gpu mem is 9216.0 bytes which is less than 80000000 bytes. Streams might fail to start. Consider assigning more memory to gpu in /boot/config.txt with the gpu_mem option
2017/12/07 07:38:26 - l_default - DEBUG - 192.168.7.58 : 1081 Connectable
2017/12/07 07:38:51 - l_default - DEBUG - stats_counter is 24. Only sending every 40
2017/12/07 07:38:51 - l_default - DEBUG - Free gpu memory value is 493M
2017/12/07 07:38:51 - l_default - DEBUG - Free memory in bytes: 516947968.0
2017/12/07 07:38:51 - l_default - DEBUG - 192.168.7.58 : 1081 Connectable

skowron0, 2017-12-07:
There is also a delay on one camera.

skowron0, 2018-01-15:
Hi,

When I set: memory_usage_check: False, the problems with the delay have gone away. I'm testing the second day and everything is ok.

SvenVD, 2018-01-17:
Thanks for your debugging can anybody else who have the same delays confirm the solution by setting memory_usage_check False?

Thanks

closed 2018-02-28 by SvenVD

dhsemp, 2018-11-15:
It didn’t work for me.  I’m using UniFi Protect (UCK-G2+) into a fully updated Rpi3 running rpisurv.  After 12-24hrs, the video will be delayed by minutes.  UBNT says it’s a problem with omxplayer even though it works fine with their legacy NVR running UniFi Video 3.

Any idea what additional omxplayer or rpisurv debug settings I can enable to figure out exactly what’s causing the delay issues?  Could it really be buffering on the Rpi3?

syadnom, 2018-11-15:
@dhsemp protect still has some serious rtsp bugs.  Try playing a stream via vlc and you'll see the same behavior.

syadnom, 2018-11-15:
@dhsemp  omxplayer does have some peculiarities, but that's not the problem here.  Protect is doing some weird things in it's RTSP implementation and ubnt knows it.  

As of right now, you cannot get a reliable RTSP stream out of protect using any known software on any platform.

I have 8 protect/ck2+ systems and they all have the exact some very predictable issues with RTSP streams.

SvenVD, 2018-11-30:
Would it just be the worker.py file I'd have to modify?--> yes. Please report back if that fixed your issue, and if ubiquity fixed their problem.