SvenVD/rpisurv #46: [REQ] Ability to switch between fullscreen views

kronflux, 2017-11-28:
Is there a possibility to add the option to "dial" into a specific stream fullscreen?

In other words, I want to have, for example, 4 camera feeds on my screen. If I hit "1", "Enter" - the first camera feed would become the only one on screen, and take up the entirety of the screen.
To get back to the quad view, I would dial "0", "Enter"

This would be a live saver, as this doesn't seem to be available in any other "basic" software of this sort.
Everything I've looked at so far is very feature bloated, and wouldn't work on a small system like the Pi or similar dev boards.

Thanks in advance!

SvenVD, 2017-11-29:
Thanks for the request. Feel free to make a pull request for this. Currently this is not on the radar, but I will register it for the future.

closed 2017-12-04 by SvenVD

SvenVD, 2018-08-12:
This has now been implemented, currently still beta, check https://github.com/SvenVD/rpisurv/blob/v2.0_branch/RELEASE_NOTES.md