SvenVD/rpisurv #47: pi3 and unifi-video losing streams

syadnom, 2017-12-04:
I'm having an issue where a pi 3 w/ 6 videos pulling rtsp from a unifi-video server (low quality ie '2' streams) is occasionally dropping a display off.

I'm rebooting the unit at 4am every day to keep this to a minimum, but it's still happening about once a week and is down until the reboot.  A reboot always cures it.

Any suggestions to resolve this?

Thanks.

SvenVD, 2017-12-04:
What do you mean by dropping a display off? Please check the logs for any useful hints why this might happen.

THanks

syadnom, 2017-12-04:
Says connecting to camera feed..., never connects.  Can connect w/ VLC to feed and a reboot gets it to connect.

SvenVD, 2017-12-04:
What do the logs say?

syadnom, 2017-12-04:
I'll post after next error.

SvenVD, 2018-01-17:
I assume no error occured again?