SvenVD/rpisurv #48: Streams work in VLC.  "Connecting..." in Rpisurv

dvdrewinder, 2018-01-15:
Hello,

All 3 of my streams work correctly in VLC, but Rpisurv shows them as connecting...

Status shows: 

Jan 15 20:00:28 raspberrypi rpisurv[1369]: 2018/01/15 20:00:28 - worker - DEBUG - Starting stream cam_stream3 with commandline ['/usr/bin/omxplayer', '--live', '--timeout', '60', '--aidx', '
Jan 15 20:00:28 raspberrypi rpisurv[1369]: have a nice day ;)
Jan 15 20:00:28 raspberrypi rpisurv[1369]: have a nice day ;)
Jan 15 20:00:29 raspberrypi rpisurv[1369]: have a nice day ;)
Jan 15 20:00:33 raspberrypi rpisurv[1369]: 2018/01/15 20:00:33 - worker - DEBUG - Starting stream cam_stream1 with commandline ['/usr/bin/omxplayer', '--live', '--timeout', '60', '--aidx', '
Jan 15 20:00:33 raspberrypi rpisurv[1369]: 2018/01/15 20:00:33 - worker - DEBUG - Starting stream cam_stream2 with commandline ['/usr/bin/omxplayer', '--live', '--timeout', '60', '--aidx', '
Jan 15 20:00:33 raspberrypi rpisurv[1369]: have a nice day ;)
Jan 15 20:00:33 raspberrypi rpisurv[1369]: 2018/01/15 20:00:33 - worker - DEBUG - Starting stream cam_stream3 with commandline ['/usr/bin/omxplayer', '--live', '--timeout', '60', '--aidx', '
Jan 15 20:00:33 raspberrypi rpisurv[1369]: have a nice day ;)
Jan 15 20:00:34 raspberrypi rpisurv[1369]: have a nice day ;)

SvenVD, 2018-01-17:
omxplayer is crashing on your urls/streams, try on commandline, also the log should show more on the commandline like what the url was trying to access.

dvdrewinder, 2018-01-19:
I think the issue was related to special characters in the password for the credentials.  After changing the passwords, all streams are working correctly.

closed 2018-01-19 by dvdrewinder