SvenVD/rpisurv #49: Looking for way to add mix of h264 and mjpeg cameras

dannyzumwalt, 2018-01-17:
Using omxplayer on raspbian, i can get mjpeg streams to play from command line with a http cgi address: 

http://192.168.1.45:2080/videostream.cgi

Is there a way to add this to /etc/ripsurv config to add it to my existing (currently 3) h264 streams? When i ender rtsp_url: "http://192.168.1.45:2080/videostream.cgi" rpisurv starts, but the mjpeg stream never starts. 

2018/01/16 20:10:44 - l_default - ERROR - 192.168.1.45 : 2080 Not Connectable (failed socket connect)

Obviously it's trying to use RTSP protocol instead of http. How to override?

SvenVD, 2018-01-17:
This is currently not supported. I am open to a  pull request for this.

dannyzumwalt, 2018-01-17:
Not familiar with git too much or pull requests.

closed 2018-02-28 by SvenVD

SvenVD, 2018-08-12:
This has now been implemented, currently still beta, check https://github.com/SvenVD/rpisurv/blob/v2.0_branch/RELEASE_NOTES.md