SvenVD/rpisurv #54: Crashing

musicman251, 2018-03-24:
I'm unable to get streams to work. service crashing...

sudo systemctl status rpisurv
● rpisurv.service - Rpisurv Raspberry Pi Surveillance
   Loaded: loaded (/etc/systemd/system/rpisurv.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Sat 2018-03-24 16:01:11 UTC; 201ms ago
  Process: 1484 ExecStart=/usr/bin/rpisurv (code=exited, status=1/FAILURE)
 Main PID: 1484 (code=exited, status=1/FAILURE)

Mar 24 16:01:11 PaulMarinePi1 systemd[1]: rpisurv.service: Failed with result 'exit-code'.
Mar 24 16:01:12 PaulMarinePi1 systemd[1]: rpisurv.service: Service hold-off time over, scheduling restart.
Mar 24 16:01:12 PaulMarinePi1 systemd[1]: Stopped Rpisurv Raspberry Pi Surveillance.
Mar 24 16:01:12 PaulMarinePi1 systemd[1]: Started Rpisurv Raspberry Pi Surveillance.

SvenVD, 2018-03-24:
There is an error in the config file for one of your streams, it seems the port was wrongly specified? Could paste you config here.

musicman251, 2018-03-24:
Here is the config I'm using...

 camera_streams:
        #Dlink dcs-5222l example
        - rtsp_url: "rtsp://admin:Taylor19&^@192.168.2.41:554/play1.sdp"

The URLworks using OMXplayer and VLC.

SvenVD, 2018-03-24:
Hi I tested it and  there is nothing wrong with that line, but still there is a problem in the config file. Can you share the whole config file. Did you do your indents correct and did you use spaces instead of tabs etc?

musicman251, 2018-03-24:
essentials:
    #These are the rtsp streams you want the program to display
    #The streams will be displayed in the order you present them here
    camera_streams:
        #Dlink dcs-5222l example
        - rtsp_url: "rtsp://admin:admin@192.168.2.41:554/play1.sdp"
          #Enable this option if you want the rtsp stream to stream over tcp in$
          #Note that you need a version of omxplayer older then 14 March 2016 f$
          rtsp_over_tcp: true
        #Foscam-fi9821w example
        - rtsp_url: "rtsp://<user>:<password>@<ip or dnsname>:<port>/videoMain"
        #Dahua IPC-HDW4200S example or IPC-HDW4300S
        - rtsp_url: "rtsp://<user>:<password>@<ip or dnsname>:<port>/cam/realmo$

    #How many columns you want the program to use, it will autocalculate the am$
    nr_of_columns: "2"

    #Autostretching makes sures all of your pixels will be uses to display stre$
    #But this will probably destroy the aspect ratio of your last stream.
    #autostretch: False

    #rpisurv detects the connectable streams on startup,
    #if this option is enabled it will keep the first calculated layout for the lifetime o$
    #When this option is set to True, rpisurv ignores all cameras configured that were not$
    keep_first_screen_layout: False

#!!!Normal users do not need to edit this section!!!
advanced:
    ##Enable this option if you want to have a fixed width of all your camera streams,
    ##By default rpisurv autocalculates this value, this can cause streams to get "stretch$
    ##if this value exceeds the available width, rpisurv will fallback to autocalculation
    #fixed_width: 500

    ##Enable this option if you want to have a fixed height for all your camera streams,
    ##By default rpisurv autocalculates this value, this can cause streams to get "stretch$
    ##if this value exceeds the available height, rpisurv will fallback to autocalculation
    #fixed_height: 500

    #Rpisurv sends usage stats to it's statistics server to give the rpisurv community an $
    #No performance impact is measured when using this option, no sensitive data is being $
    #By default this is false
    #update_stats: False

    #By default rpisurv checks every 25 seconds if it needs to redraw the screen
    #interval_check_status: 25

    #By default rpisurv will check memory usage, set to False to skip memory usage check
    #memory_usage_check: True

#These are fallbacks if autodection fails;
#Normally you do not need to configure these
fallbacks:
    resolution:
        width: "1920"
        height: "1080"

SvenVD, 2018-03-25:
You should remove or comment these lines:

#Foscam-fi9821w example
- rtsp_url: "rtsp://:@:/videoMain"
#Dahua IPC-HDW4200S example or IPC-HDW4300S
- rtsp_url: "rtsp://:@:/cam/realmo$

SvenVD, 2018-04-07:
Is your issue solved now?

musicman251, 2018-04-07:
Yes. You were correct. I had not commented out the unused sample lines. New to this...should have caught that!

closed 2018-04-07 by musicman251