SvenVD/rpisurv #55: Failing to start Rpisurv

NZRobNZ, 2018-04-16:
I'm unsure where we are going wrong here, I've checked `/etc/rpisurv` for tabs and all indentation is spaces per the example file, yet on reboot I get the following
```
[FAILED] Failed to start Rpisurv Raspberry Pi Surveillance.
See 'systemctl status status rpisurv.service' for details.
```
Running `sudo journalctl -u rpisurv` gives me these logs.
```
-- Logs begin at Mon 2018-04-16 15:51:40 NZST, end at Mon 2018-04-16 15:52:50 NZST. --
Apr 16 15:51:42 aae-rtsp01 systemd[1]: Started Rpisurv Raspberry Pi Surveillance.
Apr 16 15:51:45 aae-rtsp01 rpisurv[347]: 2018/04/16 15:51:45 - l_default - DEBUG - autodetected resolution of['1824', '984']
Apr 16 15:51:45 aae-rtsp01 rpisurv[347]: 2018/04/16 15:51:45 - l_default - DEBUG - nr_of_columns = 1
Apr 16 15:51:45 aae-rtsp01 rpisurv[347]: 2018/04/16 15:51:45 - l_default - DEBUG - interval_check_status = 25
Apr 16 15:51:45 aae-rtsp01 rpisurv[347]: 2018/04/16 15:51:45 - l_default - DEBUG - camera_streams config option exist, using this one
Apr 16 15:51:45 aae-rtsp01 rpisurv[347]: Traceback (most recent call last):
Apr 16 15:51:45 aae-rtsp01 rpisurv[347]:   File "surveillance.py", line 372, in <module>
Apr 16 15:51:45 aae-rtsp01 rpisurv[347]:     all_camera_streams=setup_camera_streams(camera_streams)
Apr 16 15:51:45 aae-rtsp01 rpisurv[347]:   File "surveillance.py", line 247, in setup_camera_streams
Apr 16 15:51:45 aae-rtsp01 rpisurv[347]:     cam_stream=CameraStream(cam_stream_name,camera_stream)
Apr 16 15:51:45 aae-rtsp01 rpisurv[347]:   File "surveillance.py", line 29, in __init__
Apr 16 15:51:45 aae-rtsp01 rpisurv[347]:     self.rtsp_url = camera_stream["rtsp_url"]
Apr 16 15:51:45 aae-rtsp01 rpisurv[347]: KeyError: 'rtsp_url'
Apr 16 15:51:45 aae-rtsp01 systemd[1]: rpisurv.service: Main process exited, code=exited, status=1/FAILURE
Apr 16 15:51:45 aae-rtsp01 systemd[1]: rpisurv.service: Unit entered failed state.
Apr 16 15:51:45 aae-rtsp01 systemd[1]: rpisurv.service: Failed with result 'exit-code'.
Apr 16 15:51:46 aae-rtsp01 systemd[1]: rpisurv.service: Service hold-off time over, scheduling restart.
Apr 16 15:51:46 aae-rtsp01 systemd[1]: Stopped Rpisurv Raspberry Pi Surveillance.
Apr 16 15:51:46 aae-rtsp01 systemd[1]: Started Rpisurv Raspberry Pi Surveillance.
Apr 16 15:51:47 aae-rtsp01 rpisurv[410]: 2018/04/16 15:51:47 - l_default - DEBUG - autodetected resolution of['1824', '984']
Apr 16 15:51:47 aae-rtsp01 rpisurv[410]: 2018/04/16 15:51:47 - l_default - DEBUG - nr_of_columns = 1
Apr 16 15:51:47 aae-rtsp01 rpisurv[410]: 2018/04/16 15:51:47 - l_default - DEBUG - interval_check_status = 25
Apr 16 15:51:47 aae-rtsp01 rpisurv[410]: 2018/04/16 15:51:47 - l_default - DEBUG - camera_streams config option exist, using this one
Apr 16 15:51:47 aae-rtsp01 rpisurv[410]: Traceback (most recent call last):
Apr 16 15:51:47 aae-rtsp01 rpisurv[410]:   File "surveillance.py", line 372, in <module>
Apr 16 15:51:47 aae-rtsp01 rpisurv[410]:     all_camera_streams=setup_camera_streams(camera_streams)
Apr 16 15:51:47 aae-rtsp01 rpisurv[410]:   File "surveillance.py", line 247, in setup_camera_streams
Apr 16 15:51:47 aae-rtsp01 rpisurv[410]:     cam_stream=CameraStream(cam_stream_name,camera_stream)
Apr 16 15:51:47 aae-rtsp01 rpisurv[410]:   File "surveillance.py", line 29, in __init__
Apr 16 15:51:47 aae-rtsp01 rpisurv[410]:     self.rtsp_url = camera_stream["rtsp_url"]
Apr 16 15:51:47 aae-rtsp01 rpisurv[410]: KeyError: 'rtsp_url'
Apr 16 15:51:47 aae-rtsp01 systemd[1]: rpisurv.service: Main process exited, code=exited, status=1/FAILURE
Apr 16 15:51:47 aae-rtsp01 systemd[1]: rpisurv.service: Unit entered failed state.
Apr 16 15:51:47 aae-rtsp01 systemd[1]: rpisurv.service: Failed with result 'exit-code'.
Apr 16 15:51:47 aae-rtsp01 systemd[1]: rpisurv.service: Service hold-off time over, scheduling restart.
Apr 16 15:51:47 aae-rtsp01 systemd[1]: Stopped Rpisurv Raspberry Pi Surveillance.
Apr 16 15:51:47 aae-rtsp01 systemd[1]: Started Rpisurv Raspberry Pi Surveillance.
Apr 16 15:51:48 aae-rtsp01 rpisurv[436]: 2018/04/16 15:51:48 - l_default - DEBUG - autodetected resolution of['1824', '984']
Apr 16 15:51:48 aae-rtsp01 rpisurv[436]: 2018/04/16 15:51:48 - l_default - DEBUG - nr_of_columns = 1
Apr 16 15:51:48 aae-rtsp01 rpisurv[436]: 2018/04/16 15:51:48 - l_default - DEBUG - interval_check_status = 25
Apr 16 15:51:48 aae-rtsp01 rpisurv[436]: 2018/04/16 15:51:48 - l_default - DEBUG - camera_streams config option exist, using this one
Apr 16 15:51:48 aae-rtsp01 rpisurv[436]: Traceback (most recent call last):
Apr 16 15:51:48 aae-rtsp01 rpisurv[436]:   File "surveillance.py", line 372, in <module>
Apr 16 15:51:48 aae-rtsp01 rpisurv[436]:     all_camera_streams=setup_camera_streams(camera_streams)
Apr 16 15:51:48 aae-rtsp01 rpisurv[436]:   File "surveillance.py", line 247, in setup_camera_streams
Apr 16 15:51:48 aae-rtsp01 rpisurv[436]:     cam_stream=CameraStream(cam_stream_name,camera_stream)
Apr 16 15:51:48 aae-rtsp01 rpisurv[436]:   File "surveillance.py", line 29, in __init__
Apr 16 15:51:48 aae-rtsp01 rpisurv[436]:     self.rtsp_url = camera_stream["rtsp_url"]
Apr 16 15:51:48 aae-rtsp01 rpisurv[436]: KeyError: 'rtsp_url'
Apr 16 15:51:48 aae-rtsp01 systemd[1]: rpisurv.service: Main process exited, code=exited, status=1/FAILURE
Apr 16 15:51:48 aae-rtsp01 systemd[1]: rpisurv.service: Unit entered failed state.
Apr 16 15:51:48 aae-rtsp01 systemd[1]: rpisurv.service: Failed with result 'exit-code'.
Apr 16 15:51:48 aae-rtsp01 systemd[1]: rpisurv.service: Service hold-off time over, scheduling restart.
Apr 16 15:51:48 aae-rtsp01 systemd[1]: Stopped Rpisurv Raspberry Pi Surveillance.
Apr 16 15:51:48 aae-rtsp01 systemd[1]: Started Rpisurv Raspberry Pi Surveillance.
Apr 16 15:51:49 aae-rtsp01 rpisurv[441]: 2018/04/16 15:51:49 - l_default - DEBUG - autodetected resolution of['1824', '984']
Apr 16 15:51:49 aae-rtsp01 rpisurv[441]: 2018/04/16 15:51:49 - l_default - DEBUG - nr_of_columns = 1
Apr 16 15:51:49 aae-rtsp01 rpisurv[441]: 2018/04/16 15:51:49 - l_default - DEBUG - interval_check_status = 25
Apr 16 15:51:49 aae-rtsp01 rpisurv[441]: 2018/04/16 15:51:49 - l_default - DEBUG - camera_streams config option exist, using this one
Apr 16 15:51:49 aae-rtsp01 rpisurv[441]: Traceback (most recent call last):
Apr 16 15:51:49 aae-rtsp01 rpisurv[441]:   File "surveillance.py", line 372, in <module>
Apr 16 15:51:49 aae-rtsp01 rpisurv[441]:     all_camera_streams=setup_camera_streams(camera_streams)
[... 17 lines cut ...]
Apr 16 15:51:51 aae-rtsp01 rpisurv[446]:     all_camera_streams=setup_camera_streams(camera_streams)
Apr 16 15:51:51 aae-rtsp01 rpisurv[446]:   File "surveillance.py", line 247, in setup_camera_streams
Apr 16 15:51:51 aae-rtsp01 rpisurv[446]:     cam_stream=CameraStream(cam_stream_name,camera_stream)
Apr 16 15:51:51 aae-rtsp01 rpisurv[446]:   File "surveillance.py", line 29, in __init__
Apr 16 15:51:51 aae-rtsp01 rpisurv[446]:     self.rtsp_url = camera_stream["rtsp_url"]
Apr 16 15:51:51 aae-rtsp01 rpisurv[446]: KeyError: 'rtsp_url'
Apr 16 15:51:51 aae-rtsp01 systemd[1]: rpisurv.service: Main process exited, code=exited, status=1/FAILURE
Apr 16 15:51:51 aae-rtsp01 systemd[1]: rpisurv.service: Unit entered failed state.
Apr 16 15:51:51 aae-rtsp01 systemd[1]: rpisurv.service: Failed with result 'exit-code'.
Apr 16 15:51:51 aae-rtsp01 systemd[1]: rpisurv.service: Service hold-off time over, scheduling restart.
Apr 16 15:51:51 aae-rtsp01 systemd[1]: Stopped Rpisurv Raspberry Pi Surveillance.
Apr 16 15:51:51 aae-rtsp01 systemd[1]: rpisurv.service: Start request repeated too quickly.
Apr 16 15:51:51 aae-rtsp01 systemd[1]: Failed to start Rpisurv Raspberry Pi Surveillance.
Apr 16 15:51:51 aae-rtsp01 systemd[1]: rpisurv.service: Unit entered failed state.
Apr 16 15:51:51 aae-rtsp01 systemd[1]: rpisurv.service: Failed with result 'exit-code'.
```

I'm not sure where to look next. If I run omxplayer directly I get
```
$ omxplayer rtsp://192.168.1.10:8228/h264 --live --hw --blank
* failed to open vchiq instance
$ sudo chmod a+rw /dev/vchiq
$ omxplayer rtsp://192.168.1.10:8228/h264 --live --hw --blank
```
I then get a full screen stream as expected.. I'm not sure where to look now to translate this into a working instance of the rpisurv service.

SvenVD, 2018-04-18:
Please post your config (with dummy password) something is wrong in your url

NZRobNZ, 2018-04-19:
Thanks Sven, config is as follows;
```

essentials:
    camera_streams:
    # Test Stream
        - rtsp url: "rtsp://192.168.10.150:8557/h264"
    #How many columns you want the program to use, it will autocalculate the amount of row needed based on the resolution of your screen
    nr_of_columns: "1"

    #Autostretching makes sures all of your pixels will be uses to display streams, so you have a maximum surveillance area.
    #But this will probably destroy the aspect ratio of your last stream.
    #autostretch: False

    #rpisurv detects the connectable streams on startup,
    #if this option is enabled it will keep the first calculated layout for the lifetime of the rpisurv running even if some cameras go down and come up, instead of redrawing the screen
    #When this option is set to True, rpisurv ignores all cameras configured that were not connectable when rpisurv was started
    keep_first_screen_layout: False

#!!!Normal users do not need to edit this section!!!
advanced:
    ##Enable this option if you want to have a fixed width of all your camera streams,
    ##By default rpisurv autocalculates this value, this can cause streams to get "stretched",
    ##if this value exceeds the available width, rpisurv will fallback to autocalculation
    #fixed_width: 500

    ##Enable this option if you want to have a fixed height for all your camera streams,
    ##By default rpisurv autocalculates this value, this can cause streams to get "stretched",
    ##if this value exceeds the available height, rpisurv will fallback to autocalculation
    #fixed_height: 500

    #Rpisurv sends usage stats to it's statistics server to give the rpisurv community an idea how widespread this software is being used
    #No performance impact is measured when using this option, no sensitive data is being sent. All data is anonymised.
    #By default this is false
    #update_stats: False

    #By default rpisurv checks every 25 seconds if it needs to redraw the screen
    #interval_check_status: 25

    #By default rpisurv will check memory usage, set to False to skip memory usage check
    #memory_usage_check: True

#These are fallbacks if autodection fails;
#Normally you do not need to configure these
fallbacks:
    resolution:
        width: "1920"
        height: "1080"
```
I hope that's what you needed

SvenVD, 2018-04-19:
It should be rtsp_url not rtsp url with space

mlschoeneman, 2018-04-21:
I am having a similar issue, RPI is booting into console with no camera viewing. Below is my config, all streams tested in omxplayer and vlc successfully.

essentials:
    #These are the rtsp streams you want the program to display
    #The streams will be displayed in the order you present them here
    camera_streams:
        #Dahua 773CamSideYard
        - rtsp_url: "rtsp://admin:**********@192.168.13.22:3000/cam/realmonitor?channel=1&subtype=0&unicast=true&proto=Onvif"
          #Enable this option if you want the rtsp stream to stream over tcp instead of udp, this may solve a "smearing" effect on some setups. Defaults to false
          #Note that you need a version of omxplayer older then 14 March 2016 for this option to work
          rtsp_over_tcp: true
        #Dahua 773CamMiddle Yard
        - rtsp_url: "rtsp://admin:**********@192.168.13.23:3000/cam/realmonitor?channel=1&subtype=0&unicast=true&proto=Onvif"
        #Hikvision 773CamFrontPorch
        - rtsp_url: "rtsp://admin:**********@192.168.13.21:3000/Streaming/Channels/1"
        #Dahua 773CamBackyard
        - rstp_url: "rtsp://admin:**********@192.168.13.24:3000/cam/realmonitor?channel=1&subtype=0&unicast=true&proto=Onvif"

    #How many columns you want the program to use, it will autocalculate the amount of row needed based on the resolution of your screen
    nr_of_columns: "2"

    #Autostretching makes sures all of your pixels will be uses to display streams, so you have a maximum surveillance area.
    #But this will probably destroy the aspect ratio of your last stream.
    #autostretch: False

    #rpisurv detects the connectable streams on startup,
    #if this option is enabled it will keep the first calculated layout for the lifetime of the rpisurv running even if some cameras go down and come up, instead of redrawing the screen
    #When this option is set to True, rpisurv ignores all cameras configured that were not connectable when rpisurv was started
    keep_first_screen_layout: True

#!!!Normal users do not need to edit this section!!!
advanced:
    ##Enable this option if you want to have a fixed width of all your camera streams,
    ##By default rpisurv autocalculates this value, this can cause streams to get "stretched",
    ##if this value exceeds the available width, rpisurv will fallback to autocalculation
    #fixed_width: 500

    ##Enable this option if you want to have a fixed height for all your camera streams,
    ##By default rpisurv autocalculates this value, this can cause streams to get "stretched",
    ##if this value exceeds the available height, rpisurv will fallback to autocalculation
    #fixed_height: 500

    #Rpisurv sends usage stats to it's statistics server to give the rpisurv community an idea how widespread this software is being used
    #No performance impact is measured when using this option, no sensitive data is being sent. All data is anonymised.
    #By default this is false
    #update_stats: False

    #By default rpisurv checks every 25 seconds if it needs to redraw the screen
    #interval_check_status: 25

    #By default rpisurv will check memory usage, set to False to skip memory usage check
    #memory_usage_check: True

#These are fallbacks if autodection fails;
#Normally you do not need to configure these
fallbacks:
    resolution:
        width: "1920"
        height: "1080"

mlschoeneman, 2018-04-21:
boot log:
Apr 21 15:26:13 rpi systemd-timesyncd[264]: Synchronized to time server 206.108.0.132:123 (2.debian.pool.ntp.org).
Apr 21 15:26:13 rpi systemd[1]: Time has been changed
Apr 21 15:26:13 rpi systemd[1]: apt-daily.timer: Adding 5h 11min 19.292349s random time.
Apr 21 15:26:13 rpi systemd[1]: apt-daily-upgrade.timer: Adding 53min 3.174921s random time.
Apr 21 15:26:14 rpi rpisurv[678]: 2018/04/21 15:26:14 - l_default - DEBUG - autodetected resolution of['1920', '1080']
Apr 21 15:26:14 rpi rpisurv[678]: 2018/04/21 15:26:14 - l_default - DEBUG - nr_of_columns = 2
Apr 21 15:26:14 rpi rpisurv[678]: 2018/04/21 15:26:14 - l_default - DEBUG - interval_check_status = 25
Apr 21 15:26:14 rpi rpisurv[678]: 2018/04/21 15:26:14 - l_default - DEBUG - camera_streams config option exist, using this one
Apr 21 15:26:14 rpi rpisurv[678]: Traceback (most recent call last):
Apr 21 15:26:14 rpi rpisurv[678]:   File "surveillance.py", line 372, in <module>
Apr 21 15:26:14 rpi rpisurv[678]:     all_camera_streams=setup_camera_streams(camera_streams)
Apr 21 15:26:14 rpi rpisurv[678]:   File "surveillance.py", line 247, in setup_camera_streams
Apr 21 15:26:14 rpi rpisurv[678]:     cam_stream=CameraStream(cam_stream_name,camera_stream)
Apr 21 15:26:14 rpi rpisurv[678]:   File "surveillance.py", line 29, in __init__
Apr 21 15:26:14 rpi rpisurv[678]:     self.rtsp_url = camera_stream["rtsp_url"]
Apr 21 15:26:14 rpi rpisurv[678]: KeyError: 'rtsp_url'
Apr 21 15:26:14 rpi systemd[1]: rpisurv.service: Main process exited, code=exited, status=1/FAILURE
Apr 21 15:26:14 rpi systemd[1]: rpisurv.service: Unit entered failed state.
Apr 21 15:26:14 rpi systemd[1]: rpisurv.service: Failed with result 'exit-code'.
Apr 21 15:26:14 rpi systemd[1]: rpisurv.service: Service hold-off time over, scheduling restart.
Apr 21 15:26:14 rpi systemd[1]: Stopped Rpisurv Raspberry Pi Surveillance.

mlschoeneman, 2018-04-21:
SvenVD--Thanks for the quick response! I figured it out and have it running flawlessly. Thank you so much for your contribution.

closed 2018-06-18 by SvenVD