SvenVD/rpisurv #56: Smearing Effect (How to downgrade omxplayer)

mlschoeneman, 2018-04-23:
I  am experiencing the smearing effect on all four of my cameras. I see that I need to downgrade my omxplayer to before 03142016. How do I go about downgrading? I am on RPi 3 B+ Stretch 04192018. 

My current omxplayer version is:
Build date: Wed, 05 Jul 2017 16:40:39 +0100
Version   : 5a25a57 [debian]
Repository: git@github.com:XECDesign/omxplayer.git

"If you have a "smearing" effect this option may resolve it. Note that you need a version of omxplayer older then 14 March 2016 (https://github.com/popcornmix/omxplayer/pull/433) to do this."

SvenVD, 2018-04-23:
No it is the way around your need an omxplayer version older then 14 March 2016 to enable rtsp_over_tcp.

SvenVD, 2018-04-23:
I will update the readme because it is stated wrong

mlschoeneman, 2018-04-23:
Ok, how do I eliminate the smearing effect then? Is there a simple way to block udp traffic to the pi from those ip addresses individually?

mlschoeneman, 2018-04-23:
Ahhh ok, I see your wording there now. Each stream needs to be tagged with the rtsp_over_tcp: true. Configured for all four cameras and have not dropped a single frame. 

Thanks for the great help!

closed 2018-04-23 by mlschoeneman

renatomarcio, 2018-05-13:
Having the same problem. Doubt: rtsp_over_tcp: true needs to be repeated after each url?
Doing so makes application crash and not load. Even leaving a single extra space after each url makes system crash.
Leaving rtsp_over_tcp: true only once, at the end keeps smearing... 
What else can I do?