SvenVD/rpisurv #60: 2.0 Beta - yaml.parser.ParserError

franzaut, 2018-08-14:
yaml.parser.ParserError: while parsing a block collection
  in "conf/surveillance.yml", line 74, column 11
expected <block end>, but found '?'
  in "conf/surveillance.yml", line 78, column 11

Error in the original surveillance.yml?

<unknown>): did not find expected '-' indicator while parsing a block collection at line 74 column 11

SvenVD, 2018-08-16:
Although it isn't the goal that the example config file is a config that passes the initialization phase of rpisurv, because of the dummy "documentation" examples, it should at least pass the yaml parser, to be a good example file and not confuse the reader with indentation errors. I fixed it in https://github.com/SvenVD/rpisurv/commit/6f541df3002af4099761e07d5a94db015dc5987d

nice catch!

SvenVD, 2018-08-16:
Can you confirm on your side it is OK now? Thanks

franzaut, 2018-08-16:
Yes thank you, it is OK.

closed 2018-08-16 by franzaut