SvenVD/rpisurv #62: rtsp_over_tcp: true and omxplayer older than Feb. 2016

chbally, 2018-08-19:
I have several cams which will not run over UDP and need TCP!
Why don't you change the installer and install the with TCP working version instead of 
apt-get install omxplayer...

I got everything running tonight exept for my tcp-cams.

Thanks for rpisurv2.0
With more cams I can test the new features ;-)

SvenVD, 2018-08-22:
Hi, normally apt-get install omxplayer gets you the latest omxplayer that the local cache on the system knows about. You should run apt-get update before this command to update the local cache, I now included the latter in the installer.

The installer does not interfere with an existing omxplayer, installed either from source or from package. It is up to the user to update his existing omxplayer binary, or not. There may be reasons to not have it updated.

Does this answer your question?

Thanks for testing version 2.0

chbally, 2018-08-22:
For all th other packets apt-get install is a good idea.
But for the omxplayer I would download the player generally from here: 
https://omxplayer.sconde.net/builds/omxplayer_0.3.7~git20160206~cb91001_armhf.deb
and let the installer do the installing.
Except the newer version of omxplayer has other advantages.

SvenVD, 2018-08-22:
On a fresh install I would like to use the latest omxplayer available to the user in the package management system. Since all kind of cameras are being used in the wild, I think, the user will benefit most with the most recent stable version of omxplayer.

On an existing install I would not like to interfere with an existing omxplayer install, since that install is most likely providing a working setup and upgrading or changing omxplayer version could change this for the worst. On top of that, some users fork omxplayer and I do not want to override their install directly from source.

closed 2018-09-24 by SvenVD