SvenVD/rpisurv #66: Rpisurv -- start and fails

ngentil, 2018-08-26:
● rpisurv.service - Rpisurv Raspberry Pi Surveillance
   Loaded: loaded (/etc/systemd/system/rpisurv.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sun 2018-08-26 02:56:22 UTC; 13s ago
  Process: 6290 ExecStart=/usr/bin/rpisurv (code=exited, status=1/FAILURE)
 Main PID: 6290 (code=exited, status=1/FAILURE)

Aug 26 02:56:22 raspberrypi systemd[1]: rpisurv.service: Unit entered failed state.
Aug 26 02:56:22 raspberrypi systemd[1]: rpisurv.service: Failed with result 'exit-code'.
Aug 26 02:56:22 raspberrypi systemd[1]: rpisurv.service: Service hold-off time over, scheduling restart.
Aug 26 02:56:22 raspberrypi systemd[1]: Stopped Rpisurv Raspberry Pi Surveillance.
Aug 26 02:56:22 raspberrypi systemd[1]: rpisurv.service: Start request repeated too quickly.
Aug 26 02:56:22 raspberrypi systemd[1]: Failed to start Rpisurv Raspberry Pi Surveillance.
Aug 26 02:56:22 raspberrypi systemd[1]: rpisurv.service: Unit entered failed state.
Aug 26 02:56:22 raspberrypi systemd[1]: rpisurv.service: Failed with result 'exit-code'.

SvenVD, 2018-08-26:
Hi, Please show me the output of /var/log/daemon.log, so I can trace the why it is not starting

ngentil, 2018-08-26:
Hey! got it working on a raspberry pi 3b 
- what worked: Commenting out all rtsp streams in /etc/rpisurv

closed 2018-08-26 by ngentil

SvenVD, 2018-08-26:
Nice you got it working. The rtsp streams in rpisurv and dummy documentation examples, they are not technically valid to be started as is. So you should remove/replace them or comment them.

ngentil, 2018-08-27:
Cheers.. Thanks! - im also dissecting what i think is a memory leak, after say 12hrs a 30 second delay builds up - although every 24hrs i have setup a cronjob for a reboot, otherwise the delay increases to about 10 mins + i have recently increased the gpu mem from 64mb to 128mb - time will only tell, will keep you posted...

SvenVD, 2018-08-27:
Which version of rpisurv are you running?

ngentil, 2018-08-28:
Hey Dude! 

Got Back home after work to check the delay on the stream - after 8 hours there was less than 2 second delay, which is the best i think i can ask for considering the hardware and software limitations, although interesting side note, the output of /var/log/daemon.log mentions low GPU MEM so i upped the split from 128mb to 256mb and reboot and this is the following output from /var/log/daemon.

===============================================================================

pi@raspberrypi:~ $ tail -f  /var/log/daemon.log
Aug 28 06:57:40 raspberrypi systemd[565]: Listening on GnuPG cryptographic agent (ssh-agent emulation).
Aug 28 06:57:40 raspberrypi systemd[565]: Reached target Sockets.
Aug 28 06:57:40 raspberrypi systemd[565]: Reached target Basic System.
Aug 28 06:57:40 raspberrypi systemd[565]: Reached target Default.
Aug 28 06:57:40 raspberrypi systemd[565]: Startup finished in 55ms.
Aug 28 06:57:40 raspberrypi systemd[1]: Started User Manager for UID 1000.
Aug 28 06:57:59 raspberrypi rpisurv[323]: 2018/08/28 06:57:59 - l_default - DEBUG - stats_counter is 5. Only sending every 40
Aug 28 06:57:59 raspberrypi rpisurv[323]: 2018/08/28 06:57:59 - l_default - DEBUG - Free gpu memory value is 190M  **<<<-------- 19 seconds in and 66mb of GPU MEM used**
Aug 28 06:57:59 raspberrypi rpisurv[323]: 2018/08/28 06:57:59 - l_default - DEBUG - Free memory in bytes: 199229440.0
Aug 28 06:57:59 raspberrypi rpisurv[323]: 2018/08/28 06:57:59 - l_default - DEBUG - 192.168.1.218 : 554 Connectable
Aug 28 06:58:24 raspberrypi rpisurv[323]: 2018/08/28 06:58:24 - l_default - DEBUG - stats_counter is 6. Only sending every 40
Aug 28 06:58:24 raspberrypi rpisurv[323]: 2018/08/28 06:58:24 - l_default - DEBUG - Free gpu memory value is 190M
Aug 28 06:58:24 raspberrypi rpisurv[323]: 2018/08/28 06:58:24 - l_default - DEBUG - Free memory in bytes: 199229440.0
Aug 28 06:58:24 raspberrypi rpisurv[323]: 2018/08/28 06:58:24 - l_default - DEBUG - 192.168.1.218 : 554 Connectable
Aug 28 06:58:49 raspberrypi rpisurv[323]: 2018/08/28 06:58:49 - l_default - DEBUG - stats_counter is 7. Only sending every 40
Aug 28 06:58:49 raspberrypi rpisurv[323]: 2018/08/28 06:58:49 - l_default - DEBUG - Free gpu memory value is 961  **<<<-------- a little over a minute in and less than 1kb of GPU MEM remaining**
Aug 28 06:58:49 raspberrypi rpisurv[323]: 2018/08/28 06:58:49 - l_default - DEBUG - Free memory in bytes: 961.0
Aug 28 06:58:49 raspberrypi rpisurv[323]: 2018/08/28 06:58:49 - l_default - ERROR - Free gpu mem is 961.0 bytes which is less than 80000000 bytes. Streams might fail to start. Consider assigning more memory to gpu in /boot/config.txt with the gpu_mem option
Aug 28 06:58:49 raspberrypi rpisurv[323]: 2018/08/28 06:58:49 - l_default - DEBUG - 192.168.1.218 : 554 Connectable
Aug 28 06:59:14 raspberrypi rpisurv[323]: 2018/08/28 06:59:14 - l_default - DEBUG - stats_counter is 8. Only sending every 40
Aug 28 06:59:14 raspberrypi rpisurv[323]: 2018/08/28 06:59:14 - l_default - DEBUG - Free gpu memory value is 190M  **<<<-------- almost 2 minutes and gpu mem has been purged and reloaded - or so it appears**
Aug 28 06:59:14 raspberrypi rpisurv[323]: 2018/08/28 06:59:14 - l_default - DEBUG - Free memory in bytes: 199229440.0
Aug 28 06:59:14 raspberrypi rpisurv[323]: 2018/08/28 06:59:14 - l_default - DEBUG - 192.168.1.218 : 554 Connectable
Aug 28 06:59:39 raspberrypi rpisurv[323]: 2018/08/28 06:59:39 - l_default - DEBUG - stats_counter is 9. Only sending every 40
Aug 28 06:59:39 raspberrypi rpisurv[323]: 2018/08/28 06:59:39 - l_default - DEBUG - Free gpu memory value is 0
Aug 28 06:59:39 raspberrypi rpisurv[323]: 2018/08/28 06:59:39 - l_default - DEBUG - Free memory in bytes: 0.0
Aug 28 06:59:39 raspberrypi rpisurv[323]: 2018/08/28 06:59:39 - l_default - ERROR - Free gpu mem is 0.0 bytes which is less than 80000000 bytes. Streams might fail to start. Consider assigning more memory to gpu in /boot/config.txt with the gpu_mem option
Aug 28 06:59:39 raspberrypi rpisurv[323]: 2018/08/28 06:59:39 - l_default - DEBUG - 192.168.1.218 : 554 Connectable
Aug 28 07:00:04 raspberrypi rpisurv[323]: 2018/08/28 07:00:04 - l_default - DEBUG - stats_counter is 10. Only sending every 40
Aug 28 07:00:05 raspberrypi rpisurv[323]: 2018/08/28 07:00:05 - l_default - DEBUG - Free gpu memory value is 9K
Aug 28 07:00:05 raspberrypi rpisurv[323]: 2018/08/28 07:00:05 - l_default - DEBUG - Free memory in bytes: 9216.0
Aug 28 07:00:05 raspberrypi rpisurv[323]: 2018/08/28 07:00:05 - l_default - ERROR - Free gpu mem is 9216.0 bytes which is less than 80000000 bytes. Streams might fail to start. Consider assigning more memory to gpu in /boot/config.txt with the gpu_mem option

===============================================================================
im led to believe this is a underlying problem within omxplayer memory managment that has never been rectified

SvenVD, 2018-08-29:
Hi, thanks for sharing, however in my test setup, I can not reproduce, my memory counters are quite stable. What version of omxplayer you are using? In the end it probably depends on the kind of feed.