SvenVD/rpisurv #71: v2.o

paczmax, 2018-09-22:
previous version was working fine after upgrade cat see y cameras
Sep 22 07:16:39 raspberrypi systemd[1]: rpisurv.service: Failed with result 'exi
lines 1-7/7 (END)...skipping...
● rpisurv.service - Rpisurv Raspberry Pi Surveillance
   Loaded: loaded (/etc/systemd/system/rpisurv.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Sat 2018-09-22 07:16:39 UTC; 6s ago
  Process: 5930 ExecStart=/usr/bin/rpisurv (code=exited, status=1/FAILURE)
 Main PID: 5930 (code=exited, status=1/FAILURE)

Sep 22 07:16:39 raspberrypi systemd[1]: rpisurv.service: Failed with result 'exit-code'.

SvenVD, 2018-09-22:
Hi, what was your previous version? Could you send me the main.log and/or /var/log/daemon.log? Thanks

paczmax, 2018-09-24:
/var/log/daemon.log
[test_test.pdf](https://github.com/SvenVD/rpisurv/files/2409483/test_test.pdf)

SvenVD, 2018-09-24:
Hi it seems you are not running rpisurv but some other software according to the daemon.log

"Sep 21 05:59:15 raspberrypi displaycameras[346]: /etc/init.d/displaycameras: line 14:
0: command not found"

paczmax, 2018-09-27:
my update im stuck now i see its running bug just geting balck screen, I
dont know linux at all but, I fallow all and not getting
[image: Capture_rpistatus.PNG]
any result. 1st I got v1 installed was ok but didnt work on RP3 after
swaping Sd to RPi 2+  after upgrading to 2.0, i cant get it to work. I
formated sd run all over again and nothing, it didt run on RPi 3(the newest
but run on 2+, all i did was swaped sd card).

paczmax, 2018-09-27:
[image: Capturerpi.PNG]

SvenVD, 2018-09-29:
See my previous post to begin with:

Hi it seems you are not running rpisurv but some other software according to the daemon.log

"Sep 21 05:59:15 raspberrypi displaycameras[346]: /etc/init.d/displaycameras: line 14:
0: command not found"

You are not even running rpisurv according to the logs u provided