SvenVD/rpisurv #72: Black Screen Shortly After Launch

tempUser99, 2018-09-23:
Shortly after launching I am seeing a completely black screen although the monitor still is detecting output from the hardware. I have attempted to use multiple different monitors and the same behavior is shown.

Steps:
1) Add four (4) or more valid RTSP feeds to the configuration file (3 streams appears to work fine).
2) Launch rpisurv
3) Observe placeholders
4) Observe each stream come up (1, 2, and 3 streams seem to load fine)
5) As 4th stream is expected to load the entire output is lost

SSH is still available and response appears to be good but hardware output appears to be lost (monitor still detects unit is attached and signal is present).

SvenVD, 2018-09-23:
What is the resolution of each stream? Most likely you are exceeding the maximum resolution the pi can handle

tempUser99, 2018-09-25:
Each stream resolution was at 720p and the monitor is at 1080p. After reducing the resolution it appears I am now able to stream up to 4 at one time. More RTSP streams appears to show the same behavior.

Is there a location for documentation of the constraint? Is omxplayer the limiting factor?

SvenVD, 2018-09-25:
I believe it was some constraint of the rpi GPU, I can't recall where it was documented

sanakess, 2018-09-26:
I found solution. You need to change raspi-config resolution to 720p on 1080p display and problem gone

SvenVD, 2018-10-01:
Maybe you are running against one of the problems as described here: https://www.raspberrypi.org/forums/viewtopic.php?p=1061684&sid=471a6a2a7d8fd82e9d5643e2b325ff50#p1061684

tempUser99, 2018-10-03:
This is really interesting. I am now actually finding that using omxplayer directly I can actually get the monitor output to fail with just two streams. When using the --win argument flag I can run two streams and the screen will go blank, while running at least one of the streams without the --win flag runs totally fine with both streams running at the same time. I think you might be right, the issue may be related to the article issues you attached. But it isn't the number of streams in the case clearly because only two can make it fail.

Below are listed the executed commands:
`omxplayer 'rtsp://username:password@10.0.1.2:80/videoSub' --avdict rtsp_transport:tcp --win "144, 756, 944, 1156" --live --video_queue 4 --fps 30`

`omxplayer 'rtsp://username:password@10.0.1.3:80/videoSub' --avdict rtsp_transport:tcp --win "802, 1546, 1154, 1788" --live --video_queue 4 --fps 30`

Although running this (below) combination of command is successful with video output:
`omxplayer 'rtsp://username:password@10.0.1.2:80/videoSub' --avdict rtsp_transport:tcp --win "144, 756, 944, 1156" --live --video_queue 4 --fps 30`

`omxplayer 'rtsp://username:password@10.0.1.3:80/videoSub' --avdict rtsp_transport:tcp --live --video_queue 4 --fps 30`

Any thoughts on possible --win issues?

SvenVD, 2018-10-03:
I have no idea on the --win issues. This is something omxplayer related. Linking your omxplayer thread: https://github.com/popcornmix/omxplayer/issues/653

tempUser99, 2018-10-03:
Much appreciated. I think it is looking like a resolution issue. I will track it there. Thanks again for the help.

closed 2018-10-03 by tempUser99