SvenVD/rpisurv #73: URLError when using MJPEG Stream

tempUser99, 2018-09-25:
When attempting to use an MJPEG stream I am receiving the following error:

`CameraStream: screen1_cam_stream1 http://<hidden_username>:<hidden_password>@10.0.1.10:80/img/video.mjpeg Not Connectable (URLerror2), URLError(gaierror(-2, 'Name or service not known'),)`

The stream is accessible and the IP ping resolves fine from the machine.

This issue is while using Beta 2.0.

SvenVD, 2018-09-25:
Hi, I can not reproduce this error. Are there any special chars in the username or password?

SvenVD, 2018-09-25:
Yes, but it does not use any credentials. I am able to reproduce an error with credentials and http basic authentication. I am working on  fix

SvenVD, 2018-09-25:
New version is online, can you test it out

tempUser99, 2018-09-25:
Absolutely, I will test shortly. Thank you for the quick turn around.

tempUser99, 2018-09-26:
Testing the new version, the MJPEG stream does report as Connectable but rpisurv now stays with the connecting.png.

I am attempting to diagnose the issue now.

SvenVD, 2018-09-26:
Hi, I managed to get a mjpg ip camera WITH basic authentication. I still can not reproduce your issue.

/usr/bin/omxplayer.bin --video_fifo 1 --video_queue 1 --live --timeout 60 --aidx -1 -o hdmi --threshold 0 http://<username>:<password>@192.168.2.12/video/mjpg.cgi --win 0 0 960 540 --dbus_name org.mpris.MediaPlayer2.screen1_cam_stream1

Works perfectly fine with 2.0.beta10 on my test setup

tempUser99, 2018-10-03:
It must be a bug in the actual mode the specific camera I am working with is serving the MJPEG. I can change the output, toggling between MJPEG and MPEG-4, and it appears moving the MPEG-4 serves the correct output.

It was still important to get the update to assess the connection properly. The MPEG-4 wouldn't have worked without it as it uses the same URL to connect. Thank you for providing that update and thank you for diagnosing this issue.

closed 2018-10-03 by tempUser99