SvenVD/rpisurv #74: "No video stream defined" error

RobertoA109, 2018-10-10:
Hi Sven,
I am successfully using rpisurv to display a set of 16 cameras on a rotating 4x4 grid.
Thank you very much and once again for your work. RPISURV is a piece of jewellery.
Recently I have purchased 10 mini-dome network cameras (brand ATLANTIS, model A11-820A-DP) and I get the error message "No video stream defined".
I have tested the stream of the camera individually by launching omxplayer and it works fine:
`omxplayer --win 0,35,640,343 rtsp://admin:pwd@IPaddr:554/ch01.264 --avdict rtsp_transport:tcp`
I have tested the stream of the camera individually with Videolan and it works fine:
`rtsp://admin:pwd@IPaddr:554/ch01.264`

If I type the corresponding url in /etc/rpisurv I get a "No video stream defined" error:
        # ATLANTIS - TAVERNA
        - url: "rtsp://admin:pwd@IPaddr:554/ch01.264"
          rtsp_over_tcp: True

Have you already experienced a similar issue or behaviour? I thank you very much and in advance for your help. (I hope this is the right place to post this request for support and if not I humbly apologise).

Ciao,
Roberto

SvenVD, 2018-10-10:
Hi Roberto, glad you like rpisurv.
Can you show me the log of rpisurv in /var/log/daemon.log and the main.log in /usr/local/bin/rpisurv/logs/

Thanks

RobertoA109, 2018-10-10:
Hi Sven,
thank you for reacting so fast, that is really cool.

I have attached the files you asked and my /boot/config.txt as well.

I have noticed that in the log file there is an ERROR message related to the ATLANTIS camera (IPaddr 10.0.220.110):

ERROR - CameraStream: screen1_cam_stream2 rtsp://<hidden_username>:<hidden_password>@10.0.220.110:554/ch01.264 Not Connectable (failed rtsp validation, no response from rtsp server)

This is strange. Looks like the camera does not accept the rtsp request.

I look forward to hear your advice, take your time and thank you in advance.

Ciao,
Roberto

--------------------

# For more options and information see
# http://rpf.io/configtxtreadme
# Some settings may impact device functionality. See link above for details

# DISPLAY_DEFAULT_LCD
# If the raspberry Pi DSI LCD is detected it will be used as the default display and will show the frame buffer.
# Setting display_default_lcd=0 will ensure the LCD is not the default display, which usually implies the HDMI output will be the default.
# The LCD can still be used by choosing its display number from supported applications, for example omxplayer.
display_default_lcd=0

# LCD_ROTATE
# This flips the display using the LCD's inbuilt flip functionality, which is a cheaper operation that using the GPU-based rotate operation
# For example, lcd_rotate=2 will compensate for an upside down display
#lcd_rotate=2

# HDMI_FORCE_HOTPLUG
# Setting hdmi_force_hotplug to 1 pretends that the HDMI hotplug signal is asserted, so it appears that a HDMI display is attached.
# in other words, HDMI output mode will be used, even if no HDMI monitor is detected.
hdmi_force_hotplug=1

# HDMI_IGNORE_HOTPLUG
# Setting hdmi_ignore_hotplug to 1 pretends that the HDMI hotplug signal is not asserted, so it appears that a HDMI display is not attached.
# In other words, composite output mode output mode will be used, even if an HDMI monitor is detected.
#hdmi_ignore_hotplug=1

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c_arm=on
#dtparam=i2s=on
dtparam=spi=on

# Uncomment this to enable the lirc-rpi module
#dtoverlay=lirc-rpi

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on
start_x=0
# Changed from 256 to 512 for RpiSurv (2018-08-05)
gpu_mem=512

# Enable 1-wire
dtoverlay=w1-gpio

RobertoA109, 2018-10-11:
Hi Sven,
thank you for your investigation and analysis of the issue I have with the ATLANTIS camera.
Tonight I will make a bunch of tests with this only camera connected to RPISURV and I will diagnose the network and the firewall settings.
I will report back to you.

Enjoy your day and ciao,
Roberto

RobertoA109, 2018-10-11:
Hello Sven,
here is what I have done an tried:

1. I have commented out all video streams and left only the ATLANTIS one and selected 1 column in the config (/etc/rpisurv)

2. I can ping the camera from an ssh session running in the RPi running rpisurv

3. The RPi running rpisurv is in the same VLAN as the ATLANTIS camera

4. No firewall rules defined to block communications between devices within this same VLAN. There is a rule that blocks IPcams in this VLAN from going outside this VLAN.

If I stop rpisurv (systemctl stop risurv) and I try to access the url by specifying  user and pwd:

5a. with omxplayer I can see the video stream of the camera by specifying user and pwd:

pi@RPI-UPS:~ $ omxplayer --win 0,0,640,343 --avdict rtsp_transport:tcp rtsp://admin:pwd@10.0.220.112:554/ch01.264
Video codec omx-h264 width 1920 height 1080 profile 100 fps 25.000000
Subtitle count: 0, state: off, index: 1, delay: 0
V:PortSettingsChanged: 1920x1080@25.00 interlace:0 deinterlace:0 anaglyph:0 par:1.00 display:0 layer:0 alpha:255 aspectMode:0

5b. If I open Videolan I can see the video stream of the camera by opening the network with:

rtsp://admin:pwd@10.0.220.112:554/ch01.264

6. If i do not specify  user and pwd I noticed that omxplayer works and Videolan not (Connection failed, VLC could not connect to “10.0.220.112:554”):

pi@RPI-UPS:~ $ omxplayer --win 0,0,640,343 --avdict rtsp_transport:tcp rtsp://10.0.220.112:554/ch01.264
Video codec omx-h264 width 1920 height 1080 profile 100 fps 25.000000
Subtitle count: 0, state: off, index: 1, delay: 0
V:PortSettingsChanged: 1920x1080@25.00 interlace:0 deinterlace:0 anaglyph:0 par:1.00 display:0 layer:0 alpha:255 aspectMode:0

7. I have taken another ATLANTIS camera and I get the same issues.

What I cannot figure out is why omxplayer can connect to the url of the camera and rpisurv cannot.
rpisurv is using omxplayer as well so where is the difference?

Could it be that using omxplayer dbus makes such a huge difference?

The network settings of the ATLANTIS camera are the following:

I have unchecked both RTSP validation to see if this could ease the access to the video stream but no change.

Let’s see, maybe you will come up with some other ideas and potential solutions.

I thank you very much for supporting me in this.

Ciao,
Roberto

-------------------

SvenVD, 2018-10-11:
A couple of things you can try:

1. Disable probing (      disable_probing_for_all_streams: False, see example config file) and see if rpisurv works. If this is the case then we have a problem with probing and we need to investigate more
2. If 1. is still not working then try running omxplayer with the defaults options that rpisurv uses
 --video_fifo 1 --video_queue 1 --live --timeout 60 --aidx -1 -o hdmi --threshold 0 
If this is not working then we have a problem with one of the omxplayer options and your camera
2. Try running omxplayer with the default options but force over --avdict rtsp_transport:tcp and report on result

RobertoA109, 2018-10-11:
Hi Sven,
here are the results:

1. In the file /etc/rpisurv i have disabled probing, one screen defined with only the ATLANTIS camera and it works !!!
disable_probing_for_all_streams: True

1a. I have removed all comments to the defined urls (4 screens of 4 cameras each, in a 4x4 grid) then the ATLANTIS camera displays again “No video stream defined”

1b. If I activate only one screen with 2 or 4 cameras, 2 columns it works !!!

1c. If I activate two (or more) screens with 4 cameras each, 2 columns, then the ATLANTIS camera displays again “No video stream defined”

2. I have tried omxplayer with your default parameters and it works fine:

pi@RPI-UPS:~ $ omxplayer --win 0,0,640,343 --video_fifo 1 --video_que 1 --live --timeout 60 --aidx -1 -o hdmi --threshold 0 rtsp://admin:pwd@10.0.220.112:554/ch01.264
Video codec omx-h264 width 1920 height 1080 profile 100 fps 25.000000
Subtitle count: 0, state: off, index: 1, delay: 0
V:PortSettingsChanged: 1920x1080@25.00 interlace:0 deinterlace:0 anaglyph:0 par:1.00 display:0 layer:0 alpha:255 aspectMode:0

3. I have tried omxplayer with your default parameters and forced over tcp and it works fine:

pi@RPI-UPS:~ $ omxplayer --win 0,0,640,343 --video_fifo 1 --video_que 1 --live --timeout 60 --aidx -1 -o hdmi --threshold 0 --avdict rtsp_transport:tcp rtsp://admin:pwd@10.0.220.112:554/ch01.264
Video codec omx-h264 width 1920 height 1080 profile 100 fps 25.000000
Subtitle count: 0, state: off, index: 1, delay: 0
V:PortSettingsChanged: 1920x1080@25.00 interlace:0 deinterlace:0 anaglyph:0 par:1.00 display:0 layer:0 alpha:255 aspectMode:0 

I hope that these details will help you further in the analysis.
I am already super happy to see that we are getting closer to the fix.

Ciao,
Roberto

---------------------

SvenVD, 2018-10-11:
disable_probing_for_all_streams: True is a per screen configuration. Did you configure it for all screens?

My guess now is that you camera model does not reply to the RTSP OPTIONS for some reason. Is it possible for you to send me a networkdump i.e. pcap for one camera while you are testing one stream with omxplayer?

RobertoA109, 2018-10-11:
Hi Sven,
I added only one instance of "disable_probing_for_all_streams: True” , sorry.
I will add it for each screen and report.

I will try to create the networkdump, never done it before.

Ciao,
Roberto

---------------

RobertoA109, 2018-10-11:
Hi Sven,

I have now added at the end of each screen definition the following block:
      disable_probing_for_all_streams: True
      nr_of_columns: "2"
      autostretch: False
      duration: 30

Now all 4 screens 4x4 are working fine. Thanks and sorry for the mistake in my config file.

I have to understand how to make a networkdump for you.
I will report as soon as I manage to so it.

Ciao,
Roberto

---------------

Hi Sven,
I added only one instance of "disable_probing_for_all_streams: True” , sorry.
I will add it for each screen and report.

I will try to create the networkdump, never done it before.

Ciao,
Roberto

---------------

RobertoA109, 2018-10-11:
Hi Sven,
I managed to do something with Wireshark and I hope that I have done it like you was expecting (see attached .pcap file).

I have launched omxplayer with your defaults:

pi@RPI-UPS:~ $ omxplayer --win 0,0,640,343 --video_fifo 1 --video_queue 1 --live --timeout 60 --aidx -1 -o hdmi --threshold 0 --avdict rtsp_transport:tcp rtsp://admin:pwd@10.0.220.112:554/ch01.264 <rtsp://admin:pwd@10.0.220.112:554/ch01.264>

Video codec omx-h264 width 1280 height 720 profile 100 fps 25.000000
Subtitle count: 0, state: off, index: 1, delay: 0
V:PortSettingsChanged: 1280x720@25.00 interlace:0 deinterlace:0 anaglyph:0 par:1.00 display:0 layer:0 alpha:255 aspectMode:0

If the logging with Wireshark has not been made correctly please tell me what you expect and tomorrow I will learn more in order to do better.

Thank you Sven and ciao,
Roberto

-------------------

SvenVD, 2018-10-12:
Hi Roberto, I do not see any attachments.I think  github  requires that you  zip it first

RobertoA109, 2018-10-13:
Hi Sven,
ok as soon as I am back home I will zip and send again, Thank you for helping me.
Ciao,
Roberto

SvenVD, 2018-10-14:
Hi Roberto, unfortunately I still can't see any attachment.

RobertoA109, 2018-10-14:
Hi Sven,
oh this is sad, I have copied the files on my dropbox.
Let’s try that, maybe the link is going though:
https://www.dropbox.com/sh/yi0wtsipj5ieeji/AACV3gnqljCcCMdJH294pbeVa?dl=0 <https://www.dropbox.com/sh/yi0wtsipj5ieeji/AACV3gnqljCcCMdJH294pbeVa?dl=0>

The name of the directory is “rpisurv” and it contains 3 files:
Wireshark__ATLANTIS_cam__No_probing_FALSE.pcap
Wireshark__ATLANTIS_cam__No_probing_TRUE.pcap
Wireshark__ATLANTIS_cam__omxplayer_defaults.pcap

Ciao,
Roberto

----------

RobertoA109, 2018-10-15:
> Thanks Roberto! However all the three files on your dropbox have the extension pcap but in reality are html files. So can you upload the real pcap files, thank you

Hi Sven,
I have opened the files I had saved on my machine and tried to SaveAs .
These are the additional files with an underscore at the end.
https://www.dropbox.com/sh/yi0wtsipj5ieeji/AACV3gnqljCcCMdJH294pbeVa?dl=0 <https://www.dropbox.com/sh/yi0wtsipj5ieeji/AACV3gnqljCcCMdJH294pbeVa?dl=0>

I had to mount all 16 cameras on a frame for the configuration/testing and debugging (see file IPcams_test_frame.JPG) and I have to finish the patching of the complete network cabinet before I can make another acquisition with Wireshark.

I fear that these are not the files you expect so please do not let me down.

As soon as I am ready with the network the first thing I will do is the tcp dump so we can continue investigating my issue.

By chance do you have a pcap file that I could see as a reference so I can create them as you need them? This would be of a great help.

Thank you in advance for your help Sven and I will be back soon.

Ciao,
Roberto

---------

SvenVD, 2018-10-16:
You can do something like this:
 tcpdump -i any host <replace with ip of 1 camera> -w /tmp/capture.pcap
to generate a pcap

RobertoA109, 2018-10-16:
Hi Sven,
I have connected a mix of cameras and 3 ATLANTIS for the TCP dump.
I have used your command and it worked like a charm.

10.0.220.110 is the ATLANTIS
10.0.220.33 is the RPISURV

I have attached my etc/rpisurv configuration file as well so you can see that in this temporary quick setup I have 4 screen with 4 cameras.
Some cameras appears in several screens.

The files can be found on my dropbox at:
https://www.dropbox.com/sh/yi0wtsipj5ieeji/AACV3gnqljCcCMdJH294pbeVa?dl=0 <https://www.dropbox.com/sh/yi0wtsipj5ieeji/AACV3gnqljCcCMdJH294pbeVa?dl=0>

As soon as I have all final cameras connected to all switches I will do a TCP dump again.

I hope these files will help you figure out what is different in the ATLANTIS cameras.

It is a pity to face such issues and to ask you to support so much.
I did searched a lot on the internet to find a webcam that was not requiring ActiveX for the setup, that was POE, mini dome and relatively cheap.
The ATLANTIS cameras are well built (aluminum housing), the quality of the image is fine, all in all it is a good product I would say.
I am sure you will come up with something.

Anyway, with probing disables it is already working,  I am not so desperate.
What I understood from your words is that probing is very important for diagnostics and healing so let see.

Good luck Sven and good night.

Ciao,
Roberto

----------

SvenVD, 2018-10-17:
Ho Roberto, thanks for your pcaps, they included the information I was after. I hope I fixed it in https://github.com/SvenVD/rpisurv/commit/97254c5ba19a85e0f675b8cfa0381fbb0b51d8aa

Could you try latest 2.0.beta to check it out?

RobertoA109, 2018-10-19:
Hi Sven,
I have copied the file CameraStream.py (97254c5 <https://github.com/SvenVD/rpisurv/commit/97254c5ba19a85e0f675b8cfa0381fbb0b51d8aa>) from Git to the directory /core/ on my RPi.
I made a systemctl restart rpisurv.

With probing deactivated it works fine.
With probing activated it still displays “No video stream defined”.

I have created a pcap for you that I have uploaded in my DropBox:
https://www.dropbox.com/sh/yi0wtsipj5ieeji/AACV3gnqljCcCMdJH294pbeVa?dl=0 <https://www.dropbox.com/sh/yi0wtsipj5ieeji/AACV3gnqljCcCMdJH294pbeVa?dl=0>

I was wondering, could it be that I have a settings in the IPcam that is causing this?
MainStream and SubStream?
Would it be useful for you to have a look at the parameters and settings of the camera?
I could create a sequence of snapshots and put them in a multi-page PDF file and upload it in my DropBox for you to check.

Please do not give up Sven, hope never dies…

Thank you Sven and ciao,
Roberto

-----------

SvenVD, 2018-10-20:
Hi Roberto, the last pcap does not show the fix I included. Are you sure you are using the latest version of rpisurv 2.0 Beta or copied  CameraStream.py and restarted rpisurv?

RobertoA109, 2018-10-20:
Hi Sven,
I just copied the file CameraStream.py from github and restarted rpisurv.
If this is not the right thing I should have done I am sorry that you lost time checking my last pcap.

I will upgrade my rpisurv with the latest version 2.0 Beta and I will report.
Sorry.

Ciao and talk to you soon.
Roberto

------------

RobertoA109, 2018-10-20:
Hi Sven,
I have followed the update instructions you have posted on github and performed the upgrade to the 2.0 beta without overwriting the config files.

How to update 

	• cd rpisurv; git pull
	• run sudo ./install.sh
	• systemctl restart rpisurv

IT WORKS! YES! YES! YES! YES!

This is really great Sven, thank you very much for your support and your commitment to this piece of jewellery.

The terminal log i the following:

pi@RPI-UPS:~/rpisurv $ sudo git pull
Updating 8990190..97254c5
error: Your local changes to the following files would be overwritten by merge:
	surveillance/core/CameraStream.py
Please, commit your changes or stash them before you can merge.
Aborting
pi@RPI-UPS:~/rpisurv $ sudo ./install.sh
Use this installer on your own risk. Make sure this host does not contain important data and is replacable
This installer will disable graphical login on your pi, please revert with the raspi-config command if needed.

The following version will be installed: "2.0.beta7"

By using this software, you agree that by default limited and non-sensitive (runtime, unique id and version) stats
will be sent on a regular interval to a collector server over an encrypted connection.
You can disable this anytime by changing the update_stats: config option to False.
This has been introduced to get an idea of how much users are testing a beta version of the software.
Once the software comes out of beta, stats sending will be disabled by default.

Do you want to continue press <Enter>, <Ctrl-C> to cancel

Get:1 http://mirrordirector.raspbian.org jessie InRelease [14.9 kB]
Get:2 http://mirrordirector.raspbian.org jessie/main armhf Packages [9537 kB]     
Hit https://deb.nodesource.com jessie InRelease                                          
Get:3 http://archive.raspberrypi.org jessie InRelease [22.9 kB]                                                   
Hit https://deb.nodesource.com jessie/main Sources             
Hit https://deb.nodesource.com jessie/main armhf Packages                         
Get:4 https://deb.nodesource.com jessie/main Translation-en [162 B]                                                     
Get:5 http://archive.raspberrypi.org jessie/main armhf Packages [171 kB]                                                
Get:6 https://deb.nodesource.com jessie/main Translation-en [162 B]                                                     
Get:7 https://deb.nodesource.com jessie/main Translation-en [162 B]                                                     
Get:8 https://deb.nodesource.com jessie/main Translation-en [162 B]                                                     
Get:9 https://deb.nodesource.com jessie/main Translation-en [162 B]                                                     
Ign https://deb.nodesource.com jessie/main Translation-en                                                               
Hit http://archive.raspberrypi.org jessie/ui armhf Packages                                                             
Ign http://archive.raspberrypi.org jessie/main Translation-en                                                           
Ign http://archive.raspberrypi.org jessie/ui Translation-en                                                             
Get:10 http://mirrordirector.raspbian.org jessie/main armhf Packages [9537 kB]                                          
Get:11 http://mirrordirector.raspbian.org jessie/contrib armhf Packages [43.3 kB]                                       
Get:12 http://mirrordirector.raspbian.org jessie/non-free armhf Packages [84.2 kB]                                      
Get:13 http://mirrordirector.raspbian.org jessie/rpi armhf Packages [1356 B]                                            
Ign http://mirrordirector.raspbian.org jessie/contrib Translation-en                                                    
Ign http://mirrordirector.raspbian.org jessie/main Translation-en                                                       
Ign http://mirrordirector.raspbian.org jessie/non-free Translation-en                                                   
Ign http://mirrordirector.raspbian.org jessie/rpi Translation-en                                                        
Fetched 4792 kB in 6min 55s (11.5 kB/s)                                                                                 
Reading package lists... Done
Reading package lists... Done
Building dependency tree       
Reading state information... Done
coreutils is already the newest version.
openssl is already the newest version.
procps is already the newest version.
python is already the newest version.
python-dbus is already the newest version.
python-openssl is already the newest version.
[... 19 lines cut ...]
core/util/
images/blank.png
images/connecting.png
images/noconnectable.png
images/placeholder.png

sent 165,880 bytes  received 170 bytes  332,100.00 bytes/sec
total size is 215,666  speedup is 1.30
'/tmp/surveillance.yml.20181021_1540074382' -> '/usr/local/bin/rpisurv/conf/surveillance.yml'
'rpisurv' -> '/usr/bin/rpisurv'
'rpisurv.service' -> '/etc/systemd/system/rpisurv.service'
pi@RPI-UPS:~/rpisurv $ sudo systemctl restart rpisurv

The “sudo git pull” command provides some error message but I am not sure if this is a real problem or not. Maybe it is linked to the fact that I have copied the CameraStream.py file yesterday from the beta version.

I have uploaded on my DropBox a last pcap file with the system working after the upgrade "to 2.0 beta” in case you want to cross-check that it is behaving like you expected:
https://www.dropbox.com/sh/yi0wtsipj5ieeji/AACV3gnqljCcCMdJH294pbeVa?dl=0 <https://www.dropbox.com/sh/yi0wtsipj5ieeji/AACV3gnqljCcCMdJH294pbeVa?dl=0>

As soon as I will have the full system working with all 16 cameras and the two rpisurv stations I will report back to you with text and pictures.
How do I do to report? Do I have to open a ticket in github?

Have a good night and ciao and thank you once again for all you did.
Roberto

———————

closed 2018-10-22 by SvenVD

colinyarwood, 2018-10-23:
Sven

I have now tried several times to mute/unsubscribe  from your mails.

Please do me a favour and delete my email… even doing it on Github seems not to work.

Colin