SvenVD/rpisurv #78: Picture Issue

burt-macklin1, 2018-10-27:
Soi have the program working the only have an issue with the picture, i have attached links to what the picture does, not sure what causes this, I have given the GPU more ram and less ram and seems to not have any effect on this at all. Not sure if anyone knows what causes this? Other than that it works amazingly.
[image]
[image]

RobertoA109, 2018-10-27:
Hi,
I think that to avoid the smearing effect at the bottom of your video stream you could try to add:
rtsp_over_tcp: True

immediately below the line where you define the url of your camera:
- url: "rtsp://user:pwd@ipaddr:554/ch01_sub.264"
  rtsp_over_tcp: True

At least this is how I managed to solve this similar issue.

Ciao,
Roberto

-------------

 <https://user-images.githubusercontent.com/44486418/47607039-b48f6c00-d9e0-11e8-88d8-0bf51bf8421b.jpg>
 <https://user-images.githubusercontent.com/44486418/47607040-b6592f80-d9e0-11e8-9033-9696959545cf.jpg>

burt-macklin1, 2018-10-28:
Thank you @RobertoA109 That seems to have fixed the issue.

closed 2018-10-28 by burt-macklin1