SvenVD/rpisurv #79: Failing to start on my fresh raspian

ozett, 2018-11-11:
i did a fresh raspian-install on a new sdcard,
cloned the repository today, like told on the github homepage,
than it failed to start.

if i call it as root directly, it got an errormessage. see below
any hints to fix it? (update tried, said it is the newest one)

```
pi@raspberrypi:~ $ sudo -i
root@raspberrypi:~# /usr/bin/rpisurv
Traceback (most recent call last):
  File "surveillance.py", line 10, in <module>
    from config import cfg
  File "/usr/local/bin/rpisurv/config.py", line 4, in <module>
    cfg = yaml.load(ymlfile)
  File "/usr/lib/python2.7/dist-packages/yaml/__init__.py", line 71, in load
    return loader.get_single_data()
  File "/usr/lib/python2.7/dist-packages/yaml/constructor.py", line 37, in get_single_data
    node = self.get_single_node()
  File "/usr/lib/python2.7/dist-packages/yaml/composer.py", line 36, in get_single_node
    document = self.compose_document()
  File "/usr/lib/python2.7/dist-packages/yaml/composer.py", line 55, in compose_document
    node = self.compose_node(None, None)
  File "/usr/lib/python2.7/dist-packages/yaml/composer.py", line 84, in compose_node
    node = self.compose_mapping_node(anchor)
  File "/usr/lib/python2.7/dist-packages/yaml/composer.py", line 133, in compose_mapping_node
    item_value = self.compose_node(node, item_key)
  File "/usr/lib/python2.7/dist-packages/yaml/composer.py", line 84, in compose_node
    node = self.compose_mapping_node(anchor)
  File "/usr/lib/python2.7/dist-packages/yaml/composer.py", line 133, in compose_mapping_node
    item_value = self.compose_node(node, item_key)
  File "/usr/lib/python2.7/dist-packages/yaml/composer.py", line 84, in compose_node
    node = self.compose_mapping_node(anchor)
  File "/usr/lib/python2.7/dist-packages/yaml/composer.py", line 127, in compose_mapping_node
    while not self.check_event(MappingEndEvent):
  File "/usr/lib/python2.7/dist-packages/yaml/parser.py", line 98, in check_event
    self.current_event = self.state()
  File "/usr/lib/python2.7/dist-packages/yaml/parser.py", line 428, in parse_block_mapping_key
    if self.check_token(KeyToken):
  File "/usr/lib/python2.7/dist-packages/yaml/scanner.py", line 116, in check_token
    self.fetch_more_tokens()
  File "/usr/lib/python2.7/dist-packages/yaml/scanner.py", line 257, in fetch_more_tokens
    % ch.encode('utf-8'), self.get_mark())
yaml.scanner.ScannerError: while scanning for the next token
found character '\t' that cannot start any token
  in "conf/surveillance.yml", line 15, column 1
root@raspberrypi:~#
```

ozett, 2018-11-11:
i am a bit lost,
after failure i thought of trying v2beta an did the cmd-line like on homepage
in the subfolder of rpisurv

```git checkout v2.0_branch```

it changed nothing, i got still the error, it fails to start

```
pi@raspberrypi:~/rpisurv $ git checkout v2.0_branch
Branch v2.0_branch konfiguriert zum Folgen von Remote-Branch v2.0_branch von origin.
Zu neuem Branch 'v2.0_branch' gewechselt
pi@raspberrypi:~/rpisurv $ git pull
Bereits aktuell.
pi@raspberrypi:~/rpisurv $
```
any help with this? would like to see my streams running 
😄

SvenVD, 2018-11-11:
Hi, you can not use tabs in the config file, this is not yaml compatible. Please use only spaces.
The error states "found character '\t' that cannot start any token"

If you want to use v2.0 use the steps as outlined in the 2.0 README. After checkout v2.0 you should run the install.sh script again.

closed 2018-11-28 by SvenVD