SvenVD/rpisurv #80: config options not clear

ozett, 2018-11-11:
as i cloned the repository today,
the default config in /etc/rpisruv has option like:
https://raw.githubusercontent.com/SvenVD/rpisurv/master/surveillance/conf/surveillance.yml

```- rtsp_url: "path"```

if i follow the github link i land on  v20 
https://raw.githubusercontent.com/SvenVD/rpisurv/v2.0_branch/surveillance/conf/surveillance.yml
the options has changed now to:

``` - url: "path"```

i see there is versioning, but how can i change versions.
if i check it out like told on the doks, i always have the same version, or wrong config?

1)
how to edit config? identation? colons? what to delete, what not?
seems difficult....
some hints?

ozett, 2018-11-11:
why do i need the dashes in the config-file? 
python or something? seems strange to me

SvenVD, 2018-11-11:
I created an example with a very minimal config for you (replace the examples with your config!). If you do not need to optional config it probably works with this option.
[Example one screen and 4 example! streams for 2.0](https://gist.github.com/SvenVD/f744af9298bce65c7eaaba96a991d36e)

closed 2018-11-28 by SvenVD