SvenVD/rpisurv #81: rpisurv can't start properly after power cut

Neo-engineering, 2018-11-13:
At first: Thank you for nice job!

I am working quite while in video Surveillance field, so few comments from my experience. I do not like rpisurv 2.0, it is slower (if use just one screen and to compare with rpisurv 1.0) and functionality is not practical.

But rpisurv 1.0 is good product. Just found one problem:
If we cut power to Raspberry Pi (not proper shutdown) and then reconnect power, we can see single "Connecting" screen for 25s. If we restart rpisurv (for example via xrdp) straight  all cameras comes on instantly. And more: if I use “keep_first_screen_layout: True” then never will get any image if Raspberry Pi is booting after power cut.
We are using Raspberry Pi 3 Model B+, newly installed OS.
It is looks like something is not ready when rpisurv starts.
Temporary: Maybe you can advise where is command to start rpisurv when raspberry Pi is booting, so I look for option to delay of starting of rpisurv.

One suggestion: to have fixed grid from beginning, with fixed cameras places, no matter cameras was connected or not, for example:
.__ __ __ __ __ __ __ __ _ .
l . . . . . . . . . l . . . . . . . . . l
l. . . . .1 . . . l . . . . 2 . . . .l
l __ __ __ __ __ __ __ __  l
l . . . . . . . . . l . . . . . . . . . l
l. . . . .3 . . . l . . . . 4 . . . .l
l __ __ __ __ __ __ __ __  l

for example by using attribute "fixed_grid" false or true.
And maybe add additional “position” attribute to the cameras settings in config file, for example:

rtsp_url: "rtsp://<user>:<password>@<ip or dnsname>:<port>/play1.sdp"
position: 1

If is no attribute used, then allocate any first available place.

SvenVD, 2018-11-14:
Hi good suggestion with the fixed_grid and the position parameter. Feature requests can be requested and voted for [here](https://feathub.com/SvenVD/rpisurv)

As for rpisurv 2.0 slower then rpisurv 1.0 for one screen, can you define slower? What do you mean by slower? It is a little slower on startup because it needs to connect to dbus to make rotation of screens and keyboard possible.

For your problem:
I power cut my raspberry pi all the time, that is how I shut it down :-), I do not have the issues you are describing. So I can not reproduce your issue. Are you using WIFI? Rpisurv2.0 has some minor dependency improvements on network at startup.
You can restart rpisurv on command line with `systemctl restart rpisurv` and the systemd unit files is located at /etc/systemd/system/rpisurv.service. Anyhow, even if the network comes up later rpisurv should self-heal and show the cameras. What is the output of /usr/local/bin/rpisurv/logs/main.log

SvenVD, 2018-11-16:
1. if keep_first_screen_layout is true you are essentially disabling rpisurv autohealing as stated in the example config. Rpisurv could not detect any cameras at startup so it is "keeping it's first screen layout" which is 0 streams.

2. Autohealing is enabled, it will keep on checking for connection and adjusting the screen properly.
If you want faster detection you can change config option "interval_check_status":  which defaults to 25 seconds

3. "Choose whether to wait for network connection during boot" blocks startup until dhcp has finished receiving an ip, another option would be to configure a static ip, or just live with option 2.

I suggest you move to rpisurv 2.0 beta though as 1.0 will not receive any updates anymore when 2.0 comes out of beta. Everything you described can be done with 2.0 also.

closed 2018-11-28 by SvenVD

Neo-engineering, 2018-12-03:
Sorry do not had time to reply…work is keeping me busy :(
1. Regarding Problem: it is option to set static ip – this also solution. But I think is good idea (this is another idea to add to project): rpisurv must wait for network, to start search for cameras.
2. Regarding “rpisurv 2.0, it is slower”. It is annoying slow appearance of the cameras  on screen (one-by-one) :) , but also “image freeze”, please find attached video files (sorry for quality).
Also want to ask: if to use just manual screens switching (by using F buttons), why is delay around 20s? disable_autorotation: True

[rpisurv.zip](https://github.com/SvenVD/rpisurv/files/2639705/rpisurv.zip)