SvenVD/rpisurv #82: Stream from NVR

musicman251, 2018-11-27:
I am trying to stream from cameras that are connected to an NVR. I have tested the rtsp url in VLC and it opens successfully but when I put the same URL into the RPISURV confing it fails to open. I simply get a blank black screen every few seconds. 

NVR WAN port is connected to same switch as Raspberry Pi. Commented all URLs expect for:
- url: "rtsp://admin:xxxxxx@192.168.2.250:554/cam/realmonitor?channel=1&subtype=1"

burt-macklin1, 2018-11-27:
I can send you my script when I get home, mine is connected to a NVR as
well and took me a while to set up but got it. What brand nvr and cameras
donyou have?

musicman251, 2018-11-27:
They are Lorex NVR and Camera.

burt-macklin1, 2018-11-28:
### From the sounds of it you had the same issue as i did. Try this code and replace the ips to yours and see if that works. also set your NVR to a static IP. let me know if this works or not and ill try to help you, took me 3 days to get the same issue fixed.

#THIS IS A YAML FILE, INDENTATION IS IMPORTANT. ALSO DO NOT USE TABS FOR INDENTATION, BUT USE SPACES
#THE FOLLOWING ARE EXAMPLES WITH ALL OPTIONS EXPLAINED, REMOVE OR ADAPT THEM WITH YOUR REAL CONFIG

essentials:
  #rpisurv will rotate over the following screens in the order they are configured here, if you only define one screen no rotation will happen

  #If you enable this option no auto rotation will happen, but you will still be able to select a screen by pressing and holding a Function Key (F1-F12)
  #Defaults to False
  #disable_autorotation: True

  screens:
      #Start config for example screen one, this is the minimal config. This will show a 2x2 if all streams are connectable,
      #otherwise rpisurv, by default, rearranges the screen to only show the connectable streams. To disable the latter set disable_probing_for_all_streams=True
      #Force this screen to be shown by pressing and holding F1 or keypad 0
    - camera_streams:
        - url: "rtsp://user:XXXX@192.168.1.60:554/cam/realmonitor?channel=2&subtype=1"
          rtsp_over_tcp: True
        - url: "rtsp://user:XXXX@192.168.1.60:554/cam/realmonitor?channel=6&subtype=1"
          rtsp_over_tcp: True
        - url: "rtsp://user:XXXX@192.168.1.60:554/cam/realmonitor?channel=3&subtype=1"
          rtsp_over_tcp: True
        - url: "rtsp://user:XXXX@192.168.1.60:554/cam/realmonitor?channel=1&subtype=1"
          rtsp_over_tcp: True
        - url: "rtsp://user:XXXX@192.168.1.60:554/cam/realmonitor?channel=4&subtype=1"
          rtsp_over_tcp: True
        - url: "rtsp://user:XXXX@192.168.1.60:554/cam/realmonitor?channel=5&subtype=1"
          rtsp_over_tcp: True

      #OPTIONAL AND ADVANCED PER SCREEN OPTIONS, YOU CAN CONFIGURE THIS PER SCREEN, IF NOT CONFIGURED A DEFAULT VALUE WILL BE USED
      #NOT recommendeded to enable disable_probing_for_all_streams, it disables some autohealing functions of rpisurv
      #Also if using multiple screens, enabling this will slow down rotation of one screen to the next if there are unconnectable screens
      #This also disabled rpisurv only drawing the connectable screens using up all pixels. In other words, you will get a placeholder for an unconnectable stream instead of not showing at all
      #Roughly the same as the obsoleted keep_first_screen_layout in rpisurv v1.0
      #Defaults to False
      disable_probing_for_all_streams: False

      #How many columns you want the program to use, it will autocalculate the amount of row needed based on the resolution of your screen
      #Default = 2
      nr_of_columns: "2"

      #Autostretching makes sures all of your pixels will be uses to display streams, so you have a maximum surveillance area.
      #But this will probably destroy the aspect ratio of your last stream.
      #Defaults to False
      autostretch: true

      #This is how long this screen will be shown each rotation.
      #If this is set to a very low value then it can happen that building up the active and/or the cached screen takes longer then this duration,
      #in that case the building and caching needs to finish first before timer will be expired and next screen is shown
      #Defaults to 20
      duration: 60

      #DEFINE AS MANY SCREENS AS YOU WANT
      #When reaching last defined screen, rotate will start at the beginning again
      #You can have as many screens as you want, they are dynamically loaded offscreen and only the next one is loaded, so this is not limited on hardware

SvenVD, 2018-11-30:
Hi, If you can show me the log of rpisurv in /var/log/daemon.log and the main.log in /usr/local/bin/rpisurv/logs/ I probably can say what you are missing.

Neo-engineering, 2018-12-03:
Maybe resolution too high or cameras set to use H.265 compression

musicman251, 2018-12-03:
I will try tonight. I have been away...they are 4K cameras but I’m only streaming the substream
    	
    	

    	Dan d'EonD’Eon Technical Services902-749-6163

musicman251, 2018-12-04:
I ended up starting from scratch and installing the beta version. I was able to make it work using the rtsp_over_tcp: True line as suggested by burt_macklin1. Thanks for your responses!

burt-macklin1, 2018-12-04:
Glad you got it working!

SvenVD, 2018-12-05:
You only have one stream configured. Can you describe what is happening, there is nothing special in the logs except for the fact that for some reason you are using up all GPU. Did you already increase gpu memory?

musicman251, 2018-12-06:
I figured it out. I had a type with a < in front of the username section in the url. Always so simple when you finally catch your mistakes! I now have several screens rotating successfully. Thanks for the great product!

 

Dan d’Eon

Cell: (902) 749-6163

logo and slogan one third size