SvenVD/rpisurv #85: don't want to rearranges.

dreamonhigh, 2019-01-03:
Hello

very nice project. this config is very sensitive ;)  had a lot of problems. but now it works. using 2.0beta
but one little thing I want to ask you.

using 2 cams. but don't want only 2 colums. because pictures are to much streched. don't looks nice. so I thought 2col,2row might be better. tried to config this. so I did a dummy entry like url: "rtsp://". it arranges correct. but after about 10-30sec. it auto arrange and I see 2cams strechted again.

It would be super, if I could set two pictures on the positions that aren't be used. 

tried to use:  → disable_probing_for_all_streams=True
but I think its a syntax fault "=" should be a ":". it didn't change a thing.

probe_timeout: 9999 → neither
 
any idea? 
Thank you very much. good work.

burt-macklin1, 2019-01-03:
Have you tried to disable stretching?

dreamonhigh, 2019-01-03:
stretching was already disabled! I only can enable it. enabling means stretched to 3+1. after 10-30sec it jumps back to in 2picture mode .. would like to have 2colum,2rows but only 2 cam.  maybe auto arrange is possible to disable?

dreamonhigh, 2019-01-03:
thanks for your help

half the vertikal size works. but so I have blank screen on the lower half of the screen.
I would like to put some kind of picture on it. Like you do by saying "Connecting.." It would be great to put some text or better png on it. using 3cams this works fine, cause you added a placeholder.png. here I can put some nice Information on it. but I have only 2cam and it auto arrange to 2cam only. 

therefore my added dummy url. now having three cams and placeholder showings my modified placeholder.png

your right 2cam mode is stupid to use. but to use it as information indicator (png) of what the cams are showing it makes sense. in my case.

maybe you can modify your programm to add a dummy cam that only scales a png/jpg in that position?
by using 2cams it makes sense.

dont you think?
anyway it works. had this smearing problem, too. added a lot of rtsp_over_tcp: true below every url.

SvenVD, 2019-01-03:
Hi,
Great it works.

What you are asking is a new feature which looks a lot like [this feature](https://feathub.com/SvenVD/rpisurv/+9)

It is on the todo list. It is not scheduled to do in the near future, but I am always open for a pull request.

dreamonhigh, 2019-01-04:
This feature is much more ambitious ;)
mine is only a still image adding on positions that are not connected or dummy. no updating would be required. 

So if I could switch this auto arrange thing off,  that happens after loosing a camera in about 15sec. 
I would be a very lucky man ;)

not recommend, I understood: disable_probing_for_all_streams: True
I only want to stop this auto arrange thing. don't know how.

imagine using 3cams and one connection get lost, auto arrage makes it to a 2pic view. I think its better to show two placeholder in 4 pic view instead of switching in 2 pic view with half of screen blank or stretched.

SvenVD, 2019-01-04:
>  mine is only a still image adding on positions that are not connected or dummy. no updating would be required.

Updating would indeed be an extra for that feature. However I still want a generic solution based on the url you provide. Rpisurv should automatically do the correct thing for a local or remote url.

> imagine using 3cams and one connection get lost, auto arrage makes it to a 2pic view
That is rpisurv algorithm default behavior. I for example have cameras that switch on timed based in the evening, I don't want empty spaces all day for those cameras that are down.
The correct solution for rpisurv would be for the user to add to custom placeholder URL, but see first point

A workaround for the time being to show something custom on the bottom row would be to change https://github.com/SvenVD/rpisurv/blob/v2.0_branch/surveillance/images/blank.png image. With the fixed_height solution enabled.

closed 2019-01-08 by dreamonhigh

SvenVD, 2019-04-20:
New features 2.1.0 ( Currently only available on latest branch: https://github.com/SvenVD/rpisurv/tree/v2_latest. Will be merged to master once it has been tested enough)

It is now possible to show an image instead of a camerastream. The image is fetched from a choosen remote url. On top of that this remote url is watched for changes If the image is changed on the remote site, rpisurv will detect this and will also fetch the new image and update it on screen. The polling frequency is controlled by advanced: interval_check_status config option in /etc/rpisurv.conf. The "image stream" follows the same health checks as a normal camera stream. This means if the specified image is unavailable it will not be shown on-screen by default (you can override this). This feature has been requested on link.