SvenVD/rpisurv #86: After adding rtsp_over_tcp: true, service fails to start

rcktboy, 2019-02-16:
Hi,
I have the  "smearing" effect and want to try to solve it by adding
rtsp_over_tcp: true tio the stream url.

But as soon as I do, the pi will show the command and nothing else.
I use the Unifi Protect camera's on the CloudKey G2+.

I added:
- url: "rtsp://<cloudkey ip>:7447/<stream code>
  rtsp_over_tcp: true

When I remote the rtsp line it works fine.
Do I miss something?

Hope you can help.

^Rich

SvenVD, 2019-02-16:
Hi, there is now a new place for support questions: http://community.rpisurv.net/. At first sight you do not miss something but to troubleshoot further:

Please paste the most recent logs of /var/log/daemon.log and /usr/local/bin/rpisurv/logs/main.log

rcktboy, 2019-02-17:
Hi, thanks for your replay.
Might look like that the file is really sensitive.
To make the log you ask for I started the pi, quit rpisurv, started startx and changed the config on there.
Use other computer before.

Behold, it started after adding rtsp_over_tcp.

So I will test some more later today and let you know (on the community) if I keep having the problem.

Thanks for now :)

closed 2019-02-28 by SvenVD