SvenVD/rpisurv #87: All connections say connecting, gpu_memory doesn't help

DJODonohue, 2019-03-06:
So I've got an RPI3B+, and I'm trying to get rpisurv setup and pulling from RTSP from an Amcrest NVR.
The RTSP format for the stream is:

`rtsp://<user>:<pass>@<ip>:<port>/cam/realmonitor?channel=1&subtype=0`

my current **config** file is:

```
#THIS IS A YAML FILE, INDENTATION IS IMPORTANT. ALSO DO NOT USE TABS FOR INDENTATION, BUT USE SPACES
#THE FOLLOWING ARE EXAMPLES WITH ALL OPTIONS EXPLAINED, REMOVE OR ADAPT THEM WITH YOUR REAL CONFIG

essentials:
  screens:
    - camera_streams:
        - url: "rtsp://<user>:<pass>@192.168.1.236:554/cam/realmonitor?channel=1&subtype=1"
        #- url: "rtsp://<user>:<pass>@192.168.1.236:554/cam/realmonitor?channel=2&subtype=0"
        #- url: "rtsp://<user>:<pass>@192.168.1.236:554/cam/realmonitor?channel=3&subtype=0"
        #- url: "rtsp://<user>:<pass>@192.168.1.236:554/cam/realmonitor?channel=4&subtype=0"
          probe_timeout: 10
          rtsp_over_tcp: true
      disable_probing_for_all_streams: False
      nr_of_columns: "2"
      autostretch: False
      duration: 60
#!!!Normal users do not need to edit this section!!!
advanced:
  ##Enable this option if you want to have a fixed width of all your camera streams,
  ##By default rpisurv autocalculates this value, this can cause streams to get "stretched",
  ##if this value exceeds the available width, rpisurv will fallback to autocalculation
  #fixed_width: 500

  ##Enable this option if you want to have a fixed height for all your camera streams,
  ##By default rpisurv autocalculates this value, this can cause streams to get "stretched",
  ##if this value exceeds the available height, rpisurv will fallback to autocalculation
  #fixed_height: 500

  #Rpisurv sends usage stats to it's statistics server to give the rpisurv community an idea how widespread this software is being used
  #No performance impact is measured when using this option, no sensitive data is being sent. All data is anonymised.
  #By default this is false

  #No performance impact is measured when using this option, no sensitive data is being sent. All data is anonymised.
  #By default this is false
  #update_stats: False

  #By default rpisurv will cache the next screen in the rotation (if more than one screen is configured), to speed up rotation events.
  #However, this action takes up some resources of the raspberry pi.
  #If you do not mind a slower rotate event (slower building up of the screen during a rotate event) and you want to free up some resources, then set this to False
  #This will also speed up first start since no cached screen needs to be started at boot time.
  #cache_next_screen: True

  #By default rpisurv checks every 19 seconds since start of rpisurv if it needs to redraw the current running screen
  #interval_check_status: 19

  #By default rpisurv will check memory usage, set to False to skip memory usage check
  #memory_usage_check: True

#These are fallbacks if autodetection fails;
#Normally you do not need to configure these
fallbacks:
  resolution:
    width: "1920"
    height: "1080"
```

I have 4 cameras that I am wanting to stream but have 3 blocked out four the sake of debugging the issue I am currently having

I have tried both subtypes and neither have been able to give an actual output, just the "connecting" image

**/var/log/daemon.log**

[... 147 lines cut ...]

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

gpu_mem_1024
```

I've tried the following _gpu_mem_ configs

```
gpu_mem_1024
gpu_mem=1024
gpu_mem_512
gpu_mem=512
```

And the log still asks for more GPU memory.
I hope I formatted this in a manner that is readable and understandable

... What am i doing wrong?

SvenVD, 2019-03-06:
Hi, the correct place to ask for support is https://community.rpisurv.net

As for you error, omxplayer can not handle the streams you give it to him

Vcodec id unknown: ae

I suggest you try other channels on your cameras on change the streaming codec on your cameras

DJODonohue, 2019-03-06:
It was the codec, Thanks Sven! sorry that I posted this here instead of the community forum! My bad!

SvenVD, 2019-03-06:
No problem, glad you have it resolved

closed 2019-03-06 by SvenVD