SvenVD/rpisurv #90: old omx player, request bundle and config option for use

syadnom, 2019-03-28:
Basically, the existing deb out there for pre-march22/16 do longer works on the newest raspbian because of other dependancies (please correct me if I'm wrong).  This means building omxplayer manually which is a chore.

It would be great to just bundle an exec of the old omxplayer and allow a config option like 'use-old-omx'.

syadnom, 2019-03-28:
For anyone else who finds this thread, the 'fix' to get the old omxplayer running is to add jessie repos so you can install libssl1.0.0, then install the old omxplayer, then run rpi-update and reboot.

SvenVD, 2019-03-28:
Hi I do not really understand what the problem is you are having.

You can just use the latest version of omxplayer in the latest version of raspbian.

I am running

root@raspberrypi:~# omxplayer --version
omxplayer - Commandline multimedia player for the Raspberry Pi
        Build date: Mon, 05 Nov 2018 15:45:07 +0000
        Version   : 061425a [master]
        Repository: https://github.com/popcornmix/omxplayer.git

Note: support questions go here also: http://community.rpisurv.net/ . Github is only for potential bugs

Thanks

syadnom, 2019-03-28:
SvenVD, it's the well known smearing with omxplayer, udp, and certain rtsp streams.

The newest omxplayer from raspbians repos does this.

Unfortunately the newest raspibian updated openssl and the old deb of omxplayer has an explicit openssl1.0.0 which is not installable without a few tricks.

SvenVD, 2019-03-28:
The only well known smearing problem I am aware of can be solved by setting rtsp_over_tcp: true.

Or do you mean another problem?

syadnom, 2019-03-28:
Issue #56  "how to downgrade omxplayer"

installed, set rtsp_over_tcp: true, smearing.  Had to jump through hoops to install old version and the dep openssl1.0.0.  Once installed, restart rpisurv and smearing is gone.

SvenVD, 2019-03-28:
Hi,  pleas re-read the thread

Especially:

> No it is the way around your need an omxplayer version younger then 14 March 2016 to enable rtsp_over_tcp.

So the bad news is you jumped to all the hoops for nothing, and the good news is that you do not need to do this anymore, just use the latest version of omxplayer packaged in raspbian :)

syadnom, 2019-03-28:
root@raspberrypi:~# omxplayer -v
omxplayer - Commandline multimedia player for the Raspberry Pi
        Build date: Mon, 05 Nov 2018 15:45:07 +0000
        Version   : 061425a [master]
        Repository: https://github.com/popcornmix/omxplayer.git

root@raspberrypi:~# lsb_release -a
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 9.8 (stretch)
Release:        9.8
Codename:       stretch

Everything up-to-date.  constant smearing.

Install omxplayer_0.3.7~git20170130~62fb580_armhf.deb

No more smearing.

SvenVD, 2019-03-29:
and  rtsp_over_tcp: true is enabled?

syadnom, 2019-03-29:
- url: "rtsp://x/y"
          rtsp_over_tcp: true

indentation is good.  With that old version it works.  If I remove rtsp_over_tcp: true the smearing comes back, so that confirms the setting is in place.

upgrade to repo version, smearing. 

This is pulling RTSP from Unifi Protect.  It seems a LOT pickier than Unifi-video.

SvenVD, 2019-03-29:
OK, than this is another problem I was not aware of. Probably only with the combination of Unifi Protect and latest version of omxplayer.

I do not agree the old version needs to be added to rpisurv. Rpisurv will use whatever version is the default on the system (KISS). I suggest you automate or document the installation of the old version of omxplayer in a separate git repo, which can be referred to in the README of rpisurv.

But the best solution would be that omxplayer and/or Unify protect do changes so that they are compatible

syadnom, 2019-03-29:
I'll nuke the pi and start fresh and see what happens.

SvenVD, 2019-03-31:
What was the result?

closed 2019-05-11 by SvenVD