SvenVD/rpisurv #92: Lost

fryhc, 2019-07-10:
I have very little programming experience and am using a raspberry pi for the first time. I have two unifi g3 flex cameras that I'm trying to set up using rpisurv. I tried using displaycameras at first but stumbled upon this program and saw that it would auto detect screen size and decided to use this instead. 

I keep getting errors like this whenever I run sudo systemctl status rpisurv in the terminal
  Loaded: loaded (/etc/systemd/system/rpisurv.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Wed 2019-07-10 09:09:42 EDT; 11s ago
  Process: 4485 ExecStart=/usr/bin/rpisurv (code=exited, status=1/FAILURE)
 Main PID: 4485 (code=exited, status=1/FAILURE)

Jul 10 09:09:42 raspberrypi systemd[1]: rpisurv.service: Failed with result 'exit-code'.
~

I cannot get the program to launch, this is my rpisurv.conf file

essentials:
    #These are the rtsp streams you want the program to display
    #The streams will be displayed in the order you present them here
    rtsp_urls:
        - "rtsp://colbyfry:Dare1234@10.30.2.10:7447/5cf69d6390089974f280630e_1"
        rtsp_over_tcp: true
        - "rtsp://colbyfry:Dare1234@10.30.2.10:7447/5cf6884b90089974f280630a_1"
        rtsp_over_tcp: true

    #How many columns you want the program to use, it will autocalculate the amount of row needed based on the resolution of your screen
    nr_of_columns: "2"

#These are fallbacks if autodection fails;
#Normally you do not need to configure these
fallbacks:
    resolution:
        width: "1920"
        height: "1080"

I found this configuration on a forum and this was working for the person who posted it, I replaced their streams with mine but it won't work and I have no idea what to do. Both of my streams work in VLC. Again I have almost no idea what I'm doing and am relying on forums/message board code to keep plugging and chugging other people's code.

SvenVD, 2019-07-10:
Hi, thanks for testing it out.

Support questions should go here: https://community.rpisurv.net/, github is only for bug reporting.

I need  /var/log/daemon.log output to trace back where your error is

SvenVD, 2019-07-10:
By the way your config file seems to be the old format for rpisurv v1. You can install and run rpisurv v1, but I would recommend you to use version 2 (latest) and update your config file to version 2.

Full example config file https://github.com/SvenVD/rpisurv/blob/master/surveillance/conf/surveillance.yml

Minimal example config file https://gist.github.com/SvenVD/f744af9298bce65c7eaaba96a991d36e

fryhc, 2019-07-17:
Hey, sorry for the late reply, I had pushed this project aside briefly.
However, it was definitely just the format of my config file, as soon as I
copied in the example one you sent with my own URLs it worked flawlessly!

SvenVD, 2019-07-17:
Ok great! I will close this ticket

closed 2019-07-17 by SvenVD