SvenVD/rpisurv #93: Draw videos over desktop

Anonym-tsk, 2019-08-20:
I'm looking for stable omxplayer wrapper to draw video from camera over my interface.
I use MagicMirror and i want to draw video over MagicMirror interface.
Is it possible to set coordinates and size for rpisurv?

SvenVD, 2019-08-22:
rpisurv/omxplayer claims all the pixels. So everything from magicmirror will be hidden, rpisurv autocalculates the positions but you can override to some extend on pixel base.

https://community.rpisurv.net is the right place to ask questions

Thanks

Anonym-tsk, 2019-08-23:
@SvenVD could you make this feature in next releases?

SvenVD, 2019-08-23:
Feature requests are handled here: https://community.rpisurv.net. What exactly is it you want.

closed 2019-09-09 by SvenVD