SvenVD/rpisurv #96: After install.sh autostart and fail continuously

Brickabrak, 2019-11-14:
I'm running on a CM3 module, installed via the install script and rebooted. 
I get a blank flash in the command line showing the mouse cursor every 23 seconds. 
this is a sample from the /usr/local/bin/rpisurv/logs/main.log

2019/11/14 18:07:17 - l_default - INFO - Unique id of this installation is 84f1b5c8e69f2f19a8a352f00c8b95e945a0b9781c5c652aef482f4dade257b1
2019/11/14 18:07:40 - l_default - INFO - Starting rpisurv 2.1.2
2019/11/14 18:07:40 - l_default - INFO - Unique id of this installation is 84f1b5c8e69f2f19a8a352f00c8b95e945a0b9781c5c652aef482f4dade257b1
2019/11/14 18:08:03 - l_default - INFO - Starting rpisurv 2.1.2
2019/11/14 18:08:03 - l_default - INFO - Unique id of this installation is 84f1b5c8e69f2f19a8a352f00c8b95e945a0b9781c5c652aef482f4dade257b1
2019/11/14 18:08:25 - l_default - INFO - Starting rpisurv 2.1.2
2019/11/14 18:08:26 - l_default - INFO - Unique id of this installation is 84f1b5c8e69f2f19a8a352f00c8b95e945a0b9781c5c652aef482f4dade257b1
2019/11/14 18:08:48 - l_default - INFO - Starting rpisurv 2.1.2
2019/11/14 18:08:48 - l_default - INFO - Unique id of this installation is 84f1b5c8e69f2f19a8a352f00c8b95e945a0b9781c5c652aef482f4dade257b1
2019/11/14 18:09:11 - l_default - INFO - Starting rpisurv 2.1.2
2019/11/14 18:09:11 - l_default - INFO - Unique id of this installation is 84f1b5c8e69f2f19a8a352f00c8b95e945a0b9781c5c652aef482f4dade257b1
2019/11/14 18:09:34 - l_default - INFO - Starting rpisurv 2.1.2
2019/11/14 18:09:34 - l_default - INFO - Unique id of this installation is 84f1b5c8e69f2f19a8a352f00c8b95e945a0b9781c5c652aef482f4dade257b1
2019/11/14 18:10:00 - l_default - INFO - Starting rpisurv 2.1.2
2019/11/14 18:10:00 - l_default - INFO - Unique id of this installation is 84f1b5c8e69f2f19a8a352f00c8b95e945a0b9781c5c652aef482f4dade257b1
2019/11/14 18:10:23 - l_default - INFO - Starting rpisurv 2.1.2
2019/11/14 18:10:23 - l_default - INFO - Unique id of this installation is 84f1b5c8e69f2f19a8a352f00c8b95e945a0b9781c5c652aef482f4dade257b1

This is using the sample /etc/rpisurv.conf but adding my streams doesn't change the output. 
I've tried still images, RTSP streams, and TCP streams with the same results. Also tried changing # of columns to just test a 1x1. 
I can play these streams using omxplayer without issue. 

Any advice or help would be appreciated?

SvenVD, 2019-11-16:
Hi, most likely you made an error in your config file.

To support you further we need some additional information:

Support questions are handled here: https://community.rpisurv.net. 
Please provide the [logs](https://www.tapatalk.com/groups/rpisurv/guidelines-t13.html) to assist you further.

Brickabrak, 2019-11-18:
Hi Sven, 

I thought the snippet from the logs would be enough, but here are the logs from the past 4 days. 
I tried again using the example config file from here
https://raw.githubusercontent.com/SvenVD/rpisurv/master/surveillance/conf/surveillance.yml
after rebooting had the same result. 

[main.log](https://github.com/SvenVD/rpisurv/files/3859591/main.log)
[main_1.log](https://github.com/SvenVD/rpisurv/files/3859599/main_1.log)
[main_2.log](https://github.com/SvenVD/rpisurv/files/3859600/main_2.log)

The streams I'm using are some traffic cams I found on the internet
"http://24.197.59.198/mjpg/video.mjpg"
"http://69.9.219.203:6002/mjpg/video.mjpg"
"http://166.169.251.203/mjpg/video.mjpg"
"http://166.140.164.188/mjpg/video.mjpg"
"http://96.57.146.155/mjpg/video.mjpg"
"http://208.139.200.133/mjpg/video.mjpg"

SvenVD, 2019-11-18:
I also need the  /var/log/daemon.log

SvenVD, 2019-11-18:
Continue in https://www.tapatalk.com/groups/rpisurv/rpisurv-repeatedly-starts-and-stops-t51.html#p211

closed 2019-11-18 by SvenVD