SvenVD/rpisurv #101: Camera Feed "Rotation"

IkaSGC, 2020-08-25:
Hi, I'd like to ask. I'm thinking about how to rotate the cameras. I use "displaycameras" at home, link:

https://github.com/Anonymousdog/displaycameras

Its creator referred me with my question to you. I would like to use a layout for 6 cameras. In reality, however, use five cameras in reduced windows. In the enlarged window, these five cameras should then alternate. Basically, an enlarged window would work like a magnifier. Can't think of a way to do it? I'm new to programming.

SvenVD, 2020-08-25:
Hi,

Questions are asked on https://community.rpisurv.net

rpisurv can do rotations of "screens". However all streams on a "screen" are of the same size. You can rotate between a screen with 5 streams -> stream1 full screen -> stream2 full screen -> stream3 full screen -> stream4 full screen -> stream5 full screen ->  screen with 5 streams

That maybe somewhat solves what your want to achieve.

SvenVD, 2020-09-12:
Did this solve your issue?

IkaSGC, 2020-09-13:
> Komu: SvenVD/rpisurv <rpisurv@noreply.github.com>
> "

Hi, 
I still haven't gotten to it. One camera broke down, so I'm waiting. Anyway,
I'm not sure if I was well understood about rotation. 
I have cameras from Unifi, classically they can be watched via RTSP. 

I would like to invent a rotation according to this diagram:

---------- Původní e-mail ----------

(https://github.com/SvenVD/rpisurv/issues/101#issuecomment-691482785), or 
unsubscribe
"

SvenVD, 2020-09-13:
I can't see your diagram.

IkaSGC, 2020-09-14:
In attachement.

---------- Původní zpráva ----------
Od: SvenVD 
Datum: 13. 9. 2020 v 22:28:14
Předmět: Re: [SvenVD/rpisurv] Camera Feed "Rotation" (#101)

I can't see your diagram.

—
(https://github.com/SvenVD/rpisurv/issues/101#issuecomment-691720931), or 
unsubscribe

SvenVD, 2020-09-14:
Sorry, still can't see your diagram. Maybe because via email it doesnt get attached?

IkaSGC, 2020-09-15:
Yes, i see, problem is in github. I send picture directly from email client, but githup delete it. 
[image]

SvenVD, 2020-09-15:
Yes that is what I read from your first post and the answer is
Rpisurv can't directly  do what you are asking but as a workaround maybe you can use rotation of screens
rpisurv can do rotations of "screens". However all streams on a "screen" are of the same size. You can rotate between a screen with 5 streams -> stream1 full screen -> stream2 full screen -> stream3 full screen -> stream4 full screen -> stream5 full screen -> screen with 5 streams again

SvenVD, 2020-10-13:
I just released 2.1.9 on the v2latest branch https://github.com/SvenVD/rpisurv/blob/v2_latest/RELEASE_NOTES.md. This release allows you to create custom layouts.

closed 2020-10-13 by SvenVD