SvenVD/rpisurv #103: RPISurv - 3 x 3 Grid Issue

tastydoosh, 2020-10-12:
Hi guys,

I've been using RPISurv as the basis for pulling RTSP streams from Unifi NVR's for over a year now across around 45 sites. Recently we found out that that old style NVR's are being discontinued and will only be supported for the unifi protect machines.

We've replaced the existing NVR's at a few sites and attempted to just replace the RTSP urls in the YAML conf file, however upon doing this we've noticed "smearing", on all cameras except the bottom right. This is on a 3 x 3 Grid on a 1080p monitor which worked fine before.

I found out that by appending RTSP_OVER_TCP: True under each url, it removes the smearing effect, however if I watch the logs when I start the service I get the following error message:

`Oct 12 15:18:13 Rasberryrpisurv[15913]: 2020/10/12 15:18:13 - l_default - ERROR - Screen: Total sum of advanced fixed_height (nr rows * fixed_height) option (1800.0) is more then available height (1080), falling back to autocalculated height: 216
`
This then shows all 9 cameras, but stretched in a grid that's 2 rows across and 5 down, distorting the feeds.

Here is a link to the yaml file, which I've checked as valid using yamlint. - https://pastebin.com/DQWwn66W

I've already specified the fixed heights of 640x360 for the feeds, so in my mind this should make 9 feeds in a 3x3 grid as before, without stretching or changing aspect ratio. Any help much appreciated!

Cheers

SvenVD, 2020-10-12:
Hi, the standard support channel is on https://community.rpisurv.net. This is for bug reports only.

The nr_of_columns entry has a wrong identation, it should go below ` - camera_streams` as it is a property of the whole screen. 

The way you configured it now, it is like it is a property of the last stream. The last stream will silently ignore this property since it does not know what it means. The screen however will just fallback to the default nr_of_columns of 2.

See https://github.com/SvenVD/rpisurv/blob/master/surveillance/conf/surveillance.yml for how the indentation should look like.

As a sidenote,  I am also working on a proper way to allow to configure complete custom layouts for a screen. Donating will help this feature to release faster.

SvenVD, 2020-10-13:
I just released 2.1.9 on the v2latest branch https://github.com/SvenVD/rpisurv/blob/v2_latest/RELEASE_NOTES.md. This release allows you to create custom layouts.

closed 2020-10-13 by SvenVD