SvenVD/rpisurv #104: High Resolution Streams

monza258, 2020-10-13:
Hi, is it possible to stream High Quality Streams?
I have a 3 Pixel camera with to Streams.

First Stream
rtsp://192.168.0.111:554/11
Resolution:2304x1296
Bit rate:1535

Second stream
rtsp://192.168.0.111:554/12
Resolution:640x352
Bit rate:1024

Now i use displaycameras from Github. When I use the First Stream I get a blank screen.
Second Stream is bad Quality and I can't change the Resolution in Camera Settings.
Is there a restriction on the image quality of the camera? Whether 1, 2 or 3px camera?

If there are no restrictions, I would like to switch.

Hardware:
PI4 2GB

Hoppe for answer

SvenVD, 2020-10-13:
Hi,

Support questions should be asked on https://community.rpisurv.net.

Anyways, the restriction is on omxplayer ( which rpisurv uses) in combo with the hardware. There seems to be some constraint between the resolution of the monitor and the resolution of the streams.  If the resolution of a stream exceeds the monitor resolution then omxplayer starts behaving oddly. I have no idea if this still is the case on a raspberry pi 4, but it was the case before.

Another constraint can be the amount of gpu memory you assign on your raspberry. Set this to the highest possible

monza258, 2020-10-15:
Thanks you for your fast comment. Can you tell me what's do with this error bellow? 

```
pi@raspberrypi:~ $ sudo systemctl status rpisurv
● rpisurv.service - Rpisurv Raspberry Pi Surveillance
   Loaded: loaded (/etc/systemd/system/rpisurv.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Thu 2020-10-15 14:44:30 CEST; 15s ago
  Process: 1308 ExecStart=/usr/bin/rpisurv (code=exited, status=1/FAILURE)
```

SvenVD, 2020-10-17:
This means that you are hitting the current assigned memory gpu limits, this can cause lost frames or other artifacts

monza258, 2020-10-17:
> This means that you are hitting the current assigned memory gpu limits, this can cause lost frames or other artifacts

Thanks for Reply. Any idea what I can do to fix it? I have set the gpu in config.txt to 512 and I got this messages.

SvenVD, 2020-10-17:
If you are visually satisfied on what is showing I would ignore the error messages.

monza258, 2020-10-17:
> If you are visually satisfied on what is showing I would ignore the error messages. 

Ok. One question pls. Can you show me a functional example? I just can't get it to work, I would like to display 4 simple cameras.

SvenVD, 2020-10-17:
https://gist.github.com/SvenVD/f744af9298bce65c7eaaba96a991d36e

monza258, 2020-10-17:
Now I am using displaycameras on my pi and I don't have any problems with it but yours have more options and I will change to it. But I don't know why I have the memory problems. Streams also dosnt work Fine. 😔

I will test it again. Hopefully 

Thanks for your great work.

adamderuwe, 2020-10-17:
@monza258 - look into this as an alternative to rpisurv. [https://github.com/raspicamplayer/camplayer]( https://github.com/raspicamplayer/camplayer)

monza258, 2020-10-17:
> If you are visually satisfied on what is showing I would ignore the error messages. 

The streams dosnt work fine. a lot of crashes.

SvenVD, 2020-10-17:
> 
> 
> > If you are visually satisfied on what is showing I would ignore the error messages.
> 
> The streams dosnt work fine. a lot of crashes.

Then probably you raspberry can not handle the resolutions of the streams you configured. Try using the substreams of the cameras

closed 2020-11-20 by SvenVD