SvenVD/rpisurv #105: Speco Camera won't connect

burt-macklin1, 2020-10-20:
Hello, ive been using this for a while and im setting one up at work now and we use Speco cameras. I can get the stream up in VLC using rtsp://<User>:<Password>@<IP>:554/ (i removed user name and password, But its filled out) but the Pi just sits on Connecting.  Not sure why VLC connects but the Pi cant. Hopefully im just missing something easy.

[image]

SvenVD, 2020-10-21:
Hi,

It is advised to use the community.rpisurv.net/ for support.  

To be able to troubleshoot, you should provide following information:
https://www.tapatalk.com/groups/rpisurv/guidelines-t13.html

If you could give the most recent logs of /var/log/daemon.log and /usr/local/bin/rpisurv/logs/main.log we probably can already see what is wrong.

Thanks

burt-macklin1, 2020-10-21:
When im at work ill grab that log, and i can open a ticket over there if that is easier

burt-macklin1, 2020-10-22:
[main.log](https://github.com/SvenVD/rpisurv/files/5423601/main.log)

SvenVD, 2020-10-22:
Ok I see that omxplayer is crashing, rpisurv detects this and tries to restart the stream however it fails because omxplayer keeps on crashing.

Omxplayer does not mention why it is crashing. So we would have to guess, what is codec and the resolution of your camerastream, maybe try with a substream and make sure basic h.264 is selected. You have to do some trial and error with different resolutions and codec setting for your camerastream to test why omxplayer does not like the current stream presented to it

Please do note that your passwords seem to be in the debug log, because for some reason the core worker logs also appear in your main.log

closed 2020-11-20 by SvenVD