SvenVD/rpisurv #106: fails to start

tonydm, 2020-12-15:
Hello,

Thank you for your code contribution.  Perhaps you can help me identify why the service fails to start.

**OS:**
Raspian Lite

```
pi@raspberrypi:~ $ cat /etc/issue
Raspbian GNU/Linux 10 \n \l

pi@raspberrypi:~ $ uname -a
Linux raspberrypi 5.4.51-v7l+ #1333 SMP Mon Aug 10 16:51:40 BST 2020 armv7l GNU/Linux
```
**Updated:**
`sudo apt-get update && sudo apt-get dist-upgrade -y && sudo apt-get auto-remove -y && sudo apt-get autoclean`

**Installation**
```
pi@raspberrypi:~ $ git clone https://github.com/SvenVD/rpisurv
pi@raspberrypi:~ $ cd rpisurv/
pi@raspberrypi:~ $ sudo ./install.sh
```

**Additional Info**

#### pi@raspberrypi:~ $ systemctl status rpisurv
```
pi@raspberrypi:~ $ sudo systemctl status rpisurv.service 
● rpisurv.service - Rpisurv Raspberry Pi Surveillance
   Loaded: loaded (/etc/systemd/system/rpisurv.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Tue 2020-12-15 09:57:27 MST; 3s ago
  Process: 725 ExecStart=/usr/bin/rpisurv (code=exited, status=1/FAILURE)
 Main PID: 725 (code=exited, status=1/FAILURE)
```

#### /etc/rpisurv.conf
##### I'm very familiar with yaml formatting and spaces vs tabs.  The - url spacing seemed suspect so I tried adjusting to the same 2 spaces but that didn't fix

##### I intend on 4 cameras so I'm using the same feeds for all four streams to set up and test.
```
pi@raspberrypi:~ $ cat /etc/rpisurv.conf 
essentials:
  screens:
    - camera_streams:
        - url: "rtsp://user:pass@192.168.1.254:554/channel=1&stream=1.sdp"
        - url: "rtsp://user:pass@192.168.1.254:554/channel=1&stream=1.sdp"
        - url: "rtsp://user:pass@192.168.1.254:554/channel=2&stream=1.sdp"
        - url: "rtsp://user:pass@192.168.1.254:554/channel=2&stream=1.sdp"
        rtsp_over_tcp: true

#Normally you do not need to configure these
#fallbacks:
#  resolution:
#    width: "1366"
#    height: "768"
```

#### /boot/config.txt (Everything is default except the following)
```
...
disable_overscan=1
...
[all]
#dtoverlay=vc4-fkms-v3d
gpu_mem=512
...
```

#### Logging
1: I implemented the logging.yml file.  FYI: You don't specify where to place that file, only the logging.yml config file itself.  So I assumed it would be located in the source dir
2: Performed a reboot
```
pi@raspberrypi:~ $ ls -alF /usr/local/bin/rpisurv
total 36
drwxr-xr-x 6 pi   pi   4096 Dec 15 10:11 ./
drwxr-xr-x 3 root root 4096 Dec 15 09:38 ../
drwxr-xr-x 2 pi   pi   4096 Dec 15 09:37 conf/
drwxr-xr-x 3 pi   pi   4096 Dec 15 10:11 core/
drwxr-xr-x 2 pi   pi   4096 Dec 15 09:37 images/
-rw-r--r-- 1 pi   pi      0 Dec 15 09:37 __init__.py
-rw-r--r-- 1 pi   pi    832 Dec 15 09:52 logging.yml
drwxr-xr-x 2 root root 4096 Dec 15 10:11 logs/
-rw-r--r-- 1 pi   pi   7946 Dec 15 09:37 surveillance.py
```

##### log file shows nothing useful
```
pi@raspberrypi:~ $ cat /usr/local/bin/rpisurv/logs/main.log 
2020/12/15 10:11:22 - l_default - INFO - Starting rpisurv 2.1.7
2020/12/15 10:11:43 - l_default - INFO - Starting rpisurv 2.1.7
2020/12/15 10:12:03 - l_default - INFO - Starting rpisurv 2.1.7
2020/12/15 10:12:24 - l_default - INFO - Starting rpisurv 2.1.7
2020/12/15 10:12:45 - l_default - INFO - Starting rpisurv 2.1.7
2020/12/15 10:13:06 - l_default - INFO - Starting rpisurv 2.1.7
2020/12/15 10:13:26 - l_default - INFO - Starting rpisurv 2.1.7
2020/12/15 10:13:47 - l_default - INFO - Starting rpisurv 2.1.7
2020/12/15 10:14:08 - l_default - INFO - Starting rpisurv 2.1.7
2020/12/15 10:14:29 - l_default - INFO - Starting rpisurv 2.1.7
2020/12/15 10:14:49 - l_default - INFO - Starting rpisurv 2.1.7
2020/12/15 10:15:10 - l_default - INFO - Starting rpisurv 2.1.7
```

SvenVD, 2020-12-15:
Support questions should be asked on https://community.rpisurv.net/

https://www.tapatalk.com/groups/rpisurv/viewtopic.php?p=16#p16 shows what we need for troubleshooting.

The error log is most probably in /var/log/daemon.log can you see if it is logged there?

Thanks

SvenVD, 2020-12-29:
No response

closed 2020-12-29 by SvenVD